kriswiner/MPU9250 #303: MPU9250 works with Arduino Uno, doesn't work with Teensy3.2

Louis8282, 2018-08-01:
Hello,
I have an mpu9250 breakout board which works on Arduino Uno R3. It has addresses at 0x0C and 0x68  when I scan from the Uno and I had to change the 'who am I' to (c == 0x73) to get it to work.
Today, I tried to get it to work on the Teensy3.2 (my first time using a Teensy) with the code MPU9250BasicAHRS_t3.ino.
Whatever I do, I get:

MPU9250 I AM FF I should be 71
Could not connect to MPU9250: 0xFF

I have tried with 4.7kohm pull up resistors, 10ko pull up resistors and no pull up resistors. I've tried connecting it to pins 17 and 16 instead of 19 and 18.  I've also tried changing 'if (c == 0x71)' from 71 to 73.  Are there any issues with Teensy 3.1 vs Teensy 3.2?

Thanks you

kriswiner, 2018-08-01:
Should just work, so I suppose you have the I2C pins switched or something
like this. If you do an I2C scan, which devices show up on the bus?

Louis8282, 2018-08-01:
If I do an i2c scan using the Teensy, it doesn't find anything.  Not only for the MPU9250 but also other i2c sensors which work with the Arduino.

I'm using the following to scan for I2C devices from the Teensy.

`// I2C Bus Scanner
// -------------------------------------------------------------------------------------------
//
// This creates an I2C master device which will scan the address space and report all
// devices which ACK.  It does not attempt to transfer data, it only reports which devices
// ACK their address.
//
// Pull the control pin low to initiate the scan.  Result will output to Serial.
//
// This example code is in the public domain.
// -------------------------------------------------------------------------------------------

#include <i2c_t3.h>

// Function prototypes
void print_scan_status(uint8_t target, uint8_t all);

uint8_t found, target, all;

void setup()
{
    pinMode(LED_BUILTIN,OUTPUT);    // LED
    pinMode(12,INPUT_PULLUP);       // pull pin 12 low to show ACK only results
    pinMode(11,INPUT_PULLUP);       // pull pin 11 low for a more verbose result (shows both ACK and NACK)

    // Setup for Master mode, pins 18/19, external pullups, 400kHz, 10ms default timeout
    Wire.begin(I2C_MASTER, 0x00, I2C_PINS_18_19, I2C_PULLUP_EXT, 400000);
    Wire.setDefaultTimeout(10000); // 10ms

    Serial.begin(115200);
}

void loop()
{
    // Scan I2C addresses
    //
    if(digitalRead(12) == LOW || digitalRead(11) == LOW)
    {
        all = (digitalRead(11) == LOW);
        found = 0;
        
        Serial.print("---------------------------------------------------\n");
        Serial.print("Starting scan...\n");
        digitalWrite(LED_BUILTIN,HIGH); // LED on
        for(target = 1; target <= 0x7F; target++) // sweep addr, skip general call
        {
            Wire.beginTransmission(target);       // slave addr
            Wire.endTransmission();               // no data, just addr
            print_scan_status(target, all);
        }
        digitalWrite(LED_BUILTIN,LOW); // LED off

        if(!found) Serial.print("No devices found.\n");
        
        delay(500); // delay to space out tests
    }
}

//
// print scan status
//
void print_scan_status(uint8_t target, uint8_t all)
{
    switch(Wire.status())
    {
    case I2C_WAITING:  
        Serial.printf("Addr: 0x%02X ACK\n", target);
        found = 1;
        break;
    case I2C_ADDR_NAK: 
        if(all) Serial.printf("Addr: 0x%02X\n", target); 
        break;
    default:
        break;
    }
}`

kriswiner, 2018-08-01:
Sounds like a question for the Teensy forum....

kriswiner, 2018-08-01:
> Ok, I'll ask a question there. The Teensy is brand new though, do you mean
> it could be using pins other than 18/19 for I2C?

I mean you don't know how to use the Teensy, go ask on their forum how to
use it with I2C devices.

kriswiner, 2018-08-01:
Can you see the address of any i2C device with your new Teensy?

Do you have one of those MPU9250 breakouts that require you enable the I2C
bus?

Louis8282, 2018-08-01:
Teensy scanner works now!
I2C device found at address 0x68  

I used the Arduino one and simply replaced include wire.h with include i2c_t3.h

Louis8282, 2018-08-11:
I must have been a wiring issue as everything works as expected now.

closed 2018-08-11 by Louis8282