kriswiner/MPU9250 #305: Does 9250 CYCLE mode also produce Gyro data or only Accel?

SoulPedal, 2018-08-12:
I'm considering a different configuration from what's shown below (maybe you see something obvious there - the ISR fires but the main code doesn't proceed). 

So I'm considering a better approach for battery life to use WOM and CYCLE to wake up my MPU, read Gyro/Accel data, and go back to sleep. However, there's no mention on reading Gyro data (only Accel) in the literature when setting PWR_MGMT_1 = 0x20 (or 0x21?) for CYCLE mode. I've considered that I'll have to keep the Gyro ready, but will new data be there each CYCLE?

Either way, I suspect I'll run into the same bug and I've tried all configurations of interrupts at both ends. (Could be specific to CC430 config. My Port 1 is set up for the I2C and the Interrupt pin so the Port ISR may be seeing both, but I thought the ISR Vector IV would filter that part. I'm sure this is out of scope here but seems I'm the only one not using Arduino or other kits so hard to get an answer that spans 2 forums.) 

This is the setup for my current approach (strictly INT driven based on Sample Rate Divider). Can you please check my Reg settings? Maybe I'm missing something obvious like a FIFO mode?

 msp430_i2c_write (MPU9250_ADDRESS, PWR_MGMT_1, 1, &InitCMP0); // 00 6B Wake up!
  __delay_cycles(100);
  msp430_i2c_write (MPU9250_ADDRESS, PWR_MGMT_1, 1, &InitCMP1); // 01 Stable PLL Clock Source
  __delay_cycles(200);
  msp430_i2c_write (MPU9250_ADDRESS, CONFIG, 1, &InitCMP3); // 03 DLPF_CFG = bits 2:0 = 011
  msp430_i2c_write (MPU9250_ADDRESS, GYRO_CONFIG, 1, &InitCMP3);   // 03 Accel rate to 1 kHz
  msp430_i2c_write (MPU9250_ADDRESS, SMPLRT_DIV, 1, &InitCMP7);   // 0x07 for slow sampling
  msp430_i2c_write (MPU9250_ADDRESS, INT_PIN_CFG, 1, &InitCMP10);  // 10 INT_Status cleared on Read 

Also tried INT_PIN_CFG 0x00 requiring Int_Status read, and 0x70 w/o INT pull-up on my end. All writes are confirmed with reads afterwards. TI's CC430 goes to sleep, Interrupt fires, but code doesn't continue for the data reads. I only get more interrupts, but I thought I've slowed this thing down enough. 

Full Code is here: https://e2e.ti.com/support/microcontrollers/msp430/f/166/t/716750

Thanks!
John

kriswiner, 2018-08-12:
Not sure I can help here, I have no experience with the MSP430. MPU9250
interrupt works fine for ESP8266, ESP32, nRF52, STM32L4, etc but this is
all with an Arduino overlay on different system layers. Your problem is
likely specific to your MCU. Did you set the MSP430 pin mode to input,
trigger on falling edge? If the MPU9250 interrupt is indeed firing as it
must what else can I say?

Can you read data via polling?

SoulPedal, 2018-08-12:
R/W is fine... I'll figure it out somehow, thanks. 
What about the cycle mode question?

kriswiner, 2018-08-12:
No idea what cycle mode means.

We have implenented simple sleep and wake functions for the MPU9250 such
that when awoken the MPU9250 provides data to the host at the specified
rate and when asleep all sensors are off and the MPU9250 portion of the
current is 20 uA or less. This is straightforward to do.

SoulPedal, 2018-08-13:
Understood. Does Invensense provide examples for the different modes they have besides the WOM flowchart in their product spec? 

With SMPLRT_DIV at 0x07, (and config as required) I should be interrupting at 1K/(1+7) = 125 Hz. 
And since this is an 8 bit register, then 0x0F should be able to div by 1+16 or 58.8 Hz right? I want it to be slow while debugging and maybe 60Hz is plenty.

FYI, our discussion the other day on gyro drift, I get why I sounded clueless, there is no "drift." Without calibration, it would only mean the angles might be say 90.1 deg vs 90. For my case, it doesn't matter but I used it anyway, thanks again.

kriswiner, 2018-08-13:
I think Invensense (now TDK) has software for the MPU9250 but you don;t
really need this.

For example, these functions:

void MPU9250::gyromagSleep()
{
uint8_t temp = 0;
temp = readByte(AK8963_ADDRESS, AK8963_CNTL);
writeByte(AK8963_ADDRESS, AK8963_CNTL, temp & ~(0x0F) ); // Clear bits 0 -
3 to power down magnetometer
temp = readByte(MPU9250_ADDRESS, PWR_MGMT_1);
writeByte(MPU9250_ADDRESS, PWR_MGMT_1, temp | 0x10); // Write bit 4 to
enable gyro standby
delay(10); // Wait for all registers to reset
}

void MPU9250::gyromagWake(uint8_t Mmode)
{
uint8_t temp = 0;
temp = readByte(AK8963_ADDRESS, AK8963_CNTL);
writeByte(AK8963_ADDRESS, AK8963_CNTL, temp | Mmode ); // Reset normal mode
for magnetometer
temp = readByte(MPU9250_ADDRESS, PWR_MGMT_1);
writeByte(MPU9250_ADDRESS, PWR_MGMT_1, 0x01); // return gyro and accel
normal mode
delay(10); // Wait for all registers to reset
}

are probably all you need to put the gyro nd mag to sleep.

They come from this repository, which shows how to manage MPU9250 wake on
motion:

https://github.com/kriswiner/MPU9250/tree/master/MPU9250_BME280_SPIFlash_Ladybug

I haven;t run it for a while so I don;t know how ready it is for your use,
but you'll get the idea of how to do this.

SoulPedal, 2018-08-14:
Getting close... any help would be great! 

Don't know if any of you were aware, but I have one setup where I send my raw foot motion & pressure data wirelessly into an Arduino Uno. If I ever get Soulpedal productized, maybe this forum would be the place for the code and I make the pedals. For now I'm targeting musicians (MIDI and effects).

Measuring my Interrupt pin on a scope, it's running constantly at about 35 KHz (30us period at 50% duty). From TDK literature, it seems the 9250 can only select WOM, Raw Data, or FIFO overflow to propagate to the Interrupt pin. (I'm using Raw Data method, maybe too raw...) 

Since I'm trying to use it to wake up my cpu, I would like read data based upon **SMPLRT_DIV** which should divide as much as down to 4Hz if needed, but that seems to have no effect on the Interrupt freq measured. How do I slow down the raw data or is that full speed no matter what - each sensor element in turn? Otherwise, I'd need another timer on my side and special code to calculate time (MPU430 has no milli() command, yuck!) Even then, I might fall behind or read the same data twice. 

So the only other way I can think of is to Interrupt on FIFO Overflow and read all at once, but that could delay my flow unless I fed into an array and nibbled on that in between read bursts. Or just pull off data sets needed on each "Full" interrupt (is that it?) If so, will the FIFO fill at the rate base on SMPLRT_DIV? They are so vague...

FYI, the WOM would not work as I'm actively controlling things while monitoring an ADC for FSR pressure on my end. It would eventually come in handy to sleep everyone when there's nothing moving (as in not being used for > x time.) And CYCLE mode is just the period used with Accel samples for WOM, so that was the wrong path forward, but sounded good by the name). 

Is everyone just taking readings whenever, and using delta milli() to determining gyro angle? Now I know why Arduino is so popular. 

Here's my latest config:
 msp430_i2c_write (MPU9250_ADDRESS, PWR_MGMT_1, 1, &InitCMP0); // 00 6B Wake up!
  __delay_cycles(100);
  msp430_i2c_write (MPU9250_ADDRESS, PWR_MGMT_1, 1, &InitCMP1); // 01 Stable PLL Clock Source
  __delay_cycles(200);
  msp430_i2c_write (MPU9250_ADDRESS, PWR_MGMT_1, 1, &InitCMP8); // Reset all registers, self clearing
  __delay_cycles(200);
  msp430_i2c_write (MPU9250_ADDRESS, CONFIG, 1, &InitCMP3); // DLPF_CFG = 011, 1KHz 
  msp430_i2c_write (MPU9250_ADDRESS, GYRO_CONFIG, 1, &InitCMP3); // FChoice = 011
  msp430_i2c_write (MPU9250_ADDRESS, **SMPLRT_DIV, 1, &InitCMPFF**); // 0x10 ~50Hz, **FF for 1K/(256+1) ~ 4Hz** 
  msp430_i2c_write (MPU9250_ADDRESS, INT_PIN_CFG, 1, &InitCMP10);  // **INT_Status cleared on any  read**
  msp430_i2c_write (MPU9250_ADDRESS, INT_ENABLE, 1, &InitCMP1); // 10 = Fifo Overflow Enabled. **01 = RAW_RDY_EN to Int pin**

kriswiner, 2018-08-14:
I think you might want to reset the MPU9250 before you select the PLL
source.

I am not sure what you are doing here, but the oscillation frequency of the
gyro is 32 kHz so you should never see 35 kHz on an interrupt. Also, if you
set the sample rates to 4 Hz you should expect to see an interrupt at 4 Hz.
Something is not configured right in your code.

I thought you don;t need the gyro and mag; why not turn them off?

Also, this:

msp430_i2c_write (MPU9250_ADDRESS, GYRO_CONFIG, 1, &InitCMP3); // FChoice =
011

turns off samplerate divider and rund the gyro at the max rate, this is
maybe why the interrupt is not working well.

This is what I am using:

void initMPU9250()
{
 // wake up device
  writeByte(MPU9250_ADDRESS, PWR_MGMT_1, 0x00); // Clear sleep mode bit
(6), enable all sensors
  delay(100); // Wait for all registers to reset

 // get stable time source
  writeByte(MPU9250_ADDRESS, PWR_MGMT_1, 0x01);  // Auto select clock
source to be PLL gyroscope reference if ready else
  delay(200);

 // Configure Gyro and Thermometer
 // Disable FSYNC and set thermometer and gyro bandwidth to 41 and 42 Hz,
respectively;
 // minimum delay time for this setting is 5.9 ms, which means sensor
fusion update rates cannot
 // be higher than 1 / 0.0059 = 170 Hz
 // DLPF_CFG = bits 2:0 = 011; this limits the sample rate to 1000 Hz for
both
 // With the MPU9250, it is possible to get gyro sample rates of 32 kHz
(!), 8 kHz, or 1 kHz
  writeByte(MPU9250_ADDRESS, CONFIG, 0x03);  // 41 Hz bandwidth, 5.9 ms
delay
 // writeByte(MPU9250_ADDRESS, CONFIG, 0x02);  // 92 Hz bandwidth, 3.9 ms
delay

 // Set sample rate = gyroscope output rate/(1 + SMPLRT_DIV)
  writeByte(MPU9250_ADDRESS, SMPLRT_DIV, 0x05);  // Use a 167 Hz rate; a
rate consistent with the filter update rate
                                    // determined in set in CONFIG above

 // Set gyroscope full scale range
 // Range selects FS_SEL and AFS_SEL are 0 - 3, so 2-bit values are
left-shifted into positions 4:3
  uint8_t c = readByte(MPU9250_ADDRESS, GYRO_CONFIG); // get current
GYRO_CONFIG register value
 // c = c & ~0xE0; // Clear self-test bits [7:5]
  c = c & ~0x02; // Clear Fchoice bits [1:0]
  c = c & ~0x18; // Clear AFS bits [4:3]
  c = c | Gscale << 3; // Set full scale range for the gyro
 // c =| 0x00; // Set Fchoice for the gyro to 11 by writing its inverse to
bits 1:0 of GYRO_CONFIG
  writeByte(MPU9250_ADDRESS, GYRO_CONFIG, c ); // Write new GYRO_CONFIG
value to register

 // Set accelerometer full-scale range configuration
  c = readByte(MPU9250_ADDRESS, ACCEL_CONFIG); // get current ACCEL_CONFIG
register value
 // c = c & ~0xE0; // Clear self-test bits [7:5]
  c = c & ~0x18;  // Clear AFS bits [4:3]
  c = c | Ascale << 3; // Set full scale range for the accelerometer
  writeByte(MPU9250_ADDRESS, ACCEL_CONFIG, c); // Write new ACCEL_CONFIG
register value

 // Set accelerometer sample rate configuration
 // It is possible to get a 4 kHz sample rate from the accelerometer by
choosing 1 for
 // accel_fchoice_b bit [3]; in this case the bandwidth is 1.13 kHz
  c = readByte(MPU9250_ADDRESS, ACCEL_CONFIG2); // get current
ACCEL_CONFIG2 register value
  c = c & ~0x0F; // Clear accel_fchoice_b (bit 3) and A_DLPFG (bits [2:0])
  c = c | 0x03;  // Set accelerometer rate to 1 kHz and bandwidth to 41 Hz
  writeByte(MPU9250_ADDRESS, ACCEL_CONFIG2, c); // Write new ACCEL_CONFIG2
register value

 // The accelerometer, gyro, and thermometer are set to 1 kHz sample rates,
 // but all these rates are further reduced by a factor of 5 to 200 Hz
because of the SMPLRT_DIV setting

  // Configure Interrupts and Bypass Enable
  // Set interrupt pin active high, push-pull, hold interrupt pin level
HIGH until interrupt cleared,
  // clear on read of INT_STATUS, and enable I2C_BYPASS_EN so additional
chips
  // can join the I2C bus and all can be controlled by the Arduino as master
   writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x12);  // INT is 50 microsecond
pulse and any read to clear
   writeByte(MPU9250_ADDRESS, INT_ENABLE, 0x01);  // Enable data ready (bit
0) interrupt
   delay(100);
}

SoulPedal, 2018-08-14:
Chris

That was it, removing the GYRO_CONFIG slowed things way down. 

FYI, I use the Gyro and Accel, but I'll play with putting them to sleep when the Accels are not moving for extended periods (eventually). 

Seriously happy!
Thanks,
John

closed 2018-08-14 by SoulPedal