kriswiner/MPU9250 #308: rotation in pitch

amir96, 2018-09-13:
Hi Kris, The rotation in pitch is from -90 to 90 and it causes you have 2 same degree in Euler Conversion(I know it's for using asin function). however in roll & yaw, all degrees are unique, because of using atan2 function.
how can I solve this problem? 
tnx

kriswiner, 2018-09-13:
Don't know what problem you mean, atan provides -80 to 180. So what's the
problem?

amir96, 2018-09-13:
ok, tell me what is the range of pitch in your algorithm?it's from -90 to 90
it increase from 0 to 90 then decrease from 90 to 0, then increase from 0 to -90 and  decrease from -90 to 0. we have for example two 45 (degree) but in opposite direction.

I want to know is it possible to have the range in pitch from -180 to 180?

kriswiner, 2018-09-13:
Pitch is -90 to +90, roll -180 to 180, yaw 0 to 360.

amir96, 2018-09-14:
I found the solution. for convert the range of your pitch from (-90/90) to (0/360), you should use other method of Conversion between quaternions and Euler angles. 
for kris winer code I uncomment the code which gives the pitch:
   yaw   = atan2(2.0f * (q[1] * q[2] + q[0] * q[3]), q[0] * q[0] + q[1] * q[1] - q[2] * q[2] - q[3] * q[3]);    
   // pitch = -asin(2.0f * (q[1] * q[3] - q[0] * q[2]));
   roll  = atan2(2.0f * (q[0] * q[1] + q[2] * q[3]), q[0] * q[0] - q[1] * q[1] - q[2] * q[2] + q[3] * q[3]);  
   pitch *= 180.0f / PI;
   yaw   *= 180.0f / PI; 

and replace it by new method for getting pitch:
    float x7 = 2 * (q[1]*q[3] - q[0]*q[2]);
    float y7 = 2 * (q[0]*q[1] + q[2]*q[3]);
    float z7 = q[0]*q[0] - q[1]*q[1] - q[2]*q[2] + q[3]*q[3];
    
    if(z7>0) //range from 0 to 90 and from 0 to -90
    {
    pitch = atan(x7 / sqrt(y7*y7 + z7*z7));
    }
    else if(z7<0) //to get range from 90 to 180 and ftom -90 to -180
    {
    pitch = - (PI)-atan(x7 / sqrt(y7*y7 + z7*z7));
    }
    if(pitch > 0) //to get range from 0 to -360
              {
                pitch = -2*PI + pitch;
              }

              pitch = abs(pitch);
The output data if you have calibrated it carefully, is very exact .