kriswiner/MPU9250 #309: What is the best way to work with two MPU9250 sensors on one I2C bus?

Sterpa, 2018-09-17:
Dear Kris,
what is the best way to work with two MPU9250 sensors on one I2C bus, 
without multiplexers or special boards like EM7180_SENtral?

This is a very old question, and he has risen again in your branches, mjs513, and Brian Taylor. 
https://github.com/kriswiner/MPU9250/issues/68
https://github.com/kriswiner/MPU9250/issues/205
https://github.com/mjs513/FreeIMU-Updates/issues/83
https://forum.pjrc.com/threads/37891-MPU-9250-Teensy-Library/page8

But we still have not come up with a simple solution to how to work with two sensors on the same bus (AK8963 always have 0x0c addres).

As I understand, there are two options:
1. Use pass through (as suggested by Kris)
```
enable 0x68 pass through
disable 0x69 pass throough
read 0x68 and 0x0c
...
disable 0x68 pass through
enable 0x69 passthrough
read 0x69 and 0x0c
```
2. Use I2C master/slave mode (as suggested by mjs513 in FreeIMU).

But the first way was too slow, slow switching pass through mode.
And on the second way there are no examples with the code... In Brian's library (the author of the idea) there are no examples for the two sensors, and FreeIMU library is very complex, and I was not able to extract from it a part of the code with the implementation of the slave mode...

Maybe someone will help with an example of an effective code for two sensors (for September 2018)))? It would be nice to add this example to one of the popular libraries for working with the MPU9250. Or explain a little more about slave mode, how to use, litle code example? How to switch between two AK8963 in master/slave mode?

kriswiner, 2018-11-04:
[This](https://github.com/kriswiner/MPU9250/tree/master/AK8963_as_slave) is the start of how to do this correctly.

kriswiner, 2018-12-09:
And [this](https://github.com/kriswiner/MPU9250/tree/master/Dual_MPU9250) is the end of how to do it correctly.