kriswiner/MPU9250 #311: a good way to calibrate the magnetometer

larryyyyy, 2018-09-22:
May I ask that is there any better way to calibrate the magnetometer?
I have tried to do the figure 8 calibration, but the result can only depends on your luck.
is it the problem of my skill or it is a common problem?

[image]
this is one of my best calibration and i cannot do this another time. 
will there be any system (hardware or software) that I could use to calibrate the magnetometer?

and one more problem, I found that you mentioned about you don't use the soldered board mpu9250 due to different reasons. I am currently using the gy-521 board for my testing. should i instead getting a raw invensense mpu9250 chip for better quality? Thank you so much!

kriswiner, 2018-09-22:
I assume you have read this
<https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration>
.

Figure eight means move the board in all possible 3D orientations, rotate
up, down, left, right sidways, backwards, forwards, tilted up left
downright down left up right while rotating slowly (one revolution per
second or two. The idea is to sample the entire 3D spherical response
surface. if you plot Mx vs My, or Mx vs Mz, or My vs Mz you should see a
circu;ar cloud of points centered at the origin.

If your MPU9250 breakout board is not well designed you could have problems
but I would try to calibrate what you have first.

This
<https://www.tindie.com/products/TleraCorp/mpu9250-add-ons-for-ladybug/> is
a well-designed MPU9250 breakout board. And there are others
<https://www.tindie.com/products/onehorse/mpu9250-teensy-3x-add-on-shields/>

larryyyyy, 2018-09-22:
Thank you so much for your quick respond! I have read the article. 
I guess I am going too fast in doing the figure eight. I will try harder to calibrate them again. Thank you!