kriswiner/MPU9250 #314: yaw slowly converged on quick rotate around z

robotdreamers, 2018-10-10:
Hi Kris, thank you very much for the great resource. I will definitely buy you a glass of beer when we meet?

I have three quick questions on yaw measurement.

I am using Madgwick algorithm. What happened is when I rotate the MPU around z quickly, yaw will always go up to `y`+~4, then it slowly come back down to a true value which `y`. The convergence would normally took 4-5s.

Also, there is a big warmup time of about 1 minute for the the yaw to converge to the stable value. Why is it?

And lastly, I see you use some kind of location based bias in your calibration algorithm `magbias`. Where did you get those numbers?

Thanks.

kriswiner, 2018-10-10:
Are you calibrating your sensors? How?

There is no location-based bias, there is correction for local magnetic
declination.

If your orientation solution is laggy you need a faster MCU.

robotdreamers, 2018-10-10:
So I did try to do the calibration but the output of the calibration is not good.
Now that you said, may be the issue is the MCU. I am using Mega2560. Do you have any comment?
I am using max sample rate + max low pass filter rate. I have tried with lower freq but still got the same problem.

kriswiner, 2018-10-10:
200 H accel and gyro rate, 100 Hz mag rate, 41 hz LPF. You could also
double gyro and accel rates or even go to 1 kHz with 100 Hz LPF. But the
fusion rate should be several kHz. Not sure if an AVR can do this. Might
try one of these
<https://www.tindie.com/products/TleraCorp/dragonfly-stm32l47696-development-board/>
or these
<https://www.tindie.com/products/TleraCorp/ladybug-stm32l432-development-board/>

robotdreamers, 2018-10-11:
I measured the time it takes between each readsensor. It is about 4ms on my board. So it seems the sample rate is only about 250Hz. Does this sound like it is too low as you mentioned the fusion rate should be several khz?

Also, if the sensor output rate is only 200Hz i.e. SRD=4. How can fusion rate to be at Khz level.
Thanks.

kriswiner, 2018-10-11:
The fusion rate will be determined by how many iterations of the fusion
filter can be accomplished by your MCU for each new data set.

robotdreamers, 2018-10-11:
Since after each sensor read, we apply the filter update once. So basically my `deltat=4ms` which means the fusion rate is 250Hz way below what you said of few Khz. is my logic correct?

kriswiner, 2018-10-11:
In my sketches the fusion function is iterated while the MCU waits for new
data, so the rate depends on how fast the MCU is. For very fast MCUs (like
STM32L4) I limit the iterations to 10 or 20.

robotdreamers, 2018-10-11:
ok so I change the code to make it read sensor on interrupt. and keep the filter update running constantly, the deltat is still around 4ms. So clearly Mega board seem too slow, do you think?

kriswiner, 2018-10-11:
Is this 16 MHz AVR? Probably yes, too slow.

robotdreamers, 2018-10-11:
yes 16Mhz

kriswiner, 2018-10-11:
Tooooo sloooooow

robotdreamers, 2018-10-11:
will order your board then :)