kriswiner/MPU9250 #315: Arduino stuck at initMPU9250()

melitacode, 2018-10-15:
Hi, I'm compiling and executing the MPU9250BasicAHRS_I2C.ino example or an Arduino nano but the program is stopping at the "myIMU.initMPU9250();" line. I know this because I placed print lines before and after this function call and the program only outputs the ones before.

The value in the WHO_AM_I_MPU9250 register is 0x73 and the self-test runs fine every time.

What could be the problem?

melitacode, 2018-10-15:
No, I've already corrected the check to look for 0x73 instead of 0x71.

The problem is related to the _csPin.

(_csPin == NOT_SPI) should return false for an I2C connection right?

kriswiner, 2018-10-15:
No idea, this is Sparkfun's code, no?

If you have a nCS pin on your breakout this needs to be HIGH to work with
I2C, but you are already getting the WHO_AM_I right? Or are you trying to
use SPI?

Somewhat confused....

melitacode, 2018-10-17:
I commented this line and it worked.

`setupMagForSPI();`

I think the MPU9250 is working better without calibration than with calibration because when I calibrate the values of the pitch and roll only cover -60 to 60 degrees.