kriswiner/MPU9250 #318: MPU9250_BMP280_BasicAHRS_t3 for GY-91 ?

funwey, 2018-10-28:
Hi Kris;

I connect two units of GY-91 (with one of the SDO connect to 3.3 V) to a mega2560 board, and use your MPU9250_BMP280_BasicAHRS_t3 code and only included the wire.h : 
#include "Wire.h"   
//#include <i2c_t3.h>
//#include <SPI.h>

plus some other modification to the code so that it compiled and uploaded.  The serial output was a follow:
Scanning...
I2C device found at address 0x68  !
I2C device found at address 0x69  !
I2C device found at address 0x76  !
I2C device found at address 0x77  !
done

MPU9250 9-axis motion sensor...
MPU9250 I AM 73 I should be 71
Could not connect to MPU9250: 0x73

...
I have also used the "MPU9250BasicAHRS" code on this same hardware and it also replied that '73' was found.  I previously ran the same code on MPU9250 and it was OK.  I have tried switching the 0x68 to 0x69 and it still gave the same problem.

Would like to know if there was previously any problem with GY-91's combined with mega 2560 with your code?

Rgds

Fun Wey

kriswiner, 2018-10-28:
These are MPU9255 on your boards, they have WHO_AM_I 0x73.

" MPU9250 I AM 73 I should be 71"

change the check to 0x73 from 0x71 and the sketch should work.

mjs513, 2018-11-13:
I've run into that a few times and after changing to 0x73 I can confirm it does work no problem.

funwey, 2018-11-14:
Thanks Kris.  Yes I have tried the code it works.  Will buy you beer sometime.