kriswiner/MPU9250 #319: which sketch is good for arduino pro mini & GY521 board?

amir96, 2018-10-30:
Hi kris, I use of arduino pro mini & GY521 board for mpu9250. First which sketch is good for my situation? because your sketches are for tenssy & adafruit. second I used your code that chnaged a litlle, my yaw after changing suddenly backed to fix mount. for example if you turn about 30 degree, it will change but after a short time will back to last mount. what is the soloution?

kriswiner, 2018-10-30:
Don't know, maybe this one:

https://github.com/kriswiner/ESP8285/tree/master/MPU9250

you will have to modify it for your particular board.

Did you calibrate your sensors? How?

amir96, 2018-10-30:
calibrating this sensor is very hard. I suggest you to produce a video for calibrating this sensor. I haven't calibrated this sensor, but I think this problem dosen't mention to calibration.
Do you think can I get good resault by arduino?

kriswiner, 2018-10-30:
No, and especially if you don't calibrate your sensors. Try something like
this <https://www.tindie.com/products/onehorse/esp8285-development-board/>
or this
<https://www.tindie.com/products/TleraCorp/ladybug-stm32l432-development-board/>
instead.