kriswiner/MPU9250 #320: Mag calibration data

amir96, 2018-11-03:
Hi kris,
These are my data for mag calibration.
first column data changes when I rotate in figure 8, but two other are fix.
what do you think? my sensor has been wasted?
Mag Calibration: Wave device in a figure eight until done!
0	0	0	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
768	2212	106	 
768	2212	106	 
902	2212	106	 
768	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
894	2212	106	 
894	2212	106	 
897	2212	106	 
900	2212	106	 
909	2212	106	 
768	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
768	2212	106	 
899	2212	106	 
768	2212	106	 
768	2212	106	 
768	2212	106	 
903	2212	106	 
768	2212	106	 
907	2212	106	 
907	2212	106	 
907	2212	106	 
907	2212	106	 
907	2212	106	 
907	2212	106	 
907	2212	106	 
907	2212	106	 
907	2212	106	 
894	2212	106	 
768	2212	106	 
768	2212	106	 
894	2212	106	 
905	2212	106	 
894	2212	106	 
910	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
901	2212	106	 
894	2212	106	 
768	2212	106	 
902	2212	106	 
894	2212	106	 
894	2212	106	 
898	2212	106	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
908	2212	106	 
901	2212	106	 
900	2212	106	 
894	2212	106	 
768	2212	106	 
768	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
894	2212	106	 
768	2212	106	 
906	2212	106	 
768	2212	106	 
894	2212	106	 
894	2212	106	 
904	2212	106	 
904	2212	106	 
904	2212	106	 
904	2212	106	 
904	2212	106	 
904	2212	106	 
904	2212	106	 
904	2212	106	 
904	2212	106	 
904	2212	106	 
906	2212	106	 
906	2212	106	 
768	2212	106	 
894	2212	106	 
897	2212	106	 
899	2212	106	 
Mag Calibration done!

amir96, 2018-11-03:
I'm using of GY521 breakout board(blue). and my micro is arduino uno.
I don't know what is passthrough mode. please explain.
and about your third quastion, the message in Serial is:
AK8963 I AM FF I should be 48

amir96, 2018-11-03:
the picture of 
[image]
[image]

my mpu 9250

kriswiner, 2018-11-03:
When you scan the I2C bus what devices do you see?

amir96, 2018-11-03:
I don't understand what is your purpose of scanning I2C bus. do you want the information of serial monitor?
MPU9250 I AM 71 I should be 71
MPU9250 is online...
gyroBias:	2.13	-0.54	0.18
accelBias:	-0.03	-0.03	-0.04
MPU9250 initialized for active data mode....
AK8963 I AM FF I should be 48
AK8963 initialized for active data mode....
Mag Calibration: Wave device in a figure eight until done!
 I have these. I have comment some. if you needed, tell me.

amir96, 2018-11-03:
I have true pitch & roll. I think my magnetometer isn't active.

kriswiner, 2018-11-03:
AK8963 I AM FF I should be 48
Means you are not reading the magnetometer

Try setting nCS to HIGH.

amir96, 2018-11-03:
I don't find nCS . where is that?

amir96, 2018-11-03:
I think my magnetometer is damaged. Is it possible?

kriswiner, 2018-11-03:
One of the pins on your board

amir96, 2018-11-03:
No change found:
AK8963 I AM FF I should be 48
It shows previous message to me.

kriswiner, 2018-11-03:
Are you using INT_CFG to set pass through mode?

amir96, 2018-11-03:
No, what should I do?

kriswiner, 2018-11-03:
set INT_CFG to 0x22

amir96, 2018-11-03:
I add this item to my code:
#define INT_CFG       0x22
but was not useful.

kriswiner, 2018-11-03:
Maybe you need a better breakout board then.

But I think you are not using it correctly.

amir96, 2018-11-03:
I change this item in your code:
#define INT_PIN_CFG     0x22
but not useful. I don't understand, how don't I using it correctly?

kriswiner, 2018-11-03:
You have to write this to the appropriate register.

amir96, 2018-11-03:
can you explain more?
I have examine my board by different libraray, but I haven't gotten magData.

amir96, 2018-11-03:
how can I recognize appropriate register?

kriswiner, 2018-11-03:
Read the data sheet

  // Configure Interrupts and Bypass Enable
  // Set interrupt pin active high, push-pull, hold interrupt pin level
HIGH until interrupt cleared,
  // clear on read of INT_STATUS, and enable I2C_BYPASS_EN so additional
chips
  // can join the I2C bus and all can be controlled by the Arduino as master
//   writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x22);
   writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x12);  // INT is 50 microsecond
pulse and any read to clear

amir96, 2018-11-03:
ok, and which new mpu breakout board do you suggest?

kriswiner, 2018-11-03:
These
<https://www.tindie.com/products/onehorse/mpu9250-teensy-3x-add-on-shields/>
work.

kriswiner, 2018-11-03:
> Excuse me kris,
> this part of my code you mentioned is:
> writeByte(MPU9250_ADDRESS, INT_ENABLE, 0x01); // Enable data ready (bit 0)
> delay(100);
> is it ok?

Could be