kriswiner/MPU9250 #321: Magnetometer calibration

gomezramones, 2018-11-05:
Hi Kris. 

We are building an application with MPU9250 to know drivers behaviour by getting G-force values of the accelerometer. However, the sensor will be in a car and the position could affect the values which are obtained of the accelerometer since the sensor could be rotated and the gravity might altering 3-axis. Therefore, we need to know the yaw, roll and pitch angles to substract the gravity value of each axis.

I calibrated accelerometer and gyroscope following your code and to reduce noise vibration I applied a kalman filter. At this points eveything is perfect but when it comes to calibrate,the magnetometer, it is when I complicated. I don't know if I'm moving the sensors correctly when the code is getting the data to calculate the Mag_Bias and Mag_Scale due to the mx,my, and mz graphs aren't even close to be circle. ( I watched some videos and most of them calibrate it by only rotating).

This is Magnetometer calibration code:

void initAK8963(float *destination)
{
  // First extract the factory calibration for each magnetometer axis
  uint8_t rawData[3];  // x/y/z gyro calibration data stored here
  writeByte(AK8963_ADDRESS, AK8963_CNTL, 0x00); // Power down magnetometer  
  delay(10);
  writeByte(AK8963_ADDRESS, AK8963_CNTL, 0x0F); // Enter Fuse ROM access mode
  delay(10);
  readBytes(AK8963_ADDRESS, AK8963_ASAX, 3, &rawData[0]);  // Read the x-, y-, and z-axis calibration values
  destination[0] =  (float)(rawData[0] - 128)/256. + 1.;   // Return x-axis sensitivity adjustment values, etc.
  destination[1] =  (float)(rawData[1] - 128)/256. + 1.;  
  destination[2] =  (float)(rawData[2] - 128)/256. + 1.;
  writeByte(AK8963_ADDRESS, AK8963_CNTL, 0x00); // Power down magnetometer
  delay(10);
  writeByte(AK8963_ADDRESS, AK8963_CNTL, 0x06); // Set to continous read at 100Hz
//  writeByte(AK8963_ADDRESS, AK8963_CNTL, Mscale << 4 | Mmode); // Set magnetometer data resolution and sample ODR
  int8_t var = readByte(AK8963_ADDRESS,AK8963_CNTL);
  delay(10);
}

void readMagData(int16_t *destination)
{
  uint8_t rawData[7];  // x/y/z gyro register data, ST2 register stored here, must read ST2 at end of data acquisition
  readBytes(AK8963_ADDRESS, AK8963_XOUT_L, 7, &rawData[0]);  // Read the six raw data and ST2 registers sequentially into data array
  uint8_t c = rawData[6]; // End data read by reading ST2 register
    if(!(c & 0x08)) { // Check if magnetic sensor overflow set, if not then report data
    destination[0] = ((int16_t)rawData[1] << 8) | rawData[0] ;  // Turn the MSB and LSB into a signed 16-bit value
    destination[1] = ((int16_t)rawData[3] << 8) | rawData[2] ;  // Data stored as little Endian
    destination[2] = ((int16_t)rawData[5] << 8) | rawData[4] ; 
   }

void Calibrate_Mag(float *Bias, float *Bias1){
  int16_t Aux[3];
  int32_t mag_bias[3] = {0, 0, 0},mag_scale[3] = {0, 0, 0};;
  int jj = 30;
  int16_t mag_max[3] = {-32767, -32767, -32767}, mag_min[3] = {32767, 32767, 32767};
  Serial.println("Calibrating Magnetometer");
  for (int i = 0; i < 1500; i++){
    readMagData(Aux);
    for (int j = 0; j < 3; j++){
      if (Aux[j] < mag_min[j])
        mag_min[j] = Aux[j];
      if (Aux[j] > mag_max[j])
        mag_max[j] = Aux[j];
    }
    delay(12);
  }
  mag_bias[0]  = (mag_max[0] + mag_min[0])/2;  // get average x mag bias in counts
  mag_bias[1]  = (mag_max[1] + mag_min[1])/2;  // get average y mag bias in counts
  mag_bias[2]  = (mag_max[2] + mag_min[2])/2;  // get average z mag bias in counts

  Bias[0] = (float)mag_bias[0]*magCalibration[0]*mRes;  // get actual magnetometer value, this depends on scale being set
  Bias[1] = (float)mag_bias[1]*magCalibration[1]*mRes;  
  Bias[2] = (float)mag_bias[2]*magCalibration[2]*mRes;

  // Get soft iron correction estimate
  mag_scale[0]  = (mag_max[0] - mag_min[0])/2;  // get average x axis max chord length in counts
  mag_scale[1]  = (mag_max[1] - mag_min[1])/2;  // get average y axis max chord length in counts
  mag_scale[2]  = (mag_max[2] - mag_min[2])/2;  // get average z axis max chord length in counts

  float avg_rad = mag_scale[0] + mag_scale[1] + mag_scale[2];
  avg_rad /= 3.0;
  Serial.println("Done Calibration");
}

kriswiner, 2018-11-05:
> I realize that my mistake was Mscale was MFS_16Bits when it had to be
> MFS_14bits. I don't if those graphs show a good calibration
>
> [image: image]
> <https://user-images.githubusercontent.com/40184703/48006960-f315dc80-e0ec-11e8-989f-873e263a04af.png>
>
> [image: image]
> <https://user-images.githubusercontent.com/40184703/48006987-045ee900-e0ed-11e8-9dd4-be06de23addd.png>
>
> [image: image]
> <https://user-images.githubusercontent.com/40184703/48007017-16408c00-e0ed-11e8-82e7-74765150a3c2.png>

You might need to increase the time for collecting and avergaing the mag
data, but this is probably good enough.

What you want to do is use the quaternions to estimate linear acceleration
(with gravity removed) this is done in most of the sketches already.

kriswiner, 2018-11-05:
> Thanks for the answer Mr. Kris. Other question, can I calculate where is
> the North, East, West and South with AK8963?

No, calculate the heading from the quaternion.

kriswiner, 2018-11-09:
> Sorry Mr. Kris but when you said to increase the time for colleting data,
> you was referring to calibration or plotting.

Calibration.

kokhua90, 2018-12-20:
Hi Kriswiner,

I implemented the calibration code.  However, AK8963 mag scale (mG) = 1.11; 1.15; 0.82. 
Then, no response to pitch roll yaw. Please enlighten me which part i am not doing right.

MPU9250 9-axis motion sensor...
MPU9250 I AM 73 I should be 71
MPU9250 is online...
x-axis self test: acceleration trim within : -5.7% of factory value
y-axis self test: acceleration trim within : -8.8% of factory value
z-axis self test: acceleration trim within : 3.9% of factory value
x-axis self test: gyration trim within : 4.2% of factory value
y-axis self test: gyration trim within : -2.1% of factory value
z-axis self test: gyration trim within : 2.6% of factory value
 Calibrate gyro and accel
accel biases (mg)
-155.94
666.56
-303.22
gyro biases (dps)
-88.24
-23.93
58.09
MPU9250 initialized for active data mode....
AK8963 I AM 48 I should be 48
AK8963 initialized for active data mode....
Mag Calibration: Wave device in a figure eight until done!
mag x min/max:
265
-443
mag y min/max:
522
-159
mag z min/max:
222
-739
Mag Calibration done!
AK8963 mag biases (mG)
-158.47
324.40
-444.26
AK8963 mag scale (mG)
1.11
1.15
0.82
X-Axis sensitivity adjustment value 1.19
Y-Axis sensitivity adjustment value 1.20
Z-Axis sensitivity adjustment value 1.15
0, 0, 0
Yaw, Pitch, Roll: 13.80, 0.00, 0.00
Grav_x, Grav_y, Grav_z: 0.00, 0.00, 1000.00 mg
Lin_ax, Lin_ay, Lin_az: 0.00, 0.00, -1000.00 mg
rate = 0.03 Hz
0, 0, 0
Yaw, Pitch, Roll: 13.80, 0.00, 0.00
Grav_x, Grav_y, Grav_z: 0.00, 0.00, 1000.00 mg
Lin_ax, Lin_ay, Lin_az: 0.00, 0.00, -1000.00 mg
rate = 10925.98 Hz

kriswiner, 2018-12-20:
No idea, look at one of the sketches in this repository for an example of
how I do it.

https://github.com/kriswiner/MPU9250/blob/master/MPU9250_BME280_SPIFlash_Ladybug/MPU9250_BME280_SPIFlash_Ladybug.ino

kokhua90, 2018-12-23:
From the function void MadgwickQuaternionUpdate(float ax, float ay, float az, float gx, float gy, float gz, float **mx**, float **my**, float mz)

however, the input  MadgwickQuaternionUpdate(-ax, ay, az, gx*pi/180.0f, -gy*pi/180.0f, -gz*pi/180.0f,  **my**,  **-mx,** mz);

may I know why is input is reverse? how to align to north?

kriswiner, 2018-12-23:
The Madgwick function requires scaled sensor data to be fed in either an
ENU or NED configuration. The user selects which accel axis will be defined
as North (meaning when this axis is pointing to True North, the qauternion
will read 1 0 0 0. The based on the relative orientation of the other
sensor axes the user identifies which are pointed East and Up or Down,
depending on the orientation convention selected. So for NED convention,
the Madgwick filter is loaded with An, Ae, Ad, Gn, Ge, Gd, Mn, Me, Md.

gomezramones, 2019-01-29:
Hi Kris.

I have a problem calculating MadgwickQuaternionsUpdate function. I followed your code that you share for MPU9250, calibrating the magnetometer (the pictures are above). So, it seems the code calculates the angles (yaw, roll and pitch) ok at the beginning, but when I start to change the IMU position the change rate is not correct. For example, at the beginning I got an angle of 90°, then I rotate 90° the device and it throws me a maximum peak of 170° to stabilize on 135 (it must be 180°)

kriswiner, 2019-01-29:
See comment above. if you are properly calibrated (accel, gyro, mag, all
three) then this is the likely problem.

gomezramones, 2019-01-29:
This are my graphs after calibration, besides I check the accelerometer values and there's not offset

[image]
[image]
[image]

I'm confused regarding how to know the orden of the parameter of Madgwick. As you said before, the Madgwick filter is loaded with An, Ae, Ad, Gn, Ge, Gd, Mn, Me, Md; thus, if my North is Y-axis, East is X-Axis and Down is -Z-axis, Madgwick parameters would be (Ay, Ax,Az,Gy,Gx,Gz,Mx,My,-Mz). Is that correct?

kriswiner, 2019-01-29:
Looks OK, mag calibration could be improved.

kriswiner, 2019-01-30:
> Hi Kris.
>
> I calibrate the magnetometer yet again and this are the graphs
>
> [image: image]
> <https://user-images.githubusercontent.com/40184703/51998448-11db0380-248f-11e9-9115-8ef206a347ba.png>
> [image: image]
> <https://user-images.githubusercontent.com/40184703/51998457-19021180-248f-11e9-8dfa-28e8197cadb9.png>
> [image: image]
> <https://user-images.githubusercontent.com/40184703/51998497-2c14e180-248f-11e9-89c8-046a40cdd27d.png>
>
> But still the gyroscope has a lot of noise when the IMU9250 is not moving:
>
> -0.02 -16.45 -0.58
> -0.63 -25.27 -0.95
> 0.05 -2.06 1.77
> 0.55 -0.52 3.25
> -0.47 -8.42 0.76
> 1.10 10.16 4.76
> -0.52 6.68 -3.31
> 0.60 2.23 -1.94
> -0.09 -13.72 1.05
> 0.14 -4.00 2.32
> -0.02 15.59 5.46
> 0.31 4.56 -3.01
> 0.40 13.87 -1.57
> -0.58 7.29 1.56
> -0.15 9.78 2.64
> 0.76 -1.74 1.88
> 0.24 20.29 1.97
> -0.12 -25.22 -4.29
> 0.00 -10.10 0.81
> 0.61 9.31 1.31
> 1.02 9.23 -0.38
> -0.17 1.80 -1.77
> 0.79 10.89 -3.54
> -0.98 0.08 0.40
> -0.11 -5.25 -1.11
> 0.00 -7.54 2.85
> -0.89 3.05 2.03
> 1.56 -11.12 3.02
> 0.61 -13.44 -2.23
> -0.58 26.49 1.79
> 0.82 -16.54 0.79
> -0.72 -8.59 -3.77
> -0.34 9.89 -0.21
> -0.17 2.84 2.64
> 0.99 4.50 3.45
> 1.07 6.67 2.70
> 0.31 9.09 1.25
> 0.46 -6.55 2.62
> -0.37 16.56 1.98
> -1.43 -4.14 -0.84
> 0.26 0.66 0.66
> 0.34 -6.68 0.23
> 0.35 -3.31 1.05
> -0.72 -1.75 0.73
> -0.43 15.15 -2.75
> -0.31 -15.69 -1.21
>
> PS: I followed your code to calibrate Accel and Gyro

Something wrong with your gyro, or conversion to dps. The readings at rest
should be 0, 0, 0 dps.

gomezramones, 2019-01-30:
So, Do I change the sensor? I'm working with an ATMega

kriswiner, 2019-01-30:
You can try another sensor breakout, or maybe something is wrong with your
sketch?

gomezramones, 2019-01-30:
Ok, I'll just keep looking at my code to see if I can solve the problem.  But it is strange since I check the register where the bias must be pushed and it save all the data

kriswiner, 2019-01-30:
This has nothing to do with the offset bias. Your gyro should show stable
values when motionless. What do the raw data registers show?

gomezramones, 2019-01-30:
This are the values
XG_OFFSET_H:255
XG_OFFSET_L:215
YG_OFFSET_H:255
YG_OFFSET_L:249
ZG_OFFSET_H:0
ZG_OFFSET_L:11

kriswiner, 2019-01-30:
OK, let's try again.

Print out the raw gyro register values (not the offsets, no one cares about
this) on the serial monitor for a few seconds while the sensor board is
flat and motionless.

They should be close to zero counts and constant, not changing...are they?

gomezramones, 2019-01-30:
Sorry, I misunderstood. This are the raw data

-12.00 -884.00 22.00
-154.00 122.00 -246.00
-142.00 2184.00 256.00
80.00 966.00 138.00
-74.00 -3172.00 640.00
-72.00 -1734.00 -420.00
96.00 -1196.00 330.00
42.00 3088.00 -644.00
-38.00 286.00 84.00
86.00 -68.00 536.00
42.00 34.00 302.00
-106.00 -574.00 148.00
144.00 -94.00 -484.00
-48.00 1780.00 70.00
106.00 1380.00 -120.00
-12.00 -3058.00 -396.00
-116.00 4540.00 42.00
40.00 3106.00 76.00
8.00 1982.00 -134.00
42.00 -126.00 118.00
-38.00 -1468.00 198.00
92.00 -724.00 -74.00
-134.00 -776.00 -244.00
74.00 404.00 452.00
-2.00 -1240.00 -52.00
-62.00 922.00 398.00
140.00 -1676.00 -158.00
180.00 -160.00 378.00
-126.00 290.00 -694.00
112.00 1840.00 14.00
34.00 2204.00 -346.00
-58.00 -1434.00 302.00
200.00 -806.00 -544.00
22.00 -936.00 610.00
it is getting the raw data awful

gomezramones, 2019-01-30:
Yes , it is the register values from 0x43 to 0x48. No, I remove the scale part of the code. The first column is Gx, the second is Gy and Gz is the last one.

readGyroData(gyroCount);  // Read the x/y/z adc values
getGres();
gx = gyroCount[0];//*gRes;  // get actual gyro value, this depends on scale being set
gy = gyroCount[1];//*gRes;  
gz = gyroCount[2];//*gRes;

void readGyroData(int16_t * destination){
  uint8_t rawData[6];  // x/y/z gyro register data stored here
  readBytes(MPU9250_ADDRESS, GYRO_XOUT_H, 6, &rawData[0]);  // Read the six raw data registers sequentially into data array
  destination[0] = ((int16_t)rawData[0] << 8) | rawData[1] ;  // Turn the MSB and LSB into a signed 16-bit value
  destination[1] = ((int16_t)rawData[2] << 8) | rawData[3] ;  
  destination[2] = ((int16_t)rawData[4] << 8) | rawData[5] ; 
}

gomezramones, 2019-01-30:
Let me print the two bytes before gathering the values

gomezramones, 2019-01-30:
this the raw data:

raw data Gx (MSB):0	 raw data Gx (LSB):204
raw data Gx (MSB):0	 raw data Gx (LSB):158
raw data Gx (MSB):0	 raw data Gx (LSB):218
raw data Gx (MSB):255	 raw data Gx (LSB):236
raw data Gx (MSB):0	 raw data Gx (LSB):122
raw data Gx (MSB):1	 raw data Gx (LSB):12
raw data Gx (MSB):0	 raw data Gx (LSB):82
raw data Gx (MSB):0	 raw data Gx (LSB):186
raw data Gx (MSB):255	 raw data Gx (LSB):218
raw data Gx (MSB):0	 raw data Gx (LSB):216
raw data Gx (MSB):0	 raw data Gx (LSB):110
raw data Gx (MSB):255	 raw data Gx (LSB):112
raw data Gx (MSB):0	 raw data Gx (LSB):88
raw data Gx (MSB):255	 raw data Gx (LSB):250
raw data Gx (MSB):1	 raw data Gx (LSB):60
raw data Gx (MSB):0	 raw data Gx (LSB):116
raw data Gx (MSB):1	 raw data Gx (LSB):24
raw data Gx (MSB):0	 raw data Gx (LSB):228
raw data Gx (MSB):1	 raw data Gx (LSB):42
raw data Gx (MSB):1	 raw data Gx (LSB):2
raw data Gx (MSB):0	 raw data Gx (LSB):248
raw data Gx (MSB):0	 raw data Gx (LSB):104
raw data Gx (MSB):0	 raw data Gx (LSB):232
raw data Gx (MSB):0	 raw data Gx (LSB):186
raw data Gx (MSB):0	 raw data Gx (LSB):144
raw data Gx (MSB):0	 raw data Gx (LSB):250
raw data Gx (MSB):0	 raw data Gx (LSB):232
raw data Gx (MSB):0	 raw data Gx (LSB):136
raw data Gx (MSB):0	 raw data Gx (LSB):56
raw data Gx (MSB):0	 raw data Gx (LSB):72
raw data Gx (MSB):0	 raw data Gx (LSB):156
raw data Gx (MSB):0	 raw data Gx (LSB):194
raw data Gx (MSB):0	 raw data Gx (LSB):208
raw data Gx (MSB):0	 raw data Gx (LSB):234
raw data Gx (MSB):0	 raw data Gx (LSB):172
raw data Gx (MSB):255	 raw data Gx (LSB):214

kriswiner, 2019-01-30:
so largest difference is ~200 counts, and at 250 dps full scale this is
200/16500 x 250 dps ~ 3 dps which is really quite a lot. Looks like either
your sensor is moving during this data collection or your gyro is broken.
You should be seeing ~25 count difference or less....

gomezramones, 2019-01-30:
I will assume that my sensor is broken since I never moved it to this test

gomezramones, 2019-01-30:
Thanks for the help Kris. Let me change the sensor

kriswiner, 2019-01-30:
Try another one.

kriswiner, 2019-01-30:
You can buy some here
<https://www.tindie.com/products/onehorse/wearable-mpu9250-nano-board/> if
needed.

gomezramones, 2019-01-30:
Hi Kris, I have few MPU9250 so I change it. However, it still send me this raw data that changes a lot

raw data Gx (MSB):255	 raw data Gx (LSB):142
raw data Gx (MSB):0	 raw data Gx (LSB):58
raw data Gx (MSB):0	 raw data Gx (LSB):56
raw data Gx (MSB):0	 raw data Gx (LSB):104
raw data Gx (MSB):0	 raw data Gx (LSB):186
raw data Gx (MSB):0	 raw data Gx (LSB):80
raw data Gx (MSB):0	 raw data Gx (LSB):172
raw data Gx (MSB):0	 raw data Gx (LSB):164
raw data Gx (MSB):0	 raw data Gx (LSB):42
raw data Gx (MSB):0	 raw data Gx (LSB):18
raw data Gx (MSB):255	 raw data Gx (LSB):254
raw data Gx (MSB):0	 raw data Gx (LSB):162
raw data Gx (MSB):1	 raw data Gx (LSB):24
raw data Gx (MSB):255	 raw data Gx (LSB):206
raw data Gx (MSB):0	 raw data Gx (LSB):86
raw data Gx (MSB):0	 raw data Gx (LSB):38
raw data Gx (MSB):0	 raw data Gx (LSB):42
raw data Gx (MSB):0	 raw data Gx (LSB):108
raw data Gx (MSB):0	 raw data Gx (LSB):60
raw data Gx (MSB):255	 raw data Gx (LSB):156
raw data Gx (MSB):0	 raw data Gx (LSB):106
raw data Gx (MSB):0	 raw data Gx (LSB):252
raw data Gx (MSB):0	 raw data Gx (LSB):110
raw data Gx (MSB):0	 raw data Gx (LSB):52
raw data Gx (MSB):0	 raw data Gx (LSB):156
raw data Gx (MSB):255	 raw data Gx (LSB):234
raw data Gx (MSB):0	 raw data Gx (LSB):92
raw data Gx (MSB):0	 raw data Gx (LSB):108
raw data Gx (MSB):0	 raw data Gx (LSB):158

kriswiner, 2019-01-30:
What are you using for gyro full scale?

kriswiner, 2019-01-30:
> Might be the frequency of the gyro? Since FSEL = 2 (10) and the frequency
> is 32KHz with a bandwidth of 8800Hz

This configuration sounds aweful to me. Why not select 1 kHz or 200 Hz gyro
sample rate with a 41 Hz bandwidth?

gomezramones, 2019-01-30:
Yes, I didn't know it had that configuration. Tomorrow I will the setting and the results. Thanks a lot for all the help you gave me Kris

gomezramones, 2019-01-30:
I will set the register and see the results*

gomezramones, 2019-01-31:
Hi Kris.

The problem persist with the Gyroscope. This are the value of registers Configuration and Gyroscope Configuration:

GYRO_CONFIG: 3 --> So FCHOICE_b[1:0] is 11 for 
CONFIG: 3 --> DLPF_CFG is 11 for 41Hz bandwidth and 1KHz

Raw data
raw data Gx (MSB):255	 raw data Gx (LSB):252
raw data Gx (MSB):0	 raw data Gx (LSB):146
raw data Gx (MSB):0	 raw data Gx (LSB):76
raw data Gx (MSB):0	 raw data Gx (LSB):20
raw data Gx (MSB):255	 raw data Gx (LSB):218
raw data Gx (MSB):0	 raw data Gx (LSB):180
raw data Gx (MSB):1	 raw data Gx (LSB):26
raw data Gx (MSB):0	 raw data Gx (LSB):226
raw data Gx (MSB):0	 raw data Gx (LSB):180
raw data Gx (MSB):1	 raw data Gx (LSB):84
raw data Gx (MSB):1	 raw data Gx (LSB):210
raw data Gx (MSB):0	 raw data Gx (LSB):192
raw data Gx (MSB):1	 raw data Gx (LSB):106
raw data Gx (MSB):1	 raw data Gx (LSB):96

kriswiner, 2019-01-31:
What is smplrtdiv?

Please copy my configuration settings and try again.

gomezramones, 2019-01-31:
Hi Kris, I realize that I call first InitMPU9250 and then CalibrationMPU9250, so all the configuration was screw because of that. But now the result of the gyroscope are:
1.05 5.94 -2.15
1.00 5.97 -2.08
0.97 5.85 -2.08
1.01 5.89 -2.11
0.97 5.94 -2.17
0.98 6.00 -2.17
0.95 6.00 -2.14
0.98 5.96 -2.15
0.96 5.97 -2.06
1.01 5.97 -2.03
0.97 5.92 -2.14
0.99 5.94 -2.17
1.04 5.81 -2.17
0.96 5.88 -2.17
1.05 5.85 -2.05
1.10 5.91 -2.12
1.06 5.96 -2.06
1.04 5.92 -2.04
1.00 5.81 -2.11
1.02 5.89 -2.13
1.00 5.97 -2.08
0.92 5.83 -2.04
0.98 5.84 -2.03
1.01 5.84 -2.06
1.01 5.77 -2.01
1.01 5.80 -2.04
0.99 5.85 -2.13
1.01 5.85 -2.08
0.98 5.91 -2.13
0.98 5.91 -2.16
1.04 5.91 -2.17
1.03 6.05 -2.17
1.07 5.91 -2.14
1.00 5.91 -2.13
1.05 5.87 -2.11

The results are not changing, but exists a offset in each axis. I guess the calibration function is not working properly. I checked the accelerometer values and there is a offset:
-2.16 -0.08 3.09
-2.16 -0.08 3.09
-2.16 -0.08 3.09
-2.17 -0.08 3.08
-2.17 -0.08 3.09
-2.17 -0.08 3.09
-2.17 -0.08 3.09
-2.17 -0.08 3.09
-2.17 -0.08 3.09
-2.16 -0.08 3.08
-2.17 -0.08 3.09
-2.16 -0.08 3.08
-2.16 -0.08 3.09
-2.16 -0.08 3.09
-2.16 -0.08 3.09
-2.16 -0.08 3.08
-2.17 -0.08 3.09
-2.17 -0.08 3.09
-2.17 -0.08 3.09
-2.17 -0.08 3.09
-2.16 -0.08 3.09
-2.16 -0.08 3.08
-2.17 -0.08 3.08
-2.16 -0.08 3.09
-2.17 -0.08 3.08
-2.17 -0.08 3.08
-2.16 -0.08 3.09
-2.17 -0.08 3.08

kriswiner, 2019-01-31:
Doesn't look right. Your code is messed up. Why not simply copy one of the
skecthes in my repository that should work well?

29676, 2020-07-01:
Hi ;
I have a question; 
What MCU you used for this job?

gomezramones, 2020-07-01:
ATMega2560 8bits. It will work but just bear in mind that it was a test. If you are going to implement for a product, just must have at least 32bits for a better performance and DSP since float operation is a overkilling for this type of MCU and in a certain point you are going to crash the code

kriswiner, 2020-07-01:
I don't understand your question, but I would use this
<https://www.tindie.com/products/tleracorp/ladybug-stm32l432-development-board/>
one.

29676, 2020-07-01:
Thanks
I also want to do just one simple test with avr and then do it with arm32. 
 Do you get Euler angles with atmega?