kriswiner/MPU9250 #322: Rotating output of Madgwick filter

amarsh, 2018-11-05:
I have a question that is really very basic but I am struggling with. Consider a scenario where you control the board containing the IMU, but you do not have control over where the board gets installed. In this case the filter is going to return the orientation of the board, not the orientation of the thing you really care about. What is the correct way to transform the filter output into the desired reference frame?

I assume you do something like the following. Collect some amount of data while the vehicle is in a steady state from the desired reference frame and then from the installation location. For each position average the acquired data and zero the gyro values (since we are at steady state). Repeatedly feed the computed values into the filter until the output converges. This should give you a quaternion representing the orientation at each point of interest.

At this point I am not sure what to do. Can you simply compute roll pitch and yaw for each location and then compute a constant to remove from measurements at the installed location? Or do you need to compute and perform a quaternion rotation prior to converting to roll, pitch, and yaw?

kriswiner, 2018-11-05:
The Madgwick (or Mahony) filter is an absolute orientation filter. When
properly used the quaternions are 1, 0, 0, 0 when the board is oriented
toward North parallel to the Earths surface.

You can select which axis (accel, say) is defined to be North. The Madwick
algorithm requires a consistent sensor data input depending on the
convention you select. I use NED but ENU can also work, so that the entry
has to be (aN. aE, aD, gN, gE, gD, mN, mE, mD) or (aE, aN, aU, gE, gN, gU,
mE, mN, mU). As long as the data entry is consistent for one of these
conventions, the yaw will always read 0 when the device points North.

amarsh, 2018-11-05:
Thank you for the response. I think my wording on the original post was off. I agree that the filter is always going to return the orientation of the sensor with respect to earth. However I really want the orientation of the vehicle with respect to earth. The sensor is installed in the vehicle in some arbitrary location and orientation that I do not have control over. Swapping an axis may not fully align the sensor with the vehicle. However once the sensor is installed it will not be moved. So the relationship between both sensor and vehicle orientation should be constant. Can I simply calculate some roll, pitch, and yaw offsets to remove from the processed sensor data to give me the vehicle's orientation? I feel like this is the correct answer but I don't know if I am violating some other assumption with this approach.

kriswiner, 2018-11-05:
Yes