kriswiner/MPU9250 #323: MPU9250 Arduino Pro Mini calibration

gojunnyo, 2018-11-09:


gojunnyo, 2018-11-09:
By the way, my hardware connection is
MPU9250/MPU9255 ------------------ ARDUINO PRO MINI
            VCC               ------------------ 3.3 Volts
           GND               ------------------- GND
            SCL                ------------------- SCL 
            SDA               ------------------- SDA

Although I can say that this doesn't help, but I want to clarify that I have done my connections properly.

[image]

kriswiner, 2018-11-09:
Just check that SDA/SCL aren;t reverswed. I can;t remember which way is
which off the top of my head. You do need a GND connection, of course.
Which MPU9250 breakout are you using? And what do you mean "does not work"?

gojunnyo, 2018-11-09:
Thank you for your fast reply sir!

I forgot to mention, in I2C scanner sketch, the sensor is acknowledged.
What will happen if the SDA and SCL are reversed? (Enlighten me with your knowledge)
Just as I checked, my connections are correct.

I just wanted to know the offsets of the gyroscope and accelerometer of my MPU9250. Because I cannot find any sketches that could help me. Unless if I have MPU6050. I cannot find any MPU9250 calibration sketches that would spit up gyroscope and accelerometer offsets. 

[image]

kriswiner, 2018-11-09:
Really, no sketches at all?

https://github.com/kriswiner/MPU9250

gojunnyo, 2018-11-09:
I already saw this
https://github.com/kriswiner/MPU9250/blob/master/MPU9250BasicAHRS.ino

but I am not sure about what part of the code spits out the offset of the sensor.

I honestly don't understand the whole process of the sketch, but I do know it outputs actual data of the orientation of the sensor. I am eager to know about where in the code that integrates values or the like (as long as it outputs an offset).

I just wanted to eliminate the runtime calibration everytime I power on the sensor.

kriswiner, 2018-11-09:
I am afraid you are going to have to understand the sketches to understand
how to manage the offset calibrations...

gojunnyo, 2018-11-09:
where do I start?

kriswiner, 2018-11-09:
Line one.

Maybe use this
<https://github.com/kriswiner/ESP32/blob/master/MPU9250_MS5637/MPU9250_MS5637_AHRS.ino>
as a template for your own sketch.

gojunnyo, 2018-11-09:
Thank you sir!

I will post my adaptation to the code, (maybe tomorrow, I'm from the Philippines by the way and I am really sleepy, I am just so hopeless that I got myself involved in this forum). Hopefully, I can get things right!
Thank you for your guidance.

kriswiner, 2018-11-10:
> Hi! it's me again.
>
> I found a sketch which would output the offsets of the accelerometer and
> gyroscope of my sensor.
> Will this suffice?
>
> /*
> offset cancellation example
> This example automatically finds offset values for each axis of the
> accelerometer and gyroscope.
>
> NOTE: Please don't move the sensor while offset is adjusted.
>
> NOTE2: All offset settings are stored in volatile memory so they
> have to be set every time the module is reset.
> */
>
> #include <MPU9255.h>//include MPU9255 library
>
> MPU9255 mpu;
>
> void print_data()//read and print raw data from the sensors
> {
>   for(int i=0;i<=10;i++)
>   {
> [... 281 lines cut ...]
>
> I am planning to modify this to get the average of 1000 samples because it
> is quick. It took less than 20 seconds to spit out offsets. Do you think it
> is worth the effort? or should I stick with the original code?
> Your help will be much appreciated!

Don't know. Why not use the gyro and accel calibration in the sketches in
the repository I linked to?

gojunnyo, 2018-11-10:
Sorry for my rookie mistakes.

I got your MPU9250_AHRS modified, and I got these data

Is this what I should expect?
Is it supposed to oscillate?

[image]

kriswiner, 2018-11-10:
I don't think so,  what are you plotting?

kriswiner, 2018-11-10:
> I got it working now! What I did is I excluded most Serial.print and
> modified to print only YPR values.
>
> Any tips to make this fast? I'm planning to implement this to our PID
> controller (Part of our "crude" VTOL tailsitter flight controller)
>
> Your help again sir is highly appreciated!
>
> If ever you are willing to waste your time with us, contact me through
> this email nino.verame@gmail.com

For fast fusion use an MCU with a floating point unit like this
<https://www.tindie.com/products/TleraCorp/dragonfly-stm32l47696-development-board/>
one. Or this
<https://www.tindie.com/products/TleraCorp/ladybug-stm32l432-development-board/>
one.

gojunnyo, 2018-11-11:
With these boards sir? Can I connect the same configuration as the pro mini Arduino board?
or there is slight changes?

gojunnyo, 2018-11-11:
With these boards sir? Can I connect the same configuration as the pro mini Arduino board?
or there are slight changes?

kriswiner, 2018-11-11:
What do you mean by same configuration??

gojunnyo, 2018-11-11:
How accurate is the selftest function in the sketch? or it depends on the orientation of the sensor during calibration?

gojunnyo, 2018-11-11:
More like, if I have connected the sensor to pin 5 (pro mini), the same configuration applies to the boards you have

gojunnyo, 2018-11-11:
Sorry by the way if I am so annoying on this thread

kriswiner, 2018-11-11:
> Need your help again sir!
>
> I based my code to yours and what this is what happened.
> Yaw is fairly consistent.
> Pitch and roll are not.
> I am confused why this happens.
>
> *TRIAL 1: (Set of 5 samples)*
> *Each trial is determined after resetting the pro mini.*
>
> Yaw, Pitch, Roll: 41.77, 1.34, -1.17
> Yaw, Pitch, Roll: 40.89, 1.47, -1.34
> Yaw, Pitch, Roll: 40.98, 1.44, -1.30
> Yaw, Pitch, Roll: 41.31, 1.47, -1.30
> Yaw, Pitch, Roll: 40.19, 1.49, -1.35
>
> *TRIAL 2*
> Yaw, Pitch, Roll: 41.92, 6.78, -1.23
> Yaw, Pitch, Roll: 42.35, 6.66, -1.12
> Yaw, Pitch, Roll: 41.27, 6.82, -1.24
> Yaw, Pitch, Roll: 42.39, 6.73, -1.18
> Yaw, Pitch, Roll: 42.07, 6.77, -1.23
>
> *TRIAL 3*
> Yaw, Pitch, Roll: 41.12, 1.36, -6.65
> Yaw, Pitch, Roll: 41.76, 1.37, -6.58
> Yaw, Pitch, Roll: 41.63, 1.36, -6.59
> Yaw, Pitch, Roll: 41.35, 1.42, -6.64
> Yaw, Pitch, Roll: 40.72, 1.36, -6.68

No idea. Poor calibration, MCU too slow?

gojunnyo, 2018-11-11:
And by calibration you specifically mean?

kriswiner, 2018-11-11:
offset biases for the accel/gyro/mag at a minimum.

gojunnyo, 2018-11-11:
How will I do that?

kriswiner, 2018-11-11:
Are you just using the sketches in the repository as a magic black box or
are you reading the sketch to see what it does?

Calibration is included in the sketches if you have it properly
implemented. Since I don;t know what you are doing (but hope you do!) all I
can advise is to look in the sketches to see how calibration is managed.

gojunnyo, 2018-11-11:
Good point. Just wanted to clarify things, because I'm still stuck at why I acquired inconsistent pitch and roll. I seriously don't know what caused it. 
Recently though, putting the sensor at close to perfect horizontal orientation, I managed to acquire close to zero pitch and roll with a deviation of approximately 0.5 on first power up.
During my third power up, the values of pitch and roll jumped to +4 even on the same setup.

I might have not known this but I managed to acquire close to zero pitch and roll if after first power on!
If I reset the pro mini, the values will jump positive or negative 4. If I want to acquire the close to zero values, I just have to drain the power, and turn it back on!

Can you confirm sir? that even though I reset my MCU, the sensor still stores the value within its volatile memory? and that during reset, the sensor is unaffected?

kriswiner, 2018-11-11:
If you cut power to the sensor it will power on to its default state.

gojunnyo, 2018-11-11:
But during MCU reset, the sensor will not go back to its default state?

kriswiner, 2018-11-11:
No idea

gojunnyo, 2018-11-11:
Okay sir! Thank you for everything!
I'll just stick with the power cut as the reset mechanism of the system, through it the beauty and functionality of the sketch are properly demonstrated. Still wanted to find out what really happened.

Hoping to buy one of your boards but currently, I'm forced to optimize and utilize what the pro mini can do. As much as I really wanted to buy beefer MCU's as you suggested, but due to financial constraints, we cannot. Thus, I will do whatever it takes (even though I become annoying more than ever on this thread).

gojunnyo, 2018-11-12:
Lastly, any tips on how to make the sensor know absolute horizon?
In which, I do not have to do runtime calibration.

I might have missed it from the sketch, but it would be a great help if you'll give me a hint on how to achieve it.

kriswiner, 2018-11-12:
> Will this development board suffice?

That's a Cortex M0. You want a Cortex M4F. Ladybug is a better choice. Is
it that much more expensive?

gojunnyo, 2018-11-13:
Too bad in our country, yes, it is indeed expensive. The shipping costs around 15 dollars. And I'm having troubles in finding similar products which cost less.

gojunnyo, 2018-11-13:
How can I make the calibration permanent?

kriswiner, 2018-11-13:
Store it in your sketch.

gojunnyo, 2018-11-13:
Correct me if I comprehended your sketch wrong.

The code below is responsible in storing the offset to the registers.
And that data[0] is just another form of gyro_bias[0]
```
  writeByte(MPU9250_ADDRESS, XG_OFFSET_H, data[0]);
  writeByte(MPU9250_ADDRESS, XG_OFFSET_L, data[1]);
  writeByte(MPU9250_ADDRESS, YG_OFFSET_H, data[2]);
  writeByte(MPU9250_ADDRESS, YG_OFFSET_L, data[3]);
  writeByte(MPU9250_ADDRESS, ZG_OFFSET_H, data[4]);
  writeByte(MPU9250_ADDRESS, ZG_OFFSET_L, data[5]);
```
So in my sketch instead of

```
void accelgyrocalMPU9250(float * dest1, float * dest2)
{  
  uint8_t data[12]; // data array to hold accelerometer and gyro x, y, z, data
  uint16_t ii, packet_count, fifo_count;
  int32_t gyro_bias[3]  = {0, 0, 0}, accel_bias[3] = {0, 0, 0};
  
 // reset device
  writeByte(MPU9250_ADDRESS, PWR_MGMT_1, 0x80); // Write a one to bit 7 reset bit; toggle reset device
  delay(100);
```
I will change the int32_t gyro_bias[3] = {3 Values I got from the sketch that output raw gyro_bias without the dest1[0] = (float) gyro_bias[0]/(float) gyrosensitivity;  } the same goes to accelerometer.

kriswiner, 2018-11-13:
Yes, then don't run the calibration anymore. But the bias is temperature
dependent so can drift.

gojunnyo, 2018-11-13:
the temperature was added to both accel and gyro biases, correct?

so is it good, if my calibration would only be getting temperature data and doing the corrections to the hard-coded biases?

The downside is I get a delay, especially on aircraft recovery, correct?

kriswiner, 2018-11-13:
How was the temperature added to the biases?

kriswiner, 2018-11-13:
> I just broke the momentum hahaha
>
>     accel_bias[0] += (int32_t) accel_temp[0];
>     accel_bias[1] += (int32_t) accel_temp[1];
>     accel_bias[2] += (int32_t) accel_temp[2];
>     gyro_bias[0]  += (int32_t) gyro_temp[0];
>     gyro_bias[1]  += (int32_t) gyro_temp[1];
>     gyro_bias[2]  += (int32_t) gyro_temp[2];
>
>
> What does this mean exactly?

temp means temporary

gojunnyo, 2018-11-13:
*facepalm*

So is there any part of the sketch that tackles with the temperature compensation? or it was the commented part?

kriswiner, 2018-11-13:
No

gojunnyo, 2018-11-14:
Sorry to bother you again sir, 

I encounter this "unusual problem"

ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = 0 my = 0 mz = 0 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Gyro temperature is 29.6 degrees C
Yaw, Pitch, Roll: 13.80, 0.00, 0.00
Grav_x, Grav_y, Grav_z: 0.00, 0.00, 1000.00 mg
Lin_ax, Lin_ay, Lin_az: 0.00, 0.00, -1000.00 mg
sumCount = 0
sum = 0.00
rate = nan Hz

This is the only data that comes up, again and again even if I move the sensor.

kriswiner, 2018-11-14:
Are you using the interrupt?

kriswiner, 2018-11-14:
> Yes, why?

If it is not set up properly, meaning not getting the interrupt, this is
the output i would expect.

gojunnyo, 2018-11-16:
If I disable the interrupt, will it be worse?

gojunnyo, 2018-11-16:
I doubled checked my Interrupt connections, there is no problem with it. It passed the continuity test.
I doubt it if there is a problem with my adaptation to your code.

gojunnyo, 2018-11-16:
I got it working now sir, I commented this line
`attachInterrupt(intPin, myinthandler, RISING);  // define interrupt for INT pin output of MPU9250`

and replaced this line  ```
if(newData == true) {  // On interrupt, read data
     newData = false;
```

with ` if (readByte(MPU9250_ADDRESS, INT_STATUS) & 0x01)`

gojunnyo, 2018-11-16:
Any reason why the first line doesn't work?

gojunnyo, 2018-11-16:
Now I don't understand why my readings are erratic.

Using Mahony filter
```
Lin_ax, Lin_ay, Lin_az: 896.33, -141.85, -402.46 mg
sumCount = 250
sum = 0.51
rate = 486.54 Hz
Yaw, Pitch, Roll: 27.04, 3.39, 65.97
Grav_x, Grav_y, Grav_z: -911.71, 59.17, 406.55 mg
Lin_ax, Lin_ay, Lin_az: 912.50, -31.95, -394.71 mg
sumCount = 250
sum = 0.51
rate = 486.28 Hz
Yaw, Pitch, Roll: 11.19, 0.79, 72.02
Grav_x, Grav_y, Grav_z: -951.10, 13.78, 308.58 mg
Lin_ax, Lin_ay, Lin_az: 952.62, 14.73, -303.70 mg
sumCount = 250
sum = 0.52
rate = 483.35 Hz
Yaw, Pitch, Roll: 21.75, 1.13, 69.27
Grav_x, Grav_y, Grav_z: -935.10, 19.74, 353.84 mg
Lin_ax, Lin_ay, Lin_az: 936.56, 9.19, -343.46 mg
sumCount = 250
sum = 0.52
rate = 485.00 Hz
Yaw, Pitch, Roll: 29.96, 2.01, 66.68
Grav_x, Grav_y, Grav_z: -917.77, 35.00, 395.58 mg
Lin_ax, Lin_ay, Lin_az: 915.75, -5.95, -385.32 mg
sumCount = 250
sum = 0.51
rate = 485.61 Hz
Yaw, Pitch, Roll: 2.53, -1.25, 65.65
Grav_x, Grav_y, Grav_z: -910.85, -21.81, 412.15 mg
Lin_ax, Lin_ay, Lin_az: 914.03, 52.58, -403.36 mg
sumCount = 250
sum = 0.52
rate = 484.86 Hz
Yaw, Pitch, Roll: 35.42, 6.64, 65.68
Grav_x, Grav_y, Grav_z: -905.18, 115.58, 409.00 mg
Lin_ax, Lin_ay, Lin_az: 907.63, -89.27, -396.98 mg
sumCount = 250
sum = 0.52
rate = 483.66 Hz
Yaw, Pitch, Roll: 46.31, -3.02, 71.85
Grav_x, Grav_y, Grav_z: -948.94, -52.77, 311.01 mg
Lin_ax, Lin_ay, Lin_az: 950.10, 80.66, -300.58 mg
sumCount = 250
sum = 0.52
rate = 483.48 Hz
Yaw, Pitch, Roll: 35.39, 1.75, 70.41
Grav_x, Grav_y, Grav_z: -941.66, 30.55, 335.16 mg
Lin_ax, Lin_ay, Lin_az: 945.45, -3.02, -324.73 mg
sumCount = 250
sum = 0.52
rate = 485.15 Hz
Yaw, Pitch, Roll: 26.92, -5.28, 74.27
Grav_x, Grav_y, Grav_z: -958.48, -92.11, 269.89 mg
Lin_ax, Lin_ay, Lin_az: 958.72, 120.31, -258.23 mg
sumCount = 250
sum = 0.52
rate = 484.02 Hz
Yaw, Pitch, Roll: 4.73, 6.85, 64.18
Grav_x, Grav_y, Grav_z: -893.74, 119.27, 432.43 mg
Lin_ax, Lin_ay, Lin_az: 894.11, -90.03, -423.40 mg
sumCount = 250
sum = 0.52
rate = 484.98 Hz
```
Using Madgwick Filter
```
Yaw, Pitch, Roll: 173.36, 5.58, 70.52
Grav_x, Grav_y, Grav_z: -938.29, 97.17, 331.93 mg
Lin_ax, Lin_ay, Lin_az: 940.79, -65.43, -322.65 mg
sumCount = 170
sum = 0.52
rate = 327.88 Hz
Yaw, Pitch, Roll: 173.63, 5.89, 70.03
Grav_x, Grav_y, Grav_z: -934.90, 102.70, 339.72 mg
Lin_ax, Lin_ay, Lin_az: 938.81, -72.49, -328.13 mg
sumCount = 170
sum = 0.52
rate = 327.59 Hz
Yaw, Pitch, Roll: 173.25, 6.10, 69.71
Grav_x, Grav_y, Grav_z: -932.63, 106.28, 344.83 mg
Lin_ax, Lin_ay, Lin_az: 935.50, -75.09, -336.22 mg
sumCount = 170
sum = 0.52
rate = 329.20 Hz
Yaw, Pitch, Roll: 172.72, 6.30, 69.41
Grav_x, Grav_y, Grav_z: -930.46, 109.82, 349.56 mg
Lin_ax, Lin_ay, Lin_az: 934.36, -79.36, -339.55 mg
sumCount = 170
sum = 0.52
rate = 329.26 Hz
Yaw, Pitch, Roll: 172.75, 5.91, 69.35
Grav_x, Grav_y, Grav_z: -930.77, 102.99, 350.81 mg
Lin_ax, Lin_ay, Lin_az: 932.11, -72.65, -342.63 mg
sumCount = 170
sum = 0.52
rate = 327.44 Hz
Yaw, Pitch, Roll: 172.30, 5.41, 69.92
Grav_x, Grav_y, Grav_z: -935.03, 94.22, 341.82 mg
Lin_ax, Lin_ay, Lin_az: 937.72, -65.77, -327.72 mg
sumCount = 170
sum = 0.52
rate = 326.42 Hz
```

gojunnyo, 2018-11-16:
This is my adaptation to your code.

```
#include "Wire.h"   

//Magnetometer Registers
#define AK8963_ADDRESS   0x0C
#define WHO_AM_I_AK8963  0x00 // should return 0x48
#define INFO             0x01
#define AK8963_ST1       0x02  // data ready status bit 0
#define AK8963_XOUT_L     0x03  // data
#define AK8963_XOUT_H    0x04
#define AK8963_YOUT_L    0x05
#define AK8963_YOUT_H    0x06
#define AK8963_ZOUT_L    0x07
#define AK8963_ZOUT_H    0x08
#define AK8963_ST2       0x09  // Data overflow bit 3 and data read error status bit 2
#define AK8963_CNTL      0x0A  // Power down (0000), single-measurement (0001), self-test (1000) and Fuse ROM (1111) modes on bits 3:0
#define AK8963_ASTC      0x0C  // Self test control
#define AK8963_I2CDIS    0x0F  // I2C disable
#define AK8963_ASAX      0x10  // Fuse ROM x-axis sensitivity adjustment value
#define AK8963_ASAY      0x11  // Fuse ROM y-axis sensitivity adjustment value
#define AK8963_ASAZ      0x12  // Fuse ROM z-axis sensitivity adjustment value

#define SELF_TEST_X_GYRO 0x00                  
#define SELF_TEST_Y_GYRO 0x01                                                                          
#define SELF_TEST_Z_GYRO 0x02

/*#define X_FINE_GAIN      0x03 // [7:0] fine gain
#define Y_FINE_GAIN      0x04
#define Z_FINE_GAIN      0x05
#define XA_OFFSET_H      0x06 // User-defined trim values for accelerometer
#define XA_OFFSET_L_TC   0x07
#define YA_OFFSET_H      0x08
#define YA_OFFSET_L_TC   0x09
#define ZA_OFFSET_H      0x0A
#define ZA_OFFSET_L_TC   0x0B */

#define SELF_TEST_X_ACCEL 0x0D
#define SELF_TEST_Y_ACCEL 0x0E    
#define SELF_TEST_Z_ACCEL 0x0F

#define SELF_TEST_A      0x10

#define XG_OFFSET_H      0x13  // User-defined trim values for gyroscope
#define XG_OFFSET_L      0x14
#define YG_OFFSET_H      0x15
#define YG_OFFSET_L      0x16
#define ZG_OFFSET_H      0x17
#define ZG_OFFSET_L      0x18
#define SMPLRT_DIV       0x19
#define CONFIG           0x1A
#define GYRO_CONFIG      0x1B
#define ACCEL_CONFIG     0x1C
#define ACCEL_CONFIG2    0x1D
#define LP_ACCEL_ODR     0x1E   
#define WOM_THR          0x1F   

#define MOT_DUR          0x20  // Duration counter threshold for motion interrupt generation, 1 kHz rate, LSB = 1 ms
#define ZMOT_THR         0x21  // Zero-motion detection threshold bits [7:0]
#define ZRMOT_DUR        0x22  // Duration counter threshold for zero motion interrupt generation, 16 Hz rate, LSB = 64 ms

#define FIFO_EN          0x23
#define I2C_MST_CTRL     0x24   
#define I2C_SLV0_ADDR    0x25
#define I2C_SLV0_REG     0x26
#define I2C_SLV0_CTRL    0x27
#define I2C_SLV1_ADDR    0x28
#define I2C_SLV1_REG     0x29
#define I2C_SLV1_CTRL    0x2A
[... 1142 lines cut ...]
            // Apply feedback terms
            gx = gx + Kp * ex + Ki * eInt[0];
            gy = gy + Kp * ey + Ki * eInt[1];
            gz = gz + Kp * ez + Ki * eInt[2];

            // Integrate rate of change of quaternion
            pa = q2;
            pb = q3;
            pc = q4;
            q1 = q1 + (-q2 * gx - q3 * gy - q4 * gz) * (0.5f * deltat);
            q2 = pa + (q1 * gx + pb * gz - pc * gy) * (0.5f * deltat);
            q3 = pb + (q1 * gy - pa * gz + pc * gx) * (0.5f * deltat);
            q4 = pc + (q1 * gz + pa * gy - pb * gx) * (0.5f * deltat);

            // Normalise quaternion
            norm = sqrtf(q1 * q1 + q2 * q2 + q3 * q3 + q4 * q4);
            norm = 1.0f / norm;
            q[0] = q1 * norm;
            q[1] = q2 * norm;
            q[2] = q3 * norm;
            q[3] = q4 * norm;
 
        }
  */
```

gojunnyo, 2018-11-16:
I used the above code, I let the sketch to output the Gyro and Accel Bias, and recorded the values.
I put the recorded values into the
`int32_t gyro_bias[3]  = {0, 0, 0}, accel_bias[3] = {0, 0, 0};`
under the function accelgyrocalMPU9250();

Anddd I expect and completely accept that whatever I am doing, It might not or might be very very STUPID. And forgive me for that.

kriswiner, 2018-11-16:
What about magnetometer calibration?

gojunnyo, 2018-11-16:
I always do the figure eight wave every run of the program.

kriswiner, 2018-11-16:
Not sure what to tell you except you might need a faster MCU. Can you check
your calibration to make sure it is reasonable? if you compare Mz when the
device is pointed up versus when it is pointed down, are the values equal
and opposite?

gojunnyo, 2018-11-16:
Do you mean Hz? or Mz is just another thing I am not familiar with?

kriswiner, 2018-11-16:
z-component of the magnetic field

gojunnyo, 2018-11-16:
The values are so rough, Although I can say that they are opposite, but they are hardly the same.
I have checked my location and updated the magnetic declination.

kriswiner, 2018-11-16:
Declination shouldn't matter.

The jitter in the mag field should be < 10%, maybe closer to a few %, not
rough.

gojunnyo, 2018-11-16:
Hmmm so its good then, all points to MCU inefficiency problem

kriswiner, 2018-11-16:
> Any suggestions sir that might help me in keeping this project? I still
> have 1 month left before our Oral Defense, enough time I suppose, to shift
> my study.

Get a better MCU.

[1 more reply]

closed 2018-11-29 by gojunnyo