kriswiner/MPU9250 #325: Confusing Timing Results

mjs513, 2018-11-13:
Hi Kris.
I was running some timing tests using your Basic_AHRS sketch, but just using wire and comparing it the results on a imrtx1050.  Using the sketch on the Arduino as is (read the interrupt status pin) I got the following results:
```
setclock(400000):
T3.5           6,365.16 Hz 
1050           8,117.8 Hz 
setclock(1000000):
T3.5           14,388.33 Hz
1050           22,005.26 Hz
```
In a post(https://forum.pjrc.com/threads/40343-MPU9250-Add-On-for-Teensy-3-6)  that manitou pointed me too, over on the prjc forum it looked like you were getting better performance using a hard interrupt (using the interrupt pin for data ready).  I just tried to that with your sketch by modifying the attachinterrupt code to read:
```
   //writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x22); 
   writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x12);
   writeByte(MPU9250_ADDRESS, INT_ENABLE, 0x01);  // Enable data ready (bit 0) interrupt
```
I think those are the right settings, just not sure why you are doing that using register 0x37 and not 0x58, but that is another question.  Anyway back to the question when I did that and set up the interrupt I got some strange timing results:
```
400K = 1695.43 Hz
1000K = rate = 3333.60 Hz
```
Dramatic drop in performance.  I must be doing something wrong but just not sure what.  On the mbed side for the imrtx1050 have to figure out how their interrupts work :) but that's next.

Any thoughts on what I messed up?

kriswiner, 2018-11-13:
Not sure. I would need to see your code.

0x58 is in the middle of the external sensor data array, not sure what this
has to do with MPU9250 interrupt.

Normally I see a speed up with the interrupt versus polling. Except when
polling you might be running the fusion filter whether you get new data or
not. What I usually do is something like this:

if (new data)

read data

for some number of iterations
run the Madgwick filter
endfor

endif

get fusion rate

Usually 10 or 20 iterations is enough. On the STM32L4 I usually get fusion
rate of 2 kHz or 4 kHz with 10 iterations.

So it depends on the structure of your loop.

Interestingly, running with the ESP32 probably about how you are doing it
(but with a hard interrupt) I am seeing 20 kHz, way to high of course, but
interesting in comparison.

mjs513, 2018-11-13:
Thanks for getting back to me - here is the sketch that I am running.  Basically your sketch with modification for newData unless I messed something up somewhere with the interrupt:  Oh, think i forgot, the strange timing results are on the T3.5 running wire.  Just wanted to be consisted when I did the comparisons.
```
/* MPU9250 Basic Example Code
 by: Kris Winer
 date: April 1, 2014
 license: Beerware - Use this code however you'd like. If you 
 find it useful you can buy me a beer some time.
 
 Demonstrate basic MPU-9250 functionality including parameterizing the register addresses, initializing the sensor, 
 getting properly scaled accelerometer, gyroscope, and magnetometer data out. Added display functions to 
 allow display to on breadboard monitor. Addition of 9 DoF sensor fusion using open source Madgwick and 
 Mahony filter algorithms. Sketch runs on the 3.3 V 8 MHz Pro Mini and the Teensy 3.1.
 
 SDA and SCL should have external pull-up resistors (to 3.3V).
 10k resistors are on the EMSENSR-9250 breakout board.
 
 Hardware setup:
 MPU9250 Breakout --------- Arduino
 VDD ---------------------- 3.3V
 VDDI --------------------- 3.3V
 SDA ----------------------- A4
 SCL ----------------------- A5
 GND ---------------------- GND
 
 Note: The MPU9250 is an I2C sensor and uses the Arduino Wire library. 
 Because the sensor is not 5V tolerant, we are using a 3.3 V 8 MHz Pro Mini or a 3.3 V Teensy 3.1.
 We have disabled the internal pull-ups used by the Wire library in the Wire.h/twi.c utility file.
 We are also using the 400 kHz fast I2C mode by setting the TWI_FREQ  to 400000L /twi.h utility file.
 */
#include "Wire.h"   
//#include <i2c_t3.h>

// See also MPU-9250 Register Map and Descriptions, Revision 4.0, RM-MPU-9250A-00, Rev. 1.4, 9/9/2013 for registers not listed in 
// above document; the MPU9250 and MPU9150 are virtually identical but the latter has a different register map
//
//Magnetometer Registers
#define AK8963_ADDRESS   0x0C
#define AK8963_WHO_AM_I  0x00 // should return 0x48
#define AK8963_INFO      0x01
#define AK8963_ST1       0x02  // data ready status bit 0
#define AK8963_XOUT_L   0x03  // data
#define AK8963_XOUT_H  0x04
#define AK8963_YOUT_L  0x05
#define AK8963_YOUT_H  0x06
#define AK8963_ZOUT_L  0x07
#define AK8963_ZOUT_H  0x08
#define AK8963_ST2       0x09  // Data overflow bit 3 and data read error status bit 2
#define AK8963_CNTL      0x0A  // Power down (0000), single-measurement (0001), self-test (1000) and Fuse ROM (1111) modes on bits 3:0
#define AK8963_ASTC      0x0C  // Self test control
#define AK8963_I2CDIS    0x0F  // I2C disable
#define AK8963_ASAX      0x10  // Fuse ROM x-axis sensitivity adjustment value
#define AK8963_ASAY      0x11  // Fuse ROM y-axis sensitivity adjustment value
#define AK8963_ASAZ      0x12  // Fuse ROM z-axis sensitivity adjustment value

#define SELF_TEST_X_GYRO 0x00                  
#define SELF_TEST_Y_GYRO 0x01                                                                          
#define SELF_TEST_Z_GYRO 0x02

/*#define X_FINE_GAIN      0x03 // [7:0] fine gain
#define Y_FINE_GAIN      0x04
#define Z_FINE_GAIN      0x05
#define XA_OFFSET_H      0x06 // User-defined trim values for accelerometer
#define XA_OFFSET_L_TC   0x07
#define YA_OFFSET_H      0x08
#define YA_OFFSET_L_TC   0x09
#define ZA_OFFSET_H      0x0A
#define ZA_OFFSET_L_TC   0x0B */

#define SELF_TEST_X_ACCEL 0x0D
#define SELF_TEST_Y_ACCEL 0x0E    
[... 847 lines cut ...]
{
  uint8_t data; // `data` will store the register data   
  Wire.beginTransmission(address);         // Initialize the Tx buffer
  Wire.write(subAddress);                  // Put slave register address in Tx buffer
  Wire.endTransmission();        // Send the Tx buffer, but send a restart to keep connection alive
//  Wire.endTransmission(false);             // Send the Tx buffer, but send a restart to keep connection alive
//  Wire.requestFrom(address, 1);  // Read one byte from slave register address 
  Wire.requestFrom(address, (size_t) 1);   // Read one byte from slave register address 
  data = Wire.read();                      // Fill Rx buffer with result
  return data;                             // Return data read from slave register
}

        void readBytes(uint8_t address, uint8_t subAddress, uint8_t count, uint8_t * dest)
{  
  Wire.beginTransmission(address);   // Initialize the Tx buffer
  Wire.write(subAddress);            // Put slave register address in Tx buffer
  Wire.endTransmission();  // Send the Tx buffer, but send a restart to keep connection alive
//  Wire.endTransmission(false);       // Send the Tx buffer, but send a restart to keep connection alive
  uint8_t i = 0;
//        Wire.requestFrom(address, count);  // Read bytes from slave register address 
        Wire.requestFrom(address, (size_t) count);  // Read bytes from slave register address 
  while (Wire.available()) {
        dest[i++] = Wire.read(); }         // Put read results in the Rx buffer
}
```

kriswiner, 2018-11-13:
There are a lot of problems with this sketch. Not sure why you chose this
one to play with.

First of all, the accel and gyro data should be read in one operation, the
resolutions (gerAres(),eg) should be done in setup just once, the mag
really ought to be run at 100 Hz, the mag biases are hard coded (obviously
wrong for your sensor), the sensor data order into the Madgwick filter is
wrong, etc. Not sure if these have anything to do with your timing results.

You might try changing PI to in-sketch defined pi = 3.14159f. This speeded
up the fuson filter by 10x when I did this for the STM32L4.

In short, this is not a good vehicle for exploring timing issues. At a
minimum I would limit the Madgwick fusion in a for loop to 10 iterations,
correct some of the obvious sketch errors and try the timing test again.

kriswiner, 2018-11-13:
By the way, I finally got around to writing a sketch for two MPU9250s on
the same bus. I get 2 kHz fusion rate for both when running at 400 kHz I2C
bus speed, 10 iterations of the filter using an STM32L4. If I up the I2C
speed to 1 MHz the rate doubles, of course.

mjs513, 2018-11-13:
Hi Kris - just used your basic version from GitHub.  No particular reason why I picked that one.  Which version should I use?

For the mag want to update the final version to use to check the mag data ready register instead of doing the way I am doing it now.?
In the mean time I will go poking around.  I went to eye doctor this morning and my eyes are still all out of sort so this should be fun :)
Will keep you posted

kriswiner, 2018-11-13:
Maybe start here:
https://github.com/kriswiner/MPU9250/tree/master/AK8963_as_slave

mjs513, 2018-11-13:
Hi Kris,
Will definitely take a look.  Any I did mod your esp32 sketch to run on the T3.5 since it had the interrupt configured as a data ready interrupt as well as the code for the mag data ready test.  I also had the 10x loop for the filter set up properly.  It also did the cal's for the accel, gyro and magnetometer and the reads of the accel/gyro at one time :)  So basically it did what you suggested in the earlier post.  

Any way, whether I used a clock freq of 400k or 1000khz I got a rate of about 2khz for the T5 at 120Mhz overclock.  Does this sound about right?

kriswiner, 2018-11-13:
No, it really should be higher unless you are liiting the iterations of the
fusion filter to ten or so.

mjs513, 2018-11-13:
Right now its set to 10, matches what you have in the esp32 sketch.  Got it ported over to the imrtx1050 and its giving me 1951hz.  a little lower than for the T3.5.

kriswiner, 2018-11-13:
This is what I would expect. At 1 MHz bus speed should jump to 4 kHz.