kriswiner/MPU9250 #326: updated y/p/r all zeros

tsaun, 2018-11-15:
Hey Kris,

I'm working with a Teensy 3.5 and an MPU-9250/6500, I uploaded what I've read to be your most updated code from your github (MPU9250_MS5637_AHRS_t3) - everything seems to initialize fine, but then the actual IMU data is all reported as zeros (see attached). When I try another sketch, MPU9250BasicAHRS_t3, the values seem to be reported appropriately. On combing through the code side-by-side, I see there are differences in how the intPin is flagging new data, as well as differences in  how the MPU9250Data is being stored (all together vs. as separate variables). I'm not sure how to reconcile the differences and get the MPU9250_MS5637_AHRS_t3 code working. Any sort of advice/explanation would be appreciated. Cheers.

```
Scanning...
I2C device found at address 0x0C  !
I2C device found at address 0x68  !
done

MPU9250 9-axis motion sensor...
MPU9250 I AM 71 I should be 71
MPU9250 is online...
x-axis self test: acceleration trim within : 1.3% of factory value
y-axis self test: acceleration trim within : -0.8% of factory value
z-axis self test: acceleration trim within : 0.1% of factory value
x-axis self test: gyration trim within : -0.7% of factory value
y-axis self test: gyration trim within : 0.7% of factory value
z-axis self test: gyration trim within : 0.3% of factory value
 Calibrate gyro and accel
accel biases (mg)
-0.49
26.37
-38.76
gyro biases (dps)
0.34
1.31
-1.48
MPU9250 initialized for active data mode....
AK8963 I AM 48 I should be 48
AK8963 initialized for active data mode....
Mag Calibration: Wave device in a figure eight until done!
Mag Calibration done!
AK8963 mag biases (mG)
-676.08
218.99
45.38
AK8963 mag scale (mG)
1.25
0.84
1.00
X-Axis sensitivity adjustment value 1.20
Y-Axis sensitivity adjustment value 1.21
Z-Axis sensitivity adjustment value 1.16
MS5637 pressure sensor reset...
PROM dta read:
C0 = 0
C1 = 0
C2 = 0
C3 = 0
C4 = 0
C5 = 0
C6 = 0
Checksum = 0 , should be 0
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = 0 my = 0 mz = 0 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Gyro temperature is 30.2 degrees C
Digital temperature value = 20.00 C
Digital temperature value = 68.00 F
Digital pressure value = 0.00 mbar
Altitude = 145366.45 feet
Yaw, Pitch, Roll: 13.80, 0.00, 0.00
Grav_x, Grav_y, Grav_z: 0.00, 0.00, 1000.00 mg
Lin_ax, Lin_ay, Lin_az: 0.00, 0.00, -1000.00 mg
rate = 0.03 Hz
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = 0 my = 0 mz = 0 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Gyro temperature is 30.2 degrees C
Digital temperature value = 20.00 C
Digital temperature value = 68.00 F
Digital pressure value = 0.00 mbar
Altitude = 145366.45 feet
Yaw, Pitch, Roll: 13.80, 0.00, 0.00
Grav_x, Grav_y, Grav_z: 0.00, 0.00, 1000.00 mg
Lin_ax, Lin_ay, Lin_az: 0.00, 0.00, -1000.00 mg
rate = 148811.73 Hz
```

kriswiner, 2018-11-15:
You are going have to set up the interrupt properly if you are using a
sketch that requires it, or get rid of tehe interrupt and use polling like
the sketch that works. if you don;t know what this means, there are Arduino
tutorials that explainhow interrupts work.

tsaun, 2018-11-15:
Got it - thank you.
A quick review on interrupts, and of course making the correct connection between the sensor and the designated intPin made all the difference :)

closed 2018-11-15 by tsaun