kriswiner/MPU9250 #327: Initiating DMP for MPU9250

dhr999, 2018-11-21:
Hey, I'm using the MPU9250 for localization using Rpi as a single board computer. I went through the registry map of the IMU but couldn't figure out a way to initialize the Dmp. Could you please tell me how to do that and also how to interpret the output from it?

Thanks a lot.

Regards

kriswiner, 2018-11-21:
Nope, contact Invensense for this information.

dhr999, 2018-11-21:
`wia = 0x00
hxl = 0x03
hxh = 0x04
hyl = 0x05
hyh = 0x06
hzl = 0x07
hzh = 0x08
ST2 = 0x09
CNTL1 = 0x0A
ASAX = 0x10
ASAY = 0x11
ASAZ = 0x12
int_config=0x37

dev_add = 0x68
bus.write_byte_data(dev_add,int_config,0x02)

def Mag_init():
	bus.write_byte_data(mag_add,CNTL1,31)
	asax = bus.read_byte_data(mag_add,ASAX)
	asay = bus.read_byte_data(mag_add,ASAY)
	asaz = bus.read_byte_data(mag_add,ASAZ)
	bus.write_byte_data(mag_add,CNTL1,6)

mag_add = 0x0C
Mag_init()

def magnetometer(addr):
	low = bus.read_byte_data(dev_add,addr)
	high = bus.read_byte_data(dev_add,addr+1)

	value = ((high <<8) | low)

	if (value & (1 << 13)) != 0:
		value = value - (1 <<14)
	return value

`

Could you please take a look at the code and tell me if I missed anything or if I am something wrong?

Thank you

Regards`

kriswiner, 2018-11-21:
No, please contact Invensense for this help.