kriswiner/MPU9250 #328: mpu9250 output

amir96, 2018-11-22:
Hi dear kris, I use of arduino pro mini 3.3 and mpu9250. For my university project I need to yaw, pitch and roll. I know pro mini isn’t suitable for this work, but I due to use this. Now if I do a good calibration, is it possible to have  rather good output?
It’s important for me that pitch doesn’t affect on yaw or roll. deflection about 5 degree isn’t important for me.

kriswiner, 2018-11-22:
I would start with this
<https://github.com/kriswiner/ESP32/blob/master/MPU9250_MS5637/MPU9250_MS5637_AHRS.ino>
sketch, you will have to add the quaternionFilters.ino sketch using the
tab. You will also have to change a few of the wire calls for the Arduino
Pro Mini, byt this should give you what you need. The Pro Mini is really
too slow to do this job properly, so you will see a lot of latency and not
very accurate orientation estimation.

amir96, 2018-11-22:
Thank you kris. And how can I plot calibration data?
What should I plot? By arduino plotter?!
Do you have sketch for plotting?

kriswiner, 2018-11-23:
> Hi, I think this part of your code has a problem:
> ` // If intPin goes high, all data registers have new data
> if(newData == true) { // On interrupt, read data
> newData = false; // reset newData flag
> readMPU9250Data(MPU9250Data); // INT cleared on any read
>
> // Now we'll calculate the accleration value into actual g's
>  ax = (float)MPU9250Data[0]*aRes - accelBias[0];  // get actual g value, this depends on scale being set
>  ay = (float)MPU9250Data[1]*aRes - accelBias[1];
>  az = (float)MPU9250Data[2]*aRes - accelBias[2];
>
> // Calculate the gyro value into actual degrees per second
>  gx = (float)MPU9250Data[4]*gRes;  // get actual gyro value, this depends on scale being set
>  gy = (float)MPU9250Data[5]*gRes;
>  gz = (float)MPU9250Data[6]*gRes;
>
> newMagData = (readByte(AK8963_ADDRESS, AK8963_ST1) & 0x01);
> if(newMagData == true) { // wait for magnetometer data ready bit to be set
>   readMagData(magCount);  // Read the x/y/z adc values
>
> // Calculate the magnetometer values in milliGauss
> // Include factory calibration per data sheet and user environmental corrections
>   mx = (float)magCount[0]*mRes*magCalibration[0] - magBias[0];  // get actual magnetometer value, this depends on scale being set
>   my = (float)magCount[1]*mRes*magCalibration[1] - magBias[1];
>   mz = (float)magCount[2]*mRes*magCalibration[2] - magBias[2];
> [... 32 lines cut ...]
>
> }`
> what's the problem of first code? how can I fix it? I used old code.
> please help me.
> tnx

The problem is you are not setting up tour interrupt properly. Which pin
are you using for interrupt?

amir96, 2018-11-24:
I didn't change default settings.

int intPin = 12;  // can be any pin
#define DMP_INT_STATUS   0x39  // Check DMP interrupt
#define INT_STATUS       0x3A

is that what you pointed or not?

amir96, 2018-11-24:
[image]
this is my calibration plot. I repeat calibration process, but no change in result. why?
I've done it by rotating in figure 8/

kriswiner, 2018-11-24:
So are you using pin 12 as the interrupt or not?

amir96, 2018-11-24:
yes, I don't change it.

kriswiner, 2018-11-24:
Show mw a picture of your setup with the interrupt connected.

amir96, 2018-11-24:
// Set up the interrupt pin, its set as active high, push-pull
  pinMode(intPin, INPUT);
  digitalWrite(intPin, LOW);
  pinMode(myLed, OUTPUT);
  digitalWrite(myLed, LOW);

kriswiner, 2018-11-24:
Delete this line:  digitalWrite(intPin, LOW);

This is not a picture, I want to see the wires.

amir96, 2018-11-24:
I didn't connect pin 12 to int pin.I though just SDA and SCL is needed.
is it possible that all of my problem caused from this part?

kriswiner, 2018-11-24:
Electronics is not magic, if you don't connect something how can it
possible work?!

amir96, 2018-11-24:
[image]
what about calibration?
this is my calibration plot. I repeat calibration process, but no change in result. why?
I've done it by rotating in figure 8.