kriswiner/MPU9250 #330: Compile the STM code in Keil uVision5

pmfmouta, 2018-12-05:
Hi! I'm trying to implement this code into an STM32f303k8 in Keil uVision 5, but it gives an error in mbed.h file. I'm tried to export the library in the mbed compiler to run in Keil uvision, but it's not working.  How you do that?

Thanks

kriswiner, 2018-12-05:
No idea, never used Keil uVision or the F303.

pmfmouta, 2018-12-06:
> No idea, never used Keil uVision or the F303.
> […](#)

In the main.cpp file, regarding the integration time, in what units is the deltat?

Now = t.read_us();
deltat = (float)((Now - lastUpdate)/1000000.0f) ;
lastUpdate = Now;
sum += deltat;
sumCount++;

kriswiner, 2018-12-06:
> No idea, never used Keil uVision or the F303.
> … <#m_-8345973089201099897_>

microseconds

kriswiner, 2018-12-06:
Sorry, micros/1000000. so seconds

rizqiuhuy, 2019-01-12:
i try compile on keil uVision 5 success with no problem the code all thanks kriswiner

pmfmouta, 2019-01-15:
> i try compile on keil uVision 5 success with no problem the code all thanks kriswiner

Could you share the Keil Project files?

kriswiner, 2019-01-15:
I have none...

rizqiuhuy, 2019-01-15:
pmfouta visit my github on i2c_mpu9250_fastmode project

pmfmouta, 2019-01-16:
> pmfouta visit my github on i2c_mpu9250_fastmode project

After I converted your project to run on STMM32F303k8Tx, it seems that the MCU gets stuck infinitely on the loop in `void I2C_Address_Search()`.
In this function, you try to determine the MPU address. But I already succeeded in getting values from the gyroscope and accelerometer and the address is:

`#define MPU9250_I2C_ADDR				0x68 << 1`
I tried to define the mpuAddress as the one above, but with no success. There is another way?

Thanks 👍

rizqiuhuy, 2019-01-16:
i think maybe you manual to define address or search the adress manually using HAL_I2C_isDeviceReady