kriswiner/MPU9250 #331: AK8963 is 0 should be 48 (and some funny values)

TByte007, 2018-12-09:
I have 2 MPU9250 one is working fine and has id 0x71. The other replies with 0x70 as ID and it's AK8963 acts strange - replies with 0 for the ID request "AK8963 is 0 should be 48" and then returns strange values (M: ... ):

A:  -0.0021  -0.0002   0.9961 | M:    -4774 -2147483648     8987 | T: 31.9C
A:  -0.0021  -0.0002   0.9961 | M:    -4774 -2147483648     8987 | T: 31.9C
A:  -0.0001   0.0003   0.9988 | M:    -4774        0     8987 | T: 31.9C
A:   0.0018  -0.0009   0.9994 | M:    -4774        0     8987 | T: 31.8C
A:   0.0005  -0.0002   0.9984 | M:    -4774 -2147483648     8987 | T: 31.8C
A:   0.0005  -0.0002   0.9984 | M:    -4774 -2147483648     8987 | T: 31.8C
A:   0.0000   0.0005   0.9987 | M:    -4774        0     8987 | T: 31.8C

I tried to wake it up (just in case) - didn't worked. I'm not really sure what is going on.

kriswiner, 2018-12-09:
MPU9250 has WHO_AM_I 0x71, MPU9255 has WHO_AM_I 0x73, I don;t know of any
kind of MPU925X that has an ID of 0x70, souncds like your breakout board is
poorly designed or broken. Is it a cheap breakout from China?

TByte007, 2018-12-09:
I think I got scammed sorry for the fuss :( . They said it's MPU-9250 and it's MPU-6500 - took me some time  too see the chip markings but it says "MP65" and then I read the register docs of MPU-6500 and it says there that it replies with 0x70 for "Who Am I" request. I hope google indexes this so more people see it and dont waste their time :)

TByte007, 2018-12-09:
The good thing is that your driver works fine with MPU-6500 !

kriswiner, 2018-12-09:
Of course, an MPU9250 is an MPU6500 with an embedded AK8963C.

TByte007, 2018-12-10:
Now I tried another module which is for sure 9250 as it answers with 0x71 - the funny part is that it's AK8963C only replies 1/10 of the times and never passes the calibration routine. Generally when I power off the system (the MCU+MPU9250) and power it on again the reply is "AK8963 is 48 should be 48" yet it returns only 0s at calibration and then only 0s for every axis. After I reset the MCU it returns the above "AK8963 is 0 should be 48" and acts as MPU6500. Another MPU9250 works fine in the same configuration/firmware Do you think it's the board or the chip is faulty ? Or may be something else ?

kriswiner, 2018-12-10:
If the software works on another MPU9250 it is the board.

kriswiner, 2018-12-10:
Are these the CJU purple boards? These are all crap and should be thrown
away.

TByte007, 2018-12-10:
Both are blue. the boards signs are different but the boards elements look the same and similarly placed. And of course the 6500 part works flawlessly every time. Only the mag acts weird. That's why I think it might be a faulty AK8963C  chip in the package. But I am far far less experienced than you :)

kriswiner, 2018-12-10:
More likely poor circuit design.

TByte007, 2018-12-10:
The tracks , elements and the elements placement look absolutely the same like copy/paste. The only difference is 2-3 signs on the board.

kriswiner, 2018-12-10:
Is pin 1 connected to 3V3?

TByte007, 2018-12-10:
Yes. Everything is the same as the other board(same cable on the same test board even) , the only difference is the disappearing mag. I just unplug the working one and plug the other one in it's place.

kriswiner, 2018-12-10:
I'm asking if the pin 1 on the MPU9250 is connected to 3V3...

TByte007, 2018-12-10:
Yes I just checked. Same as pin 13.
It is exactly like this - 1 looks like going to the other side and then going back to the front to the track where pin 13 is connected (3V3) 
[image]

TByte007, 2018-12-10:
It does I can clearly see MP92 (which made me smile as my eyes are still working and I am in front of a screen like 2/3 of the day for the last 20 years ) :)

TByte007, 2018-12-10:
nCS  is not connected currently - should i pull it high with resistor or just connect it to 3V+ ?

TByte007, 2018-12-10:
There is something funny going on. Pulled up or not for some unknown reason the mag now always reports the right ID (0x48) before it was 1 out of 10 times. The only thing I changes is this :
```
    while(d != 0x48) {
      d = MPU9250.getAK8963CID();  // Read WHO_AM_I register for AK8963
      SPRN("AK8963 is "); SPRN(d, HEX); SPRN(" should be "); SPLN(0x48, HEX);
      delay(1000);
    }
```
But it passes only ONCE through this cycle which means it replies the right way the first time. 
Yet the calibration returns all zeroes for bias and scale. And then all values are 0.
Is there a way to check if MPU9250.initAK8963() actually did it's work ?

kriswiner, 2018-12-10:
No idea. If the code works on some boards and not others this is not a
software issue...

TByte007, 2018-12-10:
I think I'm done with that thing. I think the AK8963 chip is faulty - everything works until you try to set AK8963_CNTL to 0x16 (16 bit resolution, 100 Hz update).  If you read back the register it comes as 0 and not 0x16 (another driver does that check and fails at that point). I tried it like 1000 times and it never returned 0x16

closed 2018-12-10 by TByte007