kriswiner/MPU9250 #332: Mbias

psilvao, 2018-12-21:
Hello, thank you very much for the code, the only thing I need to know is what the MBIAS parameter means, could you explain a bit of what it is about and where can you get the values that you reference in your code?

Thanks in advance
Pablo

kriswiner, 2018-12-21:
https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration

psilvao, 2019-01-02:
Hello Kris, first of all Happy New Year !!

I have read carefully the url that you sent me, according to what you indicate you have to do the movement of 8 in the three axes, after about 30 repetitions, I have managed to obtain this bias, which I imagine is close to the presicion. I attached graphic

[image]

The issue is, how do I get to accuracy? Do I still try to achieve a sphere? Or will there be some way to adjust the values to get the coveted sphere?

My bias values are:

MagBiasX -43.295
MagBiasY 133,493
MagBiasZ -224.399

Thanks in advance 
Pablo

kriswiner, 2019-01-02:
This doesn't seem very spherical.

You can increase the time for mag calibration and try to sample more of the
response surface. The closer to spherical these plots are made after the
bias is corrected for the more accurate will be the absolute orientation
estimation.

One easy check is if you point the sensor facing up and then down, is Mz
the same value but opposite? If not, the mag is not calibrated.