kriswiner/MPU9250 #334: Unstable yaw pitch roll

laurentvm, 2019-01-04:
Hi Kris, 

I am now looking at yaw pitch roll and I have unstable values which never trully converges:

`MPU9250 9-axis motion sensor...
MPU9250 I AM 71 I should be 71
MPU9250 is online...
x-axis self test: acceleration trim within : 2.4% of factory value
y-axis self test: acceleration trim within : 0.3% of factory value
z-axis self test: acceleration trim within : 2.3% of factory value
x-axis self test: gyration trim within : -5.2% of factory value
y-axis self test: gyration trim within : -1.3% of factory value
z-axis self test: gyration trim within : -0.3% of factory value
 Calibrate gyro and accel
accel biases (mg)
56.34
-116.33
65.12
gyro biases (dps)
-1.65
0.60
-0.14
MPU9250 initialized for active data mode....
AK8963 I AM 48 I should be 48
AK8963 initialized for active data mode....
Mag Calibration: Wave device in a figure eight until done!
Mag Calibration done!
AK8963 mag biases (mG)
-468.51
308.03
-102.97
AK8963 mag scale (mG)
1.03
0.89
1.10
X-Axis sensitivity adjustment value 1.18
Y-Axis sensitivity adjustment value 1.19
Z-Axis sensitivity adjustment value 1.14
a(x,y,z) -2.75 106.81 1006.53 [mg] g(x,y,z) -0.09 0.02 -0.14 [dps] o(y,p,r) 126.78 -34.44 5.73 rate = 0.03 Hz
a(x,y,z) -0.61 108.15 1006.90 [mg] g(x,y,z) 0.00 -0.06 -0.13 [dps] o(y,p,r) 126.93 -33.08 5.69 rate = 941.38 Hz
a(x,y,z) -3.23 106.26 1008.12 [mg] g(x,y,z) -0.04 0.05 -0.08 [dps] o(y,p,r) 127.03 -31.72 5.67 rate = 945.49 Hz
a(x,y,z) -3.23 107.54 1007.51 [mg] g(x,y,z) 0.02 0.00 -0.08 [dps] o(y,p,r) 127.16 -30.39 5.65 rate = 943.27 Hz
a(x,y,z) -2.38 108.28 1005.31 [mg] g(x,y,z) -0.05 -0.01 -0.08 [dps] o(y,p,r) 127.27 -29.07 5.63 rate = 945.55 Hz
a(x,y,z) -1.34 106.63 1002.56 [mg] g(x,y,z) 0.00 0.01 -0.14 [dps] o(y,p,r) 127.38 -27.79 5.62 rate = 942.90 Hz
a(x,y,z) -2.44 106.45 1003.05 [mg] g(x,y,z) -0.05 0.01 -0.09 [dps] o(y,p,r) 127.53 -26.54 5.60 rate = 945.51 Hz
a(x,y,z) 0.92 110.17 1005.13 [mg] g(x,y,z) -0.07 0.01 -0.08 [dps] o(y,p,r) 127.67 -25.33 5.58 rate = 943.27 Hz
a(x,y,z) -2.99 106.38 1008.42 [mg] g(x,y,z) -0.03 -0.05 -0.14 [dps] o(y,p,r) 127.81 -24.15 5.58 rate = 945.55 Hz
a(x,y,z) 1.28 108.89 1004.70 [mg] g(x,y,z) 0.07 0.02 -0.11 [dps] o(y,p,r) 127.94 -23.02 5.57 rate = 942.95 Hz
a(x,y,z) -1.95 111.33 1006.35 [mg] g(x,y,z) 0.03 -0.02 -0.03 [dps] o(y,p,r) 128.13 -21.94 5.55 rate = 945.48 Hz
a(x,y,z) -0.73 104.19 1004.94 [mg] g(x,y,z) 0.06 0.06 -0.17 [dps] o(y,p,r) 128.26 -20.87 5.55 rate = 943.26 Hz
a(x,y,z) -1.95 107.54 1006.29 [mg] g(x,y,z) 0.05 0.05 -0.05 [dps] o(y,p,r) 128.43 -19.86 5.54 rate = 945.53 Hz
a(x,y,z) 0.85 106.93 1002.87 [mg] g(x,y,z) 0.00 0.00 -0.14 [dps] o(y,p,r) 128.57 -18.89 5.53 rate = 942.98 Hz
a(x,y,z) -3.85 105.41 1004.82 [mg] g(x,y,z) -0.06 0.02 -0.13 [dps] o(y,p,r) 128.77 -17.98 5.52 rate = 945.44 Hz
a(x,y,z) 5.74 105.77 996.83 [mg] g(x,y,z) -0.14 -0.04 -0.18 [dps] o(y,p,r) 128.96 -17.10 5.53 rate = 943.27 Hz
a(x,y,z) -1.22 108.03 1006.10 [mg] g(x,y,z) -0.05 0.07 -0.08 [dps] o(y,p,r) 129.14 -16.27 5.52 rate = 945.54 Hz
a(x,y,z) -0.31 109.19 1005.92 [mg] g(x,y,z) -0.02 0.02 -0.08 [dps] o(y,p,r) 129.26 -15.43 5.54 rate = 942.97 Hz
a(x,y,z) -1.65 107.79 1004.88 [mg] g(x,y,z) 0.02 0.04 -0.14 [dps] o(y,p,r) 129.44 -14.67 5.53 rate = 945.46 Hz
a(x,y,z) -1.53 109.19 1005.43 [mg] g(x,y,z) -0.03 -0.02 -0.21 [dps] o(y,p,r) 129.62 -13.94 5.53 rate = 943.30 Hz
a(x,y,z) -0.37 108.95 1004.09 [mg] g(x,y,z) -0.01 -0.01 -0.11 [dps] o(y,p,r) 129.80 -13.25 5.54 rate = 945.47 Hz
a(x,y,z) -1.46 107.06 1004.46 [mg] g(x,y,z) -0.05 -0.02 -0.07 [dps] o(y,p,r) 129.94 -12.56 5.55 rate = 943.27 Hz
a(x,y,z) -0.85 106.51 1004.88 [mg] g(x,y,z) -0.05 0.00 -0.06 [dps] o(y,p,r) 130.04 -11.91 5.56 rate = 945.46 Hz
a(x,y,z) -1.77 108.03 1004.52 [mg] g(x,y,z) -0.04 0.05 -0.13 [dps] o(y,p,r) 130.17 -11.30 5.56 rate = 943.29 Hz
a(x,y,z) -0.18 105.29 1005.80 [mg] g(x,y,z) -0.05 0.02 -0.08 [dps] o(y,p,r) 130.32 -10.72 5.57 rate = 945.50 Hz
a(x,y,z) -1.16 108.34 1004.70 [mg] g(x,y,z) -0.01 0.04 -0.06 [dps] o(y,p,r) 130.45 -10.17 5.59 rate = 943.26 Hz
a(x,y,z) -1.65 108.40 1004.46 [mg] g(x,y,z) -0.03 0.03 -0.05 [dps] o(y,p,r) 130.60 -9.65 5.59 rate = 945.44 Hz
a(x,y,z) -1.16 108.46 1003.54 [mg] g(x,y,z) -0.02 0.02 -0.08 [dps] o(y,p,r) 130.74 -9.16 5.60 rate = 943.32 Hz
a(x,y,z) -2.14 109.19 1007.45 [mg] g(x,y,z) -0.03 0.08 -0.11 [dps] o(y,p,r) 130.87 -8.69 5.61 rate = 945.48 Hz
a(x,y,z) -3.72 107.91 1004.46 [mg] g(x,y,z) -0.05 0.04 -0.06 [dps] o(y,p,r) 131.00 -8.24 5.62 rate = 943.26 Hz
a(x,y,z) -2.99 107.91 1005.19 [mg] g(x,y,z) 0.02 -0.01 -0.08 [dps] o(y,p,r) 131.19 -7.83 5.63 rate = 945.44 Hz
a(x,y,z) -0.85 106.63 1005.43 [mg] g(x,y,z) 0.03 0.04 -0.11 [dps] o(y,p,r) 131.33 -7.43 5.64 rate = 943.31 Hz
[... 278 lines cut ...]

        

        // With these settings the filter is updating at a ~145 Hz rate using the Madgwick scheme and 
        // >200 Hz using the Mahony scheme even though the display refreshes at only 2 Hz.
        // The filter update rate is determined mostly by the mathematical steps in the respective algorithms, 
        // the processor speed (8 MHz for the 3.3V Pro Mini), and the magnetometer ODR:
        // an ODR of 10 Hz for the magnetometer produce the above rates, maximum magnetometer ODR of 100 Hz produces
        // filter update rates of 36 - 145 and ~38 Hz for the Madgwick and Mahony schemes, respectively. 
        // This is presumably because the magnetometer read takes longer than the gyro or accelerometer reads.
        // This filter update rate should be fast enough to maintain accurate platform orientation for 
        // stabilization control of a fast-moving robot or quadcopter. Compare to the update rate of 200 Hz
        // produced by the on-board Digital Motion Processor of Invensense's MPU6050 6 DoF and MPU9150 9DoF sensors.
        // The 3.3 V 8 MHz Pro Mini is doing pretty well!

        //digitalWrite(myLed, !digitalRead(myLed));
        count = millis(); 
        sumCount = 0;
        sum = 0;    
    }`

Could advise me on what is going wrong?
Also, in many threads on mention filter rate and iteration but I never see any for loop indicating of an iterative process in your code. Do you have some material ?

Thanks

kriswiner, 2019-01-04:
Sorry, I can;t wade through all of this junk.

What is the problem exactly?

laurentvm, 2019-01-04:
With standing still device, the yaw pitch and roll never stabilizes, 

I dump below the serial output, y,p,r are the yaw pitch and roll.

You can see that for instance the yaw is moving from 19 degress to 2 degrees while the device is not moving at all.

```
MPU9250 9-axis motion sensor...
MPU9250 I AM 71 I should be 71
MPU9250 is online...
x-axis self test: acceleration trim within : 1.9% of factory value
y-axis self test: acceleration trim within : 0.3% of factory value
z-axis self test: acceleration trim within : 1.9% of factory value
x-axis self test: gyration trim within : -4.4% of factory value
y-axis self test: gyration trim within : -0.5% of factory value
z-axis self test: gyration trim within : 0.2% of factory value
 Calibrate gyro and accel
accel biases (mg)
47.73
-12.57
63.72
gyro biases (dps)
-1.50
0.64
-0.24
MPU9250 initialized for active data mode....
AK8963 I AM 48 I should be 48
AK8963 initialized for active data mode....
Mag Calibration: Wave device in a figure eight until done!
Mag Calibration done!
AK8963 mag biases (mG)
-454.32
188.74
-54.92
AK8963 mag scale (mG)
0.90
1.02
1.10
X-Axis sensitivity adjustment value 1.18
Y-Axis sensitivity adjustment value 1.19
Z-Axis sensitivity adjustment value 1.14
g(x,y,z) 0.04 -0.14 -0.05 [dps] o(y,p,r) 18.93 -43.43 -132.57 rate = 0.03 Hz
g(x,y,z) -0.06 -0.08 0.02 [dps] o(y,p,r) 18.10 -44.00 -131.37 rate = 937.11 Hz
g(x,y,z) 0.01 -0.08 -0.05 [dps] o(y,p,r) 17.27 -44.61 -130.06 rate = 939.82 Hz
g(x,y,z) -0.01 -0.05 -0.02 [dps] o(y,p,r) 16.35 -45.20 -128.76 rate = 938.97 Hz
g(x,y,z) -0.05 -0.07 -0.08 [dps] o(y,p,r) 15.30 -45.74 -127.39 rate = 938.93 Hz
g(x,y,z) 0.03 -0.11 0.06 [dps] o(y,p,r) 14.29 -46.28 -125.97 rate = 938.87 Hz
g(x,y,z) -0.03 0.00 -0.06 [dps] o(y,p,r) 13.31 -46.81 -124.47 rate = 939.13 Hz
g(x,y,z) 0.07 -0.14 -0.01 [dps] o(y,p,r) 12.09 -47.35 -122.91 rate = 939.88 Hz
g(x,y,z) 0.02 -0.02 -0.02 [dps] o(y,p,r) 10.90 -47.88 -121.28 rate = 938.97 Hz
g(x,y,z) 0.06 -0.11 0.02 [dps] o(y,p,r) 9.77 -48.39 -119.70 rate = 938.79 Hz
g(x,y,z) -0.04 -0.08 0.08 [dps] o(y,p,r) 8.59 -48.89 -117.99 rate = 939.13 Hz
g(x,y,z) 0.00 -0.18 0.00 [dps] o(y,p,r) 7.22 -49.36 -116.17 rate = 940.03 Hz
g (x,y,z) -0.08 -0.14 0.02 [dps] o(y,p,r) 5.75 -49.82 -114.29 rate = 939.00 Hz
g(x,y,z) 0.02 0.01 0.06 [dps] o(y,p,r) 4.19 -50.24 -112.34 rate = 939.03 Hz
g(x,y,z) 0.05 -0.10 -0.05 [dps] o(y,p,r) 2.59 -50.65 -110.28 rate = 938.81 Hz
```

kriswiner, 2019-01-04:
Two possibilities, poor calibration or not using NED or ENU orientation
when feeding the fusion filter.

kriswiner, 2019-01-04:
> ok,
>
> So I did 2 graph. One without doing anything during the magnetometer
> calibration and one where I've tried to make 8 shapes during the
> calibration.
>
> [image: 2019-01-04 22_26_28-book1 - excel]
> <https://user-images.githubusercontent.com/37613497/50711900-eae80800-106f-11e9-9476-884efed1d4ee.png>
>
> [image: 2019-01-04 22_26_44-book1 - excel]
> <https://user-images.githubusercontent.com/37613497/50711909-f50a0680-106f-11e9-96ff-19e9529b5bdd.png>
>
> Does it look good to you?

The question is, are the three post-calibration plots circular and centered
on the origin.

No, is my answer.

laurentvm, 2019-01-04:
So, the solution is to make better '8-shape' ? Increase sampling time during calibration?

kriswiner, 2019-01-04:
The calibration method is to sample every bit of the 3D response surface
multiple times. The quality of the calibration can be checked. You can
establish your own criteria for goodness. One easy check is Mz up == -Mz
down.

laurentvm, 2019-01-05:
I get that but it is not the case.

If I start step by step. I get this graph during calibration

[image]

Is it normal that I do not get eveything already overlapping? It is spread over 3 different quadrant.

One thing worth mentionning is that I am not using the Wire class:

```
void writeByte(uint8_t address, uint8_t subAddress, uint8_t data)
{
    /*
    Wire.beginTransmission(address);  // Initialize the Tx buffer
    Wire.write(subAddress);           // Put slave register address in Tx buffer
    Wire.write(data);                 // Put data in Tx buffer
    Wire.endTransmission();           // Send the Tx buffer
    */
    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, ( address << 1 ) | WRITE_BIT, ACK_CHECK_EN);
    // write sub-address and data
    uint8_t buf[2] = {subAddress, data};
    i2c_master_write(cmd, buf, sizeof(buf), ACK_CHECK_EN);
    i2c_master_stop(cmd);
    i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);
}

uint8_t readByte(uint8_t address, uint8_t subAddress)
{
    /*
    uint8_t data; // `data` will store the register data	 
    Wire.beginTransmission(address);         // Initialize the Tx buffer
    Wire.write(subAddress);	                 // Put slave register address in Tx buffer
    Wire.endTransmission(I2C_NOSTOP);        // Send the Tx buffer, but send a restart to keep connection alive
    //	Wire.endTransmission(false);             // Send the Tx buffer, but send a restart to keep connection alive
    //	Wire.requestFrom(address, 1);  // Read one byte from slave register address 
    Wire.requestFrom(address, (size_t) 1);   // Read one byte from slave register address 
    data = Wire.read();                      // Fill Rx buffer with result
    return data;                             // Return data read from slave register
    */
      // write sub-address
    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, ( address << 1 ) | WRITE_BIT, ACK_CHECK_DIS);
    i2c_master_write_byte(cmd, subAddress, ACK_CHECK_EN);
    i2c_master_stop(cmd);
    i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);
    // read data
    uint8_t data = 0;
    cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, ( address << 1 ) | READ_BIT, ACK_CHECK_EN);
    i2c_master_read_byte(cmd, &data, NACK_VAL);
    i2c_master_stop(cmd);
    i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000 / portTICK_RATE_MS);
    i2c_cmd_link_delete(cmd);
    return data;
}

void readBytes(uint8_t address, uint8_t subAddress, uint8_t count, uint8_t * dest)
{  
    /*
    Wire.beginTransmission(address);   // Initialize the Tx buffer
    Wire.write(subAddress);            // Put slave register address in Tx buffer
    Wire.endTransmission(I2C_NOSTOP);  // Send the Tx buffer, but send a restart to keep connection alive
    //	Wire.endTransmission(false);       // Send the Tx buffer, but send a restart to keep connection alive
    uint8_t i = 0;
    //        Wire.requestFrom(address, count);  // Read bytes from slave register address 
    Wire.requestFrom(address, (size_t) count);  // Read bytes from slave register address 
    while (Wire.available()) {
        dest[i++] = Wire.read(); }         // Put read results in the Rx buffer
  */

  // write sub-address
  i2c_cmd_handle_t cmd = i2c_cmd_link_create();
  i2c_master_start(cmd);
  i2c_master_write_byte(cmd, ( address << 1 ) | WRITE_BIT, ACK_CHECK_DIS);
  i2c_master_write_byte(cmd, subAddress, ACK_CHECK_EN);
  i2c_master_stop(cmd);
  esp_err_t ret = i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000 / portTICK_RATE_MS);
  i2c_cmd_link_delete(cmd);
  if (ret != ESP_OK) return ;//false;
  // read data
  cmd = i2c_cmd_link_create();
  i2c_master_start(cmd);
  i2c_master_write_byte(cmd, ( address << 1 ) | READ_BIT, ACK_CHECK_EN);
  if (count > 1) {
      i2c_master_read(cmd, dest, count - 1, ACK_VAL);
  }
  i2c_master_read_byte(cmd, dest + count - 1, NACK_VAL);
  i2c_master_stop(cmd);
  ret = i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000 / portTICK_RATE_MS);
  i2c_cmd_link_delete(cmd);

}

```

laurentvm, 2019-01-05:
Hi Kris, 

I changed the calibration to a while loop that continue for a for sample_count cycle until I do not find min/max value bigger/smaller. Like that I ensure that I have reached the bounds on each axis. Doing so gives me:

[image]

This is good right?

kriswiner, 2019-01-05:
Looks much better.

closed 2019-01-08 by laurentvm

shehanshaman, 2019-07-10:
I have the same problem, I couldn't understand how the graph was drawn and what is the exact way of rotating (we can rotate  according to 8 in many ways)

laurentvm, 2019-07-10:
https://www.youtube.com/watch?v=vUf6BNDToyY