kriswiner/MPU9250 #335: Why the accelerometer output is always 0,0,1 at rest whatever initial Z axis orientation

laurentvm, 2019-01-07:
Kris, 

Still working towards have a filter working.

I've remarked something that puzzle me.

If I start the device with accelerometer z point to earth' center I get this:
```
Bias : 46.33/-14.95/68.97
```
Then obvisouly:
```
a(x,y,z) -0.79 -1.16 1007.20 [mg]
```
I tilt to put x aligned with g:
```
a(x,y,z) 985.84 2.69 70.07 [mg]
```
Still make sense.

Now if I start the device with x aligned with gravity I have:
```
Bias: 1041.81/-7.87/-933.41
```
Start:
```
a(x,y,z) 0.12 2.32 998.47 [mg]
```
Rotate to have z aligned with gravity:
```
a(x,y,z) -998.41 -9.70 1998.11 [mg]
```

My question is:

1. The sensor is doing post-processing to output always the gravity on z axis? Whatever its own labeled axis internally?
2. How can I get something that makes sense when the device is not started in a good aligned position?
3. What if gravity is perfectly aligned at sqrt(3)/3 vs x,y,z ? How the sensor would pick the z?

Now, I also see something strange in gyro/acc axis.

I start with Z>0 (I guess pointing to ground). I flip around y to get g on X axis. This is for me a positive Y rotation but I get <0 gyro => Yacc = -YGyro.
I flip back and then flip to y pointing sky => positive rotation around x but I get gyro < 0 ==> Xacc = -Xacc. Does this sound right for you? The orientation is completely different from what I see and all issues. 

[image]

Then how do get sign of mag?

kriswiner, 2019-01-07:
You need to have a more sophisticated bias calculation then. I use this
kind of test, which usually works:

     if(offset[2] > +0.5f) offset[2] = offset[2] - 1.0f;
     if(offset[2] < -0.5f) offset[2] = offset[2] + 1.0f;

then apply to all axes and you should get reasonable results no matter what
crazy way you d the calibration.

laurentvm, 2019-01-07:
Trying to find how are oriented the axis of the sensors: Accelero and Gyro as starting point. 
To this end, I move the sensors in know direction and look at the sign of the gyro output to deduct how the  axis are oriented.

I do not have a board where the axis are written on it as I am playing with an ODB dongle with other components on the board.

kriswiner, 2019-01-07:
Check the data sheet for the correct orientation and sensor response.

laurentvm, 2019-01-07:
Well, I would love to have a datasheet available ;)

kriswiner, 2019-01-07:
Huh?

https://www.invensense.com/products/motion-tracking/9-axis/mpu-9250/

laurentvm, 2019-01-08:
ok, I was confused becuse many issue treat this axis orientation and sign issues. That is true if they want to use your code for another sensor but the axis you defined are correct for the MPU.

closed 2019-01-08 by laurentvm