kriswiner/MPU9250 #336: slow convergence of Yaw Pitch Roll

laurentvm, 2019-01-08:
Running on ESP32, iteraring 5 times the filter each time I get new data from the sensors I have a very slow convergence:

At start, it takes 60 seconds to converge
[image]

If I move it, for unknown reason it seems to be faster to converge, but still slow to me
[image]

Frequency is 4kHz, does that make sense?
```
8408 a(x,y,z) 1.65 1.83 1008.00 [mg] g(x,y,z) -0.06 -0.01 -0.02 [dps] m(x,y,z) 124 -126 363 [mG] q(0,x,y,z)0.89 -0.19 0.40 0.09 o(y,p,r) 0.88 47.95 -23.40 rate = 3.85 kHz
8513 a(x,y,z) 1.16 1.59 1009.64 [mg] g(x,y,z) -0.09 0.01 -0.08 [dps] m(x,y,z) 121 -111 357 [mG] q(0,x,y,z)0.90 -0.19 0.39 0.09 o(y,p,r) 1.19 47.48 -22.96 rate = 4.17 kHz
```

kriswiner, 2019-01-08:
4 kHz seems about right. I would recommend running the gyro/accel rates at
500 Hz or even 1 kHz, and the mag at 100 Hz.

I see much less "slowness" on my ESP32. Not sure what you are doing wrong.
Are you using 400 kHz I2C bus speed and using the interrupt for data ready?

laurentvm, 2019-01-09:
Hi, I am using your sketch [https://github.com/kriswiner/MPU9250/blob/master/MPU9250BasicAHRS_t3.ino](url) where I modified the call the filter based on [https://github.com/kriswiner/MPU9250/blob/master/AK8963_as_slave/AK8963Slave0_MPU9250_Ladybug.ino](url) to add iteration on the filter.

kriswiner, 2019-01-09:
> Hi, I am using your sketch
> https://github.com/kriswiner/MPU9250/blob/master/MPU9250BasicAHRS_t3.ino
> <http://url> where I modified the call to the filter based on
> https://github.com/kriswiner/MPU9250/blob/master/AK8963_as_slave/AK8963Slave0_MPU9250_Ladybug.ino
> <http://url> to add iteration.

Are you using 400 kHz I2C bus speed and using the interrupt for data
ready?

kriswiner, 2019-01-10:
> Yes but I have a modified I2C call compared to you
>
>
>
> bool initI2C()
> {
>     i2c_port_t i2c_master_port = I2C_NUM_0;
>     i2c_config_t conf;
>     conf.mode = I2C_MODE_MASTER;
>     conf.sda_io_num = (gpio_num_t)21;
>     conf.sda_pullup_en = GPIO_PULLUP_ENABLE;
>     conf.scl_io_num = (gpio_num_t)22;
>     conf.scl_pullup_en = GPIO_PULLUP_ENABLE;
>     //conf.master.clk_speed = 100000;
>     conf.master.clk_speed = 400000L;
>     //400000L
>     return i2c_param_config(i2c_master_port, &conf) == ESP_OK &&
>     i2c_driver_install(i2c_master_port, conf.mode, 0, 0, 0) == ESP_OK;
> }
>
> #define WRITE_BIT         I2C_MASTER_WRITE /*!< I2C master write */
> #define READ_BIT          I2C_MASTER_READ  /*!< I2C master read */
> #define ACK_CHECK_EN      0x1              /*!< I2C master will check ack from slave*/
> #define ACK_CHECK_DIS     0x0              /*!< I2C master will not check ack from slave */
> #define ACK_VAL           (i2c_ack_type_t)0x0              /*!< I2C ack value */
> [... 93 lines cut ...]
>
> This is a timing plot
>
> [image: image]
> <https://user-images.githubusercontent.com/37613497/50988707-b364da00-150d-11e9-8d83-3d498af92038.png>

So you are sending data to the serial terminal at 10 Hz?

laurentvm, 2019-01-10:
Yes

kriswiner, 2019-01-10:
Doesn't this affect the quaternion update rate?

laurentvm, 2019-01-10:
I'm not sure but I do not think it affects. 

It affects for sure how fast we can retrieve a newData from sensor. But the sensor is not moving, so having new data is not helping or at least I do not see why it would help, nothing is changing. 

Changing to 1hz or 0.2Hz output does not change:
[image]

I'm iterating 5 times the filter on the same newData, should i push this much higher?

laurentvm, 2019-01-10:
Kris, 

I have this in the init
`writeByte(MPU9250_ADDRESS, SMPLRT_DIV, 0x04); `
I guess that's wrong, seems to be 200hz?
it should be 
` writeByte(MPU9250_ADDRESS, SMPLRT_DIV, 0x00);  `
as in the calibration routine

Where do I change the gyro and Mag rate?
Mag, I have this
`uint8_t Mmode = 0x06;  `

kriswiner, 2019-01-10:
I believe changing SMPLRT_DIV to 0 sets both accel and gyro rates to 1 kHz.
Mag at 0x06 is already at 100 Hz.

// Configure Gyro and Thermometer
 // Disable FSYNC and set thermometer and gyro bandwidth to 41 and 42 Hz,
respectively;
 // minimum delay time for this setting is 5.9 ms, which means sensor
fusion update rates cannot
 // be higher than 1 / 0.0059 = 170 Hz
 // DLPF_CFG = bits 2:0 = 011; this limits the sample rate to 1000 Hz for
both
 // With the MPU9250, it is possible to get gyro sample rates of 32 kHz
(!), 8 kHz, or 1 kHz
  writeByte(MPU9250_ADDRESS, CONFIG, 0x03);  // 41 Hz bandwidth, 5.9 ms
delay
 // writeByte(MPU9250_ADDRESS, CONFIG, 0x02);  // 92 Hz bandwidth, 3.9 ms
delay

 // Set sample rate = gyroscope output rate/(1 + SMPLRT_DIV)
  writeByte(MPU9250_ADDRESS, SMPLRT_DIV, 0x05);  // Use a 167 Hz rate; a
rate consistent with the filter update rate
                                    // determined in set in CONFIG above

// Set accelerometer sample rate configuration
 // It is possible to get a 4 kHz sample rate from the accelerometer by
choosing 1 for
 // accel_fchoice_b bit [3]; in this case the bandwidth is 1.13 kHz
  c = readByte(MPU9250_ADDRESS, ACCEL_CONFIG2); // get current
ACCEL_CONFIG2 register value
  c = c & ~0x0F; // Clear accel_fchoice_b (bit 3) and A_DLPFG (bits [2:0])
  c = c | 0x03;  // Set accelerometer rate to 1 kHz and bandwidth to 41 Hz
  writeByte(MPU9250_ADDRESS, ACCEL_CONFIG2, c); // Write new ACCEL_CONFIG2
register value

kriswiner, 2019-01-10:
> It does not change anything. I got approximately the same timings
> void initMPU9250()
>     // wake up device
>     writeByte(MPU9250_ADDRESS, PWR_MGMT_1, 0x00); // Clear sleep mode bit (6), enable all sensors
>     delay(100); // Wait for all registers to reset
>     // get stable time source
>     writeByte(MPU9250_ADDRESS, PWR_MGMT_1, 0x01);  // Auto select clock source to be PLL gyroscope reference if ready else
>     delay(200);
>     writeByte(MPU9250_ADDRESS, SMPLRT_DIV, 0x00);  // Use a 200 Hz rate; a rate consistent with the filter update rate
>     // Set gyroscope full scale range
>     // Range selects FS_SEL and GFS_SEL are 0 - 3, so 2-bit values are left-shifted into positions 4:3
>     uint8_t c = readByte(MPU9250_ADDRESS, GYRO_CONFIG); // get current GYRO_CONFIG register value
>     // c = c & ~0xE0; // Clear self-test bits [7:5]
>     //c = c & ~0x02; // Clear Fchoice bits [1:0]
>     c = c & ~0x03; // Clear Fchoice bits [1:0]
>     c = c & ~0x18; // Clear GFS bits [4:3]
>     c = c | Gscale << 3; // Set full scale range for the gyro
>     // c =| 0x00; // Set Fchoice for the gyro to 11 by writing its inverse to bits 1:0 of GYRO_CONFIG
>     writeByte(MPU9250_ADDRESS, GYRO_CONFIG, c ); // Write new GYRO_CONFIG value to register
>     // Set accelerometer full-scale range configuration
>     c = readByte(MPU9250_ADDRESS, ACCEL_CONFIG); // get current ACCEL_CONFIG register value
>     c = c & ~0x18;  // Clear AFS bits [4:3]
>     c = c | Ascale << 3; // Set full scale range for the accelerometer
>     writeByte(MPU9250_ADDRESS, ACCEL_CONFIG, c); // Write new ACCEL_CONFIG register value
>     // The accelerometer, gyro, and thermometer are set to 1 kHz sample rates,
> [... 4 lines cut ...]
> [... cut ...] ntrolled by the Arduino as master
>     //   writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x22);
>     writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x12);  // INT is 50 microsecond pulse and any read to clear
>     writeByte(MPU9250_ADDRESS, INT_ENABLE, 0x01);  // Enable data ready (bit 0) interrupt
>     delay(100);

Don't know what to tell you...Maybe try another MCU and/or senor breakout
to be sure?

laurentvm, 2019-01-10:
I will try to use the Wire class as you do. There is maybe something wrong.

Could you advise me where to buy a good board (Teensy?). I'm in europe so a european should would be better and faster.

There still something I do not understand on the concept. If the filter is a gradient based algo, it's higly dependent on the starting point. You get a sensor data point, previous quaternion and iterate a few time to converge the filter from previous quaternion.

When the sensor is not moving at all, I probably start with an initial bad starting point. If you look at previous graph, we see that. It reach very slowly an optimum and then it's faster because you follow a movement from previous points that is not too far from current position.

When you are standing still, basically, the update rate of sensor (if resonnable) does not affect anything because you do not have to follow a movement. You are iterating with an initial point that is the same after the filter.

kriswiner, 2019-01-10:
If the sensor is well calibrated, yes it will take some (short) time
(seconds) to converge to a stable absolute orientation when starting from
the arbitrary 1 0 0 0 quaternion, but it shouldn;t take 60 seconds.

I recommend this
<https://www.tindie.com/products/TleraCorp/ladybug-stm32l432-development-board/>
device. or this
<https://www.tindie.com/products/TleraCorp/dragonfly-stm32l47696-development-board/>
one.