kriswiner/MPU9250 #338: Poor Euler Angle Reading while rotating on z axis flat on table

WilliamPitman, 2019-01-09:
I am having trouble using the sensor fusion algorithm when the IMU is flat on a table and rotated on the z axis. The euler angle does not change very much. We are using the version of the Mahony algorithm that does not take the magnetometer data.

[image]

When I rotate the IMU on another axis, I get good results, also if the IMU is not flat on the table but rotated on the z axis I get good results. I think that the reason for the limited response is because when the IMU is rotating flat on the table its position relative to the gravitational field is not changing. Is there a way to fix this? Can I tune the IMU to be more responsive to gyro changes?

kriswiner, 2019-01-09:
Wasn;t aware there was a 6 DoF version of the mahony algorithm, do you mean
Madgwick?

Not sure why you'd expect accurate heading from a 6 DoF fusion algorithm in
any case. How did you calibrate your sensors?

WilliamPitman, 2019-01-10:
I am using the MATLAB files from http://x-io.co.uk/open-source-imu-and-ahrs-algorithms/ 
which contains a function UpdateIMU(obj, Gyroscope, Accelerometer) within the Mahoney function. It looks to be different from the UpdateIMU(obj, Gyroscope, Accelerometer) within the Madgwick Function. 

I was originally using the Magnetometer but I found it to be troublesome with occasional dropouts. Currently I am calibrating the gyroscope and accelerator by leaving the sensor flat on a table while the calibration runs.

I don't expect an extremely accurate heading but the algorithm output gives almost nothing using this. On the other hand if the sensor is rotated on another plane (with a change in the gravitational field) the output is pretty good. Is this just a restriction of the sensor?

kriswiner, 2019-01-10:
This is a restrction of the fusion algorithm.

WilliamPitman, 2019-01-10:
Okay, thank you. So is my only option to try to use the Magnetometer to fix this?

kriswiner, 2019-01-10:
Not sure this is the only option, you might still be able to make 6 DoF
give you a sensible heading fro your application but I would recommend a 9
DoF approach for most applications. Not sure what problem you had but we
obtain 4-degree absolute heading accuracy with the MPU9250 and the Madgwick
or Mahony fusion algorithms running on a fast enough MCU. Why would anyone
settle for 6 DoF?