kriswiner/MPU9250 #340: Verical Data Restricted

VGiaglisis, 2019-01-11:
Hello Kris,
First of all thank you for the great contributions you have made with the MPU9250.
I have followed your instructions on calibrating the magnetometer and used the most recent code on an Arduino Nano. I am getting really stable results in terms of lateral and horizontal axes (pitch, roll) but the vertical data seem to be resticted between around +- 40 deg. I am transmiting quaternion data directly to a Unity script where the calculation are made for a visualisation on a simple cubic object. I have not used any king of Euler or Tait-Bryan conversion functions. Also I checked the local mag bias and compared them to the calibration results. One axis seemed to be off bt a lot. Could you please suggest some ideas on getting correct yaw data without restrictions?

Here is my code:

`
#include "quaternionFilters.h"
#include "MPU9250.h"

#define AHRS true         // Set to false for basic data read
#define SerialDebug true  // Set to true to get Serial output for debugging

// Pin definitions
int intPin = 12;  // These can be changed, 2 and 3 are the Arduinos ext int pins
int myLed  = 13;  // Set up pin 13 led for toggling
int j = 1;
int buttonState;
#define I2Cclock 400000
#define I2Cport Wire
//#define MPU9250_ADDRESS MPU9250_ADDRESS_AD0   // Use either this line or the next to select which I2C address your device is using
#define MPU9250_ADDRESS MPU9250_ADDRESS_AD1

MPU9250 myIMU(MPU9250_ADDRESS, I2Cport, I2Cclock);

void setup()
{
  Wire.begin();
  // TWBR = 12;  // 400 kbit/sec I2C speed
  Serial.begin(38400);

  while(!Serial){};

  // Set up the interrupt pin, its set as active high, push-pull
  pinMode(intPin, INPUT);
  digitalWrite(intPin, LOW);
  pinMode(myLed, OUTPUT);
  digitalWrite(myLed, HIGH);

  // Read the WHO_AM_I register, this is a good test of communication
  byte c = myIMU.readByte(MPU9250_ADDRESS, WHO_AM_I_MPU9250);
  Serial.print(F("MPU9250 I AM 0x"));
  Serial.print(c, HEX);
  Serial.print(F(" I should be 0x"));
  Serial.println(0x71, HEX);

  if (c == 0x71) // WHO_AM_I should always be 0x71
  {
    Serial.println(F("MPU9250 is online..."));

    // Start by performing self test and reporting values
    myIMU.MPU9250SelfTest(myIMU.selfTest);
    Serial.print(F("x-axis self test: acceleration trim within : "));
    Serial.print(myIMU.selfTest[0],1); Serial.println("% of factory value");
    Serial.print(F("y-axis self test: acceleration trim within : "));
    Serial.print(myIMU.selfTest[1],1); Serial.println("% of factory value");
    Serial.print(F("z-axis self test: acceleration trim within : "));
    Serial.print(myIMU.selfTest[2],1); Serial.println("% of factory value");
    Serial.print(F("x-axis self test: gyration trim within : "));
    Serial.print(myIMU.selfTest[3],1); Serial.println("% of factory value");
    Serial.print(F("y-axis self test: gyration trim within : "));
    Serial.print(myIMU.selfTest[4],1); Serial.println("% of factory value");
    Serial.print(F("z-axis self test: gyration trim within : "));
    Serial.print(myIMU.selfTest[5],1); Serial.println("% of factory value");

    // Calibrate gyro and accelerometers, load biases in bias registers
    myIMU.calibrateMPU9250(myIMU.gyroBias, myIMU.accelBias);

    myIMU.initMPU9250();
    // Initialize device for active mode read of acclerometer, gyroscope, and
[... 178 lines cut ...]
        Serial.print(*getQ());
        Serial.print(","); 
        Serial.print(*(getQ() + 1));
        Serial.print(","); 
        Serial.print(*(getQ() + 2));
        Serial.print(","); 
        Serial.print(*(getQ() + 3));
        Serial.print(","); 
        Serial.println(buttonState);
      }

 /*     if(SerialDebug)
      {
        Serial.print("rate = ");
        Serial.print((float)myIMU.sumCount / myIMU.sum, 2);
        Serial.println(" Hz");
      }
*/

      myIMU.count = micros();
      myIMU.sumCount = 0;
      myIMU.sum = 0;
    } // if (myIMU.delt_t > 500)
  } // if (AHRS)
}`

VGiaglisis, 2019-01-11:
Also these are my magnetometer plots before and after fixed calibration 

[image]
[image]

kriswiner, 2019-01-11:
Could be better, especially Mxz.

Are you feeding the sensor data to the fusion algorithm as NED or ENU?

VGiaglisis, 2019-01-11:
I am using NED I think. As for the calibration the variably showing big divergence from the expected is my

kriswiner, 2019-01-11:
How do you know you are using NED? It is a user choice, and the sensor data
need to be provided then in a particular order.

Maybe increase the calibration time to get My better calibrated?

VGiaglisis, 2019-01-12:
Ok let me rephrase, the biases are set based on NED longitude and latitude data. I am not sure about the order I am using. The calibration timer is set at 15 seconds but just to be sure I need to ask; When you are saying move in a figure 8 movement you are not referring to a 2D planar figure 8 movement, but a spatial movement trying to achieve various acceleration vectors in all the axes, right?
If so I will try changing mpu in case this one is faulty and adjust the timer to 20 sec for some new bias comparisons.

kriswiner, 2019-01-12:
Yes, the idea of mag calibration is to sample the entire 3D response
surface multiple times during the calibration.

The sensor data needs to be fed to the Madgwick filter as AN, AE, AD, GN,
GE, GD, MN, ME, MD for NED or you can also use ENU convention, but either
way, it needs to be consistent.

VGiaglisis, 2019-01-12:
Ok thank you, I will try to verify that they are fed correctly and repost with an update.
Really appreciate the help!

VGiaglisis, 2019-01-21:
Hello again,

I am sorry for the delays. I made numerous atemts of manual calibration (15 seconds of waving) and each time i got different numbers on the mag biases. I am feeding the sesor with:
MahonyQuaternionUpdate(-myIMU.ax, -myIMU.ay, myIMU.az, myIMU.gx * DEG_TO_RAD,
myIMU.gy * DEG_TO_RAD, -myIMU.gz * DEG_TO_RAD, myIMU.my,
myIMU.mx, myIMU.mz, myIMU.deltat);

Based on NED biases or the Magcal biases.
Any ideas?

kriswiner, 2019-01-21:
> I also get values like:
> AK8963 mag scale (mG)
> 1.01
> 0.76
> 1.44
> Which sound way off to me.

Yeah, looks like 15 seconds is not enought to fully calibrate the mag. it
needs to sample all 3 dimensions of the response surface several times.

VGiaglisis, 2019-01-23:
Update !
I found the problem, my waving calibration motions were too repetitive and i was geting huge (or very small) scalars in 2 planes. I tried waving to greate a planar 8-like shape in each plane (sagittal,coronal,transverse) and then did random figure 8 motions until the calibration was finished.
I got way better scales, and much more consistant biases and there seemed to be no restrictions in movement. Maybe if I increased to 20 seconds, I might have gotten even better numbers. I will try it
Thank you Kris, the problem was solved.
I will try posting a video of the calibration motion just to help others with the same problem.