kriswiner/MPU9250 #341: Inaccurate Yaw Values, with a seemingly calibrated Magnetometer

SilasWaxter, 2019-01-12:
Dear Kris,
I'm a junior in high and am working on a science research project.  I am using an MPU9250, a Teensy 3.2, and the "MPU9250BasicAHRS_t3.ino" with the display code removed.  I am having issues getting an accurate Yaw.  I have read some of the other issues and it seems that it would be an issue with the magnetometer not being calibrated.  However, I followed the steps to calibrate it (see video to correct me if the figure-eight motion is wrong) and graphed my values (see graph link below).  The graph shows concentric spheres that have a radius of ~250 mG.  I also have an filtering rate > 1kHz.  Below is a video of my device.  Notice that when I accelerate along an axis, the yaw values drift.  I have also had issues in the past where the degrees will be increasing until they hit about 260 then they start decreasing even though my rotational velocity is approx. constant.  Any help would be greatly appreciated.
Thanks,
Silas

**Graph:**https://drive.google.com/file/d/1wPDlHVcQsQg9OosKEtmkQXdNJDlerU_m/view?usp=sharing
**Video:**https://drive.google.com/file/d/1cub582lOsj0S1ebk31tZGa7HnVYIuDq0/view?usp=sharing

kriswiner, 2019-01-12:
Hard to tell from your plot but accel and gyro offset biases have to be
corrected as well as mag offset bias and scaling for best results. Easy
checks on these are" sensor flat andat rest, accle = 0, 0, 1 G, gyro - 0,
0, 0 dps. For mag, Mz should be equal and opposite when sensor facinf up vs
facing down.

Likely causesof your problems 1) need faster sample rate, maybe run accel
and gyro at 1 kHz, mag at 100 Hz. Feed sensor data into Madgwick/Mahony
ilter in consisten NED or ENU order, i.e.

Madgwick(AN, AE, AD, GN, GE, GD, MN, ME, MD). Or similar for ENU.

Are you taking your local magnetic declination into account. Won;t affect
yaw drift but will affect where 0 degree heading points.

SilasWaxter, 2019-01-12:
While laying flat with the z axis normal to the ground, my gyro readings are less than +- 0.10 deg/s, ax is between 0 and -5 mg, ay is -25 mg +- 5, az is 1000 mg +- 5, mz started at 600 then I flipped it and it was at -440, then I flipped back to normal position and it was 440.  Tested it back and forth two more times and it was 440 and -440.  Why is the mag reading in the beginning different?   While retesting, I tested my angles again.  This time I tested the Pitch and Roll.  My yaw would go up to 90 then start coming back down, even thought my rotational velocity was constant.  My roll, however, would range from -180 to +180.  I am confused as to why there is a difference in my range of degrees.  This is the code used to convert between quaternions and Euler Angles.
    a12 =   2.0f * (q[1] * q[2] + q[0] * q[3]);
    a22 =   q[0] * q[0] + q[1] * q[1] - q[2] * q[2] - q[3] * q[3];
    a31 =   2.0f * (q[0] * q[1] + q[2] * q[3]);
    a32 =   2.0f * (q[1] * q[3] - q[0] * q[2]);
    a33 =   q[0] * q[0] - q[1] * q[1] - q[2] * q[2] + q[3] * q[3];
    pitch = -asinf(a32);
    roll  = atan2f(a31, a33);
    yaw   = atan2f(a12, a22);
    pitch *= 180.0f / PI;
    yaw   *= 180.0f / PI; 
    yaw   += 15.2f; // Declination at Seaside, Oregon is 15 degrees 20 minutes
    if(yaw < 0) yaw   += 360.0f; // Ensure yaw stays between 0 and 360
    roll  *= 180.0f / PI;
    lin_ax = ax + a31;
    lin_ay = ay + a32;
    lin_az = az - a33;

I adjusted for my local magnetic declination, but it doesn't point in one consistent location as 0. It changes especially after I accelerate one way around the z-axis, then return it.  As far as update rate, I am using the magnetometer on 100 hz.  I haven't changed the update rate of the accel or gyro, but the function you included, which returns your update rate of the filter is running at an average of 1875 hz and drops to about 800 every so often. I assume this is when the new data is read from the sensor.  Is the default order of the Mahony/Madgwick Filter not NED or ENU?  Is there a better filter to use? I am using the Mahony right now.
Thanks again,
Silas

kriswiner, 2019-01-12:
You choose which accel axis is to point to true North. Then feed the filter
data in the right order, NED or ENU.

SilasWaxter, 2019-01-12:
I didn't change the order of what was fed to the filter.  I am using the default, NED, with the accel's x-axis pointed north, but the value is the opposite because gravity is positive down.  This is my input into the function.

MahonyQuaternionUpdate(-ax, ay, az, gx*PI/180.0f, -gy*PI/180.0f, -gz*PI/180.0f,  my,  -mx, mz);

kriswiner, 2019-01-12:
That should work.

SilasWaxter, 2019-01-12:
What is the problem with my yaw?

kriswiner, 2019-01-12:
No idea

SilasWaxter, 2019-01-12:
Okay, thanks for your help anyways.  And thank you for the code it is definitely appreciated.

kriswiner, 2019-01-13:
Took a look at that old sketch you are using. I hope you understood enough
to comment out these lines:

readMagData(magCount); // Read the x/y/z adc values
getMres();
magbias[0] = +470.; // User environmental x-axis correction in milliGauss,
should be automatically calculated
magbias[1] = +120.; // User environmental x-axis correction in milliGauss
magbias[2] = +125.; // User environmental x-axis correction in milliGauss

How are you managing the mag calibration?

kriswiner, 2019-01-13:
> I didn't have "readMagData(magCount); // Read the x/y/z adc values"
> commented out. I assume if it wasn't commented out, it would read the mag
> values whenever the 'new-data flag' was raised as well as whenever the
> 'new-mag-data flag' was raised. However, I tested it with the line
> commented out. The yaw was stuck between 210 and 225. When uncommented, the
> yaw was a lot better. I also noticed that my desk is metal. When I moved my
> device away from the desk, the angles were within 5 degrees when moving 90
> degrees. I also shook the sensor all around then brought it back.
> It would return quite fast and would be within +-2 degrees. The yaw seemed
> to deal better with accelerations after some time had gone by. Does that
> make sense with the filter?
>
> I removed all the other lines. I was reading another post and you were
> saying the best yaw you can expect to get is within 5 degrees. #115
> <https://github.com/kriswiner/MPU9250/issues/115>. Would that change with
> a higher filtering rate? For my application, I plan to display the euler
> angles in a C# app I'm programming for windows. The sensor module will
> communicate to the computer with a bluetooth module, the HC-08 (1mb/s).
> Sending over the raw accelerometer, gyroscope, and magnetometer data and
> filtering on the computer would mean a faster filtering rate. Would this
> increase the accuracy of yaw?
>
> My mag calibration uses the code below. I am rotating the sensor along all
> of its axis while moving in a figure eight at 1-2 rev/sec.
>
> [... 52 lines cut ...]
> dest2[1] = avg_rad/((float)mag_scale[1]);
> dest2[2] = avg_rad/((float)mag_scale[2]);
>
> Serial.println("Mag Calibration done!");
> }

4 degree heading accuracy is what you should expect with good calibration
using the MPU9250.

And, yes, keep the sensor away from strong external mag fields is required
to get best results

These:

magbias[0] = +470.; // User environmental x-axis correction in milliGauss,
should be automatically calculated
magbias[1] = +120.; // User environmental x-axis correction in milliGauss
magbias[2] = +125.; // User environmental x-axis correction in milliGauss

are obviously offset biases for a particular MPU9250 I used years ago, not
applicable to your sensor. Some people keep these values in their sketch
and wonder why they can;t get accurate heading!.

SilasWaxter, 2019-01-13:
Would a faster filter rate increase the accuracy? Should I switch to the MAX21100/LIS3MDL or the EM7180/LSM9DS0?

kriswiner, 2019-01-13:
Faster filter rate all else being equal will not improve accuracy, but
could improve stability and lower latency.

Accuracy depends, in order, on:

1. Sensor quality
2. Sensor calibration
...
10. fusion filter update rate
...
99. sensor fusion filter.

On the other hand, the fusion filter in the EM7180 is much better than the
simple Madgwick or Mahony filters, and is capable of routinely providing 2
degree heading accuracy or better. See this
<https://github.com/kriswiner/EM7180_SENtral_sensor_hub/wiki/K.-Limits-of-Absolute-Heading-Accuracy-Using-Inexpensive-MEMS-Sensors>
.

So for improvement in heading accuracy, I would suggest using the MPU9250
and a fast MCU like an STM32L4 for Madgwick/Mahony fusion on the host (or
off the host on a separate compute engine) for up to 4 degree heading
accuracy.

LSM6DSM+LIS2MDL and a fast MCU like an STM32L4 for  Madgwick/Mahony fusion
on the host (or off the host on a separate compute engine) for up to 3
degree heading accuracy.

The MPU9250 and EM7180 for up to 2 degree accuracy, or the
LSM6DSM+LIS2MDL+EM7180 for 1 - 2 degree heading accuracy.

These results require excellent sensor calibration, with somewhat more
laborious but easily done calibration methods.

SilasWaxter, 2019-01-13:
On further reading, it seems  each sensor would have there own EM7180.  I need the best accuracy possible, as the device I'm creating is used for medical purposes.

kriswiner, 2019-01-13:
My advice is to buy one of these
<https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution-lsm6dsm-lis2md/>
and test it for your application then. The only problem will be if you want
to design your own version, then you will need to license the EM7180
firmware from us.

SilasWaxter, 2019-01-13:
Ok.  I plan to turn the device into a product eventually, but I don't think I am going to create my own version of the sensor.  Theoretically speaking, how many units could you produce in a month?

kriswiner, 2019-01-13:
First of all, the EM7180 manages all of the sensors on the board, meaning
the accel, gyro, magnetometer and pressure sensor. So just one EM7180 is
required. I have hundreds of the boad I linked to and can have produced
thousands given 45 days notice. How many will you be needing?

SilasWaxter, 2019-01-13:
I'm just ordering 1 right now so I can produce my prototype, but within the next year, I plan to start my business and start selling my product, which uses two of them.  I think the business will take off, but we will see.  Thanks again for all your help I really appreciate it.

SilasWaxter, 2019-01-20:
I just got the EM7180_LSM6DSM_LIS2MDL_LPS22HB. I am unsure how to get started.

kriswiner, 2019-01-20:
Run this one a sensoble MCU:

https://github.com/kriswiner/EM7180_SENtral_sensor_hub/tree/master/EM7180_LSM6DSM_LIS2MDL_LPS22HB_Butterfly

after connecting 3V3, GND, SDA. SCL and an interrupt pin.

SilasWaxter, 2019-01-20:
I tried that, but it is for the butterfly MCU correct?  I don't have that.  Is there a sketch for an arduino or teensy?

kriswiner, 2019-01-20:
You'll have to make some small modification for those. Not too hard...

kriswiner, 2019-01-20:
Maybe try this library instead.

https://github.com/simondlevy/EM7180/blob/master/examples/MasterTest/MasterTest.ino

kriswiner, 2019-01-20:
This one os probbly closer for an AVR arduino.

https://github.com/kriswiner/ESP32/tree/master/EM7180/EM7180_LSM6DSM_LIS2MDL_LPS22HB_ESP32

SilasWaxter, 2019-01-20:
"Global variables use 2702 bytes (131%) of dynamic memory, leaving -654 bytes for local variables. Maximum is 2048 bytes."    Is there any variables I can remove without messing up the filtering?  I only need the orientation in Euler angles.  However, accuracy is most important.

kriswiner, 2019-01-20:
Better get a real MCU...

kriswiner, 2019-01-20:
You mentioned Teensy, this should work just fine.

SilasWaxter, 2019-01-20:
Are the hardware quaternions/angles what I want for accuracy or do I need to do some processing on the mcu?

kriswiner, 2019-01-20:
You can choose to output quaternions and construct Euler angles on the host
or you can choose to output Euler angles directly.

kriswiner, 2019-01-20:
> Ok do you have a link?

The one I sent you?

SilasWaxter, 2019-01-20:
I don't think so.  It doesn't mention Simon anywhere.  The ones you sent me have all the other data in them.

kriswiner, 2019-01-20:
https://github.com/simondlevy/EM7180/blob/master/examples/MasterTest/MasterTest.ino

SilasWaxter, 2019-01-27:
Dear Kris,
I ordered another sensor.  I intend to use 2 of them on the same device.  Is there a way I can change the i2c address?  I know I can use an i2c multiplexer, but is there a pin on the sensor similar to the MPU9250's AD0/SD0, which will alter the address of the EM7180?

SilasWaxter, 2019-01-27:
I cant.  I switched to the ESP-WROOM-32.  It has integrated bluetooth, which is nice bc it will cut down the size and complication of the device.  I will just use a multiplexer.  Thanks.

kriswiner, 2019-01-27:
Or use the nRF52 add-on board, much lower power than ESP32 especially if
coupled with the Butterfly.

SilasWaxter, 2019-01-27:
I think I'm gonna stick with the esp.  I can't use BLE bc the c# library im using, 32 feet, doesnt support it.  If I have issues with battery life I'll reconsider.

kriswiner, 2019-01-27:
> When you make the produce the sensors, do you place the components on the
> pcbs by hand?

Only the prototypes, then I send them to the fab for production if I decide
to sell them.

kriswiner, 2019-01-27:
> What is the minimum number of units per order? I am asking because
> eventually I need to have something like this on the EM7180 sensor.
> micro-usb_breakout
> <https://www.aliexpress.com/item/10pcs-MICRO-USB-To-DIP-Adapter-5pin-Female-Connector-B-Type-PCB-Converter-Breadboard-USB-01/32867946438.html?spm=2114.search0104.3.58.b1131037sUzjaq&ws_ab_test=searchweb0_0,searchweb201602_1_10065_10068_10130_10890_10547_319_10546_317_10548_10545_10696_453_10084_454_10083_10618_10307_538_537_536_10059_10884_10887_100031_321_322_10103-10890,searchweb201603_53,ppcSwitch_0&algo_expid=ca64687f-327c-4d11-acc1-177abda41f17-8&algo_pvid=ca64687f-327c-4d11-acc1-177abda41f17>
> Originally I was thinking that I would get a single layer pcb made that
> would connect the EM7180 and the micro-usb breakout. Now I wondering if it
> would be better to modify your pcb design and add a female micro-usb
> component.
>
> P.S. sorry if the link got messed up this is the first time I tried that.

Why do you need usb on the sensor breakout?

SilasWaxter, 2019-01-27:
Each sensor is going to connect to the mcu via a usb cable.

SilasWaxter, 2019-01-27:
Not for the consumer.  If their wire gets worn out they may have to buy a whole new product and thats crappy for them.  If I was someone using it and that happened, I would ask "why didn't they just use a replaceable cord?".  I am set using usb cables.

SilasWaxter, 2019-01-27:
I don't think many of the customers would feel confident enough to replace wires.

SilasWaxter, 2019-01-27:
When the sensor starts up, is there initialization code it has to run?  A time delay?

kriswiner, 2019-01-27:
Yes, probably won't work...

SilasWaxter, 2019-01-27:
Ok.  What do you think about the pcb?  Should I try to customize your pcb with the sensor, or should I get a single layer pcb that connects the sensor and the micor-usb pinout?

kriswiner, 2019-01-27:
depends on a lot of things.

SilasWaxter, 2019-01-27:
Like what?

kriswiner, 2019-01-27:
Skill level, experience, application, end user, cost, etc

SilasWaxter, 2019-02-27:
Hey Kris,
I need the angles on all three axises (roll, pitch, yaw) to be a range of 0 to 360.  Currently, I have both roll and yaw as this range, but the pitch range is different.  The range is from -90 to 90 (see [https://drive.google.com/file/d/1kULqbL3oiiO0bxU3yDOV19UMZs2hc9sI/view?usp=sharing](url)).  Do you know the equation to change pitch to a 0 to 360?

Current COde
`  float roll  = atan2(2.0f * (qw1 * qx1 + qy1 * qz1), qw1 * qw1 - qx1 * qx1 - qy1 * qy1 + qz1 * qz1);
  float pitch = asin(2.0f * qx1 * qy1 + 2.0f * qz1 * qw1);
  float yaw   = atan2(2.0f * (qx1 * qy1 + qw1 * qz1), qw1 * qw1 + qx1 * qx1 - qy1 * qy1 - qz1 * qz1);   

  pitch *= 180.0f / PI;
  yaw   *= 180.0f / PI; 
  if(yaw < 0) yaw   += 360.0f;          // Ensure yaw stays between 0 and 360
  roll  *= 180.0f / PI;
  if(roll <0) roll  += 360.0f;          // Ensure roll stays between 0 and 360`

kriswiner, 2019-02-27:
Heading is 0 to 360, pitch from -180 to 180 and roll from -90 to 90 by
definition, see https://en.wikipedia.org/wiki/Euler_angles

SilasWaxter, 2019-02-27:
Alright then.  Is it possible to measure the angle of the shortest path between two quaternions?

kriswiner, 2019-02-27:
What does this mean?

SilasWaxter, 2019-02-27:
Quaternions represent an orientation in 3d space.  Between two orientations, there is an axis that both orientations lie on.  Is there a way to measure the angle between the two quaternions along this axis?

kriswiner, 2019-02-27:
Yes.

SilasWaxter, 2019-02-27:
Do you know how to do that?

kriswiner, 2019-02-27:
Maybe look at using something like the dot product of the quaternions?

SilasWaxter, 2019-03-12:
Hey Kris thank you for your previous help.  

There seems to be industrial-grade IMUs that are even more accurate then these lower cost sensors.  What separates these categories?  

I am aiming for the highest accuracy of rotation, meaning I am less worried about accuracy in relation to true orientation (based on Earth) and more interested in the accuracy of rotation between two orientations. I realize that these are intertwined, but does this effect what an optimal sensor fusion algorithm looks like for me?

SilasWaxter, 2019-03-12:
Ok thank you.

SilasWaxter, 2019-04-08:
Hey Kris, how familiar are you with machine learning?

kriswiner, 2019-04-08:
Not too...why?

I do know that the latest LSM6DS0X accel/gyro has a state machine plus
machine learning capability built in, and I will eventually try to make use
of this...

SilasWaxter, 2019-04-09:
I am familiarizing myself with machine learning.  From what I understand, you can create a neural network with inputs, outputs, and layers in between.  You then need a large set of data, which is labeled.  In other words you have many data sets of inputs with what the outputs should give.  The "neural network algortithm" then forms an algortithm to convert the inputs into the outputs.

I am wondering if this would be more efficient then using a traditional method like a kalman, mahony, madgwick filter or if a combination would be optimal.  I am planning on contacting some prof. of computer science who specialize in machine learning.  Perhaps they can explain the limitations and any flaws in my thinking as well as point me in the right direction.

Mat-Lab has support for machine learning.  I was thinking of using it.  It also has support for of offloading the code in c to embedded processors.

While researching better IMUs, I came across a few that had "static gps heading".

http://www.airmar.com/productdescription.html?id=95              **(heading: 1deg static, 2deg dynamic)**
https://inertialsense.com/products/dual-gnss-ins-compass/       **(heading: gps 0.3deg w/ 1 meter baseline)**

For my application, the device is strapped to a person and measuring their joints.  They device would not be moving in a vector, which could be turned into a heading.  Does the gps/gnss only aid in accuary when the sensor is traveling in a direction?  It probably depends on the accuracy of the gnss system.

I'm interested to hear your thoughts.

SilasWaxter, 2019-04-09:
How is sensor calibration done now?  I rember waving the device in a figure eight for the magnetometer.  How about the accel and the gyro?

Do you think a machine learning algotrithm could be used instead of the senor fusion algorithm?  Would it be more effective?

What if the machine learning algorithm was more like a correction filter?  The sensor fusion algorithm would run off the raw data.  The output would be taken with the raw data and put into the machine learnging algorithm to refine the results.

Note:  My project is creating a device to measure the range of motion of a joint.  I know these already exist, but they are expensive, relatively innaccurate, and not user friendly.

[8 more replies]