kriswiner/MPU9250 #345: Data order for Madgwick filter

egen1um, 2019-02-05:
Hi Kris!

First of all I'd like to thank you for all the help you're giving here. Your Github issues page is definitely most useful resource on internet for IMU related questions :)

I've probably read all your answers, but somehow I still can't wrap my head around order and signs of data that you pass to the filter. Could you please explain it once more "for dummies" way?

[image]

For example, I have this same type of orientation for acc, gyro and magnetometer and I'm using NWU Madgwick filter. What would be the correct order and more importantly why?

Also, if you have any experience with Unity, how would you convert Madgwick quaternion to Unity quaternion?

kriswiner, 2019-02-05:
No idea about Unity.

The Madgwick and Mahony filters (and quaternions in general I believe) work
in a right-handed coordinate system. So the data have to be "provided" to
conform to this. Thus NED, ENU (the two most common orientation
conventions) or even NWU will all work. As long as the sensor data is
provided in a manner consistent with the chosen convention.

So first step, the user decides which edge of the sensor board will be
pointing to true North when the quaternions read 1 0 0 0. This is one of
two absolute references in the system (the other is gravity). Once the
board edge facing North is decided (yes, it is your choice!), then it is a
simply matter to find out which accel axis point North, no? Then West, then
Up. Then the filter should get the data as AN, AW, AU...same for the other
two sensors.

Then once you do this, and assuming you have chosen the sensor
configurations reasonably, calibrated the sensors sufficiently, are running
the fusioniterations sufficiently fast, you have taken into account the
local magnetic declination, etc you can check the behavior of your absolute
orientation engine. What do the quaternions say when the sensors are
pointed North (Yaw or Heading = 0)? Should be 1 0 0 0. Try turnin the board
90 degrees. What happens now? etc.

kriswiner, 2019-02-05:
> So if we take picture from previous post and use upper edge as North, then
> the order for NWU will be (ay, ax, az, gy, gx, gz, my, mx, mz)?
>
> As for calibration, what do you think about Magneto
> <http://sailboatinstruments.blogspot.com/2011/09/improved-magnetometer-calibration-part.html>?
> Is it good enough?

Did you look at the MPU9250 data sheet?

egen1um, 2019-02-05:
I'm sorry, it's not exactly about MPU9250, it's more of a general question, I didn't know where else to ask :)

kriswiner, 2019-02-05:
West is -x, no?

This isn't really that hard, or shouldn't be....;>

Do you know what right-handed coordinate system means? Look it up, google
is your frind in this case...

egen1um, 2019-02-05:
Yes, -x, my bad. Ok, I think I finally got it. It's just that I've seen some confusing posts here, but I guess that's because of specifics of physical orientation of different sensors on different boards (I mean this answer: 
> we need to feed into the madgwick function Ay, Ax, -Az, Gy, Gx, -Gz,
Mx, My, and Mz. But because gravity is by convention
// positive down, we need to invert the accel data, so we pass -Ay, -Ax,
Az, Gy, Gx, -Gz, Mx, My, and Mz into the Madgwick...

Anyway, now instead of answering same question for 1001 time you can  give link to this post :) Thanks again for your time and patience!

kriswiner, 2019-02-05:
All sensors are different, and user chooses which axis is North so there is
no one right answer for all people for all time. In tother words, the user
has to understand what he/she is doing!;>

palemao, 2021-02-21:
[image]

Looking at this illustration of MPU9250 given that I choose Y-axis to point North and choose to used the NED system. Would my input be madgwickquaternionupdate(ay, ax, az, gy, gx, gz, my, mx, mz)? 

Y=North X=East and Z= down?

kriswiner, 2021-02-21:
Which accel axis is painting North then?

palemao, 2021-02-22:
I have just seen that the axis of Mag ang accel & gyro are misaligned on the MPU9250 datasheet. 
[image]

so let me ask the question again with this new knowledge: 
[image]

Using NED system:
Given I point **x-axis** of my mag North, East=Y and Down is Z.

For ACCEL & GYRO: North = Y, EAST=X and Down = -Z

This is my thought process is this correct? 

Thanks again :)

kriswiner, 2021-02-22:
yes, so N axes are Ay, Gy, Mx, etc and the Madgiwck filter expects N, E, D
input...so filter(AN, AE, AD, GN, GE, GD, MN, ME, MD)