kriswiner/MPU9250 #346: Multiple MPU adress problem

VGiaglisis, 2019-02-07:
Hello again Kris,

I have mooved on with the project and am now trying to connect multiple MPU-9250 with a Multiplexer. I send the SCLK directly to the arduino and enable the SDA of each sensor through the multiplexer. This seems to work for 4-5. I added a 6th and i am getting I am 0xFF I should be 0x48 error only for the sixth sensor. I tried substituting it with amother 9250t sensor but nothing changed, so its probably not a problem of the MPU as a module. Do you have any ideas, as to why this could be happening?
Thank you in advance.

kriswiner, 2019-02-07:
Look at the I2C signals with an oscilloscope, your bus capacitance is
probably getting to large for your pullups. Possible solution is to reduce
the pullups to 4K7 or 2K or even 1 K from whatever they are now.

VGiaglisis, 2019-02-07:
Do you mean the pull-ups resistors for sclk and sda? I have 4.7K resistors diverging from each sclk to 5V as suggested in another schematic. Should I try 1K resistors if I want 9 MPU?

kriswiner, 2019-02-07:
Generally, one set of pullups is sufficient on a single I2C bus. If you
have 5 x 4.7 K on each line in parallel the effective resistance is < 1 K.
You might have too much resistance on SCK. If you only have 4K7 on SDA you
might have to little, etc...Use an O-scope to see...

VGiaglisis, 2019-02-08:
For 5 sensors, I have 5 4.7k for each of their sck and 5 4.7k for each of their sda. Should minimize the sck resistances to 1la single 4.7k or even a 1k and keep the sda resistances?
What am I looking for in the oscilloscope?

kriswiner, 2019-02-08:
" What am I looking for in the oscilloscope?"

square waves

duminduud, 2019-05-22:
I'm also having similar issues but the difference is magnetometer issue of giving FF is random some time 4-5 sensors work out of 7 but some time all fail but other sensors are working properly

kriswiner, 2019-05-22:
If some sensors work all of the time this is likely not a software issue
but the result of poor connections or using crappy breakout boards, etc,
meaning hardware.

If all fail some of the time, sounds like your I2C APIs need work...or
switch to a more reliable MCU.