kriswiner/MPU9250 #347: Inaccurate Yaw angle

blameitonworld, 2019-02-08:
Hello ,
I'm using  your code  to get raw data from the MPU9250. I'm also using several sensor fusion algorithm (including Madgwick and Mahony) on Matlab to get better performance than directly on the board. However, no matter the algorithm, my yaw angle is stable but inaccurate. If I move from 0 to 90 degrees, on paper I get around 44degrees of difference on the yaw angle. 
I calibrated  the magnetometer using  your method and the excel plot looks good. Also Roll and pitch angles  look good. 

1.what do you think is the issue with the accuracy ?
2. my calibrated  data  look like this at rest 
timeStamp  ax     ay        az          gx     gy     gz    mx  my  mz
 145.970 -85.57 -83.56 1002.32 -0.05 -0.30 -0.01 -212 1 606
 145.977 -84.53 -84.41 989.93 -0.07 -0.27 -0.02 -212 1 606
 145.983 -86.61 -81.60 984.50 -0.10 -0.31 -0.14 -212 1 606
 145.988 -86.98 -78.55 984.92 0.10 -0.36 -0.11 -212 1 606
 145.994 -89.60 -77.15 990.97 0.17 -0.36 -0.06 -212 1 606
 146.000 -89.11 -80.51 997.56 0.10 -0.40 -0.10 -212 1 606
 146.006 -86.43 -81.05 1004.94 0.05 -0.43 -0.09 -212 1 606
 146.010 -82.82 -78.61 997.74 -0.05 -0.27 -0.04 -212 1 606
 146.015 -84.35 -82.64 995.00 0.06 -0.34 -0.11 -212 1 606
 146.022 -81.91 -79.04 993.96 0.08 -0.33 -0.14 -212 1 606
 146.028 -83.50 -78.86 1001.04 0.02 -0.19 -0.05 -207 10 608
 146.034 -85.63 -80.51 1000.85 0.13 -0.27 -0.09 -207 10 608
 146.039 -87.77 -81.67 997.38 0.07 -0.10 -0.08 -207 10 608
 146.046 -87.22 -82.46 993.23 0.07 -0.24 -0.06 -207 10 608
 146.053 -89.66 -82.34 990.97 0.16 -0.21 -0.03 -207 10 608
 146.058 -87.52 -81.48 992.86 0.18 -0.24 -0.11 -207 10 608
 146.064 -85.75 -82.70 993.77 0.11 -0.25 -0.11 -207 10 608
 146.070 -87.71 -81.36 993.84 0.11 -0.25 -0.02 -207 10 608
 146.076 -84.90 -82.70 994.14 0.03 -0.33 -0.02 -207 10 608
 146.081 -88.26 -81.12 994.02 0.03 -0.31 0.00 -207 10 608
 146.085 -86.98 -85.14 996.34 0.00 -0.28 -0.02 -207 10 608
 146.090 -86.24 -79.90 1001.46 0.05 -0.19 -0.05 -207 10 608
 146.097 -87.46 -86.24 1000.12 -0.02 -0.33 -0.02 -207 10 608
 146.103 -87.46 -82.03 990.91 0.02 -0.32 0.02 -221 7 608
 146.108 -91.00 -81.48 986.08 0.09 -0.36 -0.05 -221 7 608
 146.114 -87.40 -80.26 989.62 0.13 -0.33 -0.08 -221 7 608
 146.120 -86.36 -80.57 995.54 0.15 -0.42 -0.13 -221 7 608
 146.126 -86.24 -80.81 998.66 0.08 -0.34 -0.11 -221 7 608
 146.131 -84.05 -81.73 991.46 0.02 -0.29 -0.05 -221 7 608
 146.137 -88.75 -81.73 990.97 -0.08 -0.34 -0.04 -221 7 608
 146.141 -89.29 -80.51 994.26 0.05 -0.36 -0.11 -221 7 608
 146.146 -85.45 -80.32 997.50 0.05 -0.32 -0.05 -221 7 608
 146.153 -86.43 -79.59 998.05 0.05 -0.34 -0.04 -221 7 608
 146.159 -86.61 -83.25 998.72 0.02 -0.25 -0.02 -221 7 608
 146.164 -84.72 -81.60 997.92 0.05 -0.21 -0.04 -221 7 608
 146.170 -84.59 -80.93 996.22 0.15 -0.27 0.01 -221 7 608
 146.176 -84.72 -83.31 991.09 0.08 -0.10 -0.11 -216 1 613
 146.182 -81.36 -79.41 1013.43 0.04 -0.17 -0.06 -216 1 613

Thank you.

kriswiner, 2019-02-08:
Magnetometer values don't look right. Mean Earth field is 500 milliGauss,
your is showing ~630 milliGauss. Accel values should show 0, 0, 1000 with
jitter at ~10 milligs, your are a lot higher. I suspect your sensors are
not properly calibrated.

If you flip a properly-calibrated sensor 180 degrees the Mz should be equal
and opposite in sign. Is it in your case?

blameitonworld, 2019-02-12:
I did the Calibration again and i get values in the appropriate range. When I flip it Mz is opposite in sign

blameitonworld, 2019-02-12:
I did the Calibration again and i get values in the appropriate range. When I flip it Mz is opposite in sign. However I still struggle with accelerometer jitter at rest

kriswiner, 2019-02-12:
The accel at rest with the sensor flat should be showing 0, 0, 1000 milligs
(if the offset biases are properly calculated and removed) with jitter of <
20 milligs. If the jitter is much larger than 20 milligs there is something
wrong with your sensor configuration or your sensor...most likely the
former.

blameitonworld, 2019-02-12:
what do you mean by sensor configuration  ? Scale ? 
I'm currently testing the sensor with your code Basic_AHRS function on arduino and I assumed calibrateMPU9250(gyroBias, accelBias); was doing the calibration and there was nothing more I needed to add to make it more accurate

kriswiner, 2019-02-12:
sample rate, bandwidth, full scale, interrupt vs. polling, etc

kriswiner, 2019-02-12:
" I assumed calibrateMPU9250(gyroBias, accelBias); was doing the
calibration and there was nothing more I needed to add to make it more
accurate"

Did you check the calibration? You still need to do this...

blameitonworld, 2019-02-12:
Oh okay !  SR is 200Hz  for  Gyro and Accel and 100Hz for Mag
Bandwidth is set at 44.8Hz and the Accel full scale is 2g. with this configuration i get logical but kinda random response : sometimes  it's almost 0 0 1000 and stable but the accurary changes everytime i reset the board

kriswiner, 2019-02-12:
Is the jitter of the accel ~< 20 millig when at rest?

blameitonworld, 2019-02-12:
Yes sometimes. When I reset the board multiple times at rest  I notice that Accel data is accurate at some point but at other times the jitter can go up to ~100 (still at rest)

kriswiner, 2019-02-12:
I would try another sensor breakout then, this is not normal behavior.

blameitonworld, 2019-02-12:
Alright, I'll try with a new sensor ! Thank You :)