kriswiner/MPU9250 #348: IMU Data Read

rafaybeg, 2019-02-15:
Hey, I am new to IMU and Arduinos and I am trying to get an Arduino to work with IMU with proper calibration. I using the code MPU9250_MS5637_AHRS_t3. I removed teh code for the pressure sensor and the display, however when I run the code, I don't get any data for. Everything shows up as 0. I am using an Arduino Uno with IMU 9250. I have attached the code I am using.
[image]
[MPU9250_MS5637_AHRS_t3.zip](https://github.com/kriswiner/MPU9250/files/2870462/MPU9250_MS5637_AHRS_t3.zip)

kriswiner, 2019-02-15:
Are you (properly) using the interrupt?

rafaybeg, 2019-02-15:
It worked, I was using the wrong pin. Thanks!

closed 2019-02-15 by rafaybeg