kriswiner/MPU9250 #350: Clarification and Guidance

rafaybeg, 2019-02-22:
Hey Kris, I am new to arduino and IMUs in general and learning as I go. I am using versions of you code, which I have adapted to take any number of IMUs. I am working on a wearable device which might will use 8-9 IMUs and a few other sensors. I am using ATMega 2560 right now and beginning to realize it might not be the best for me. Which MCU would you recommend? I also realize that i2c bus also has its limitations, so would it be recommended to split the sensors across two MCUs? All the IMUs will be within 1 ft of the MCU.
[MPU9250_MS5637_AHRS_t3.zip](https://github.com/kriswiner/MPU9250/files/2894953/MPU9250_MS5637_AHRS_t3.zip)
[Multiple_MPU9250.zip](https://github.com/kriswiner/MPU9250/files/2894954/Multiple_MPU9250.zip)

I know that higher fusion rates can reduce jitter in data and for this I need to increases the sensor sample rate as well. How do I do this? How do I determine my fusion rate or is it determined by this line? I am getting the calibration data from the code in MPU9250_MS5637_AHRS_t3.zip.

Serial.print("rate "); Serial.print(i); Serial.print(" : "); Serial.print((float)sumCount[i]/sum[i], 2); Serial.println(" Hz");

kriswiner, 2019-02-22:
I would recommend something like
<https://www.tindie.com/products/TleraCorp/dragonfly-stm32l47696-development-board/>
this since it is has an FPU, and more than one I2C bus, if necessary. The
fusion rate is set by the speed of the MCU, and/or the number of iterations
you do in the main loop for each new data set.

Generally, there is a trade between absolute orientation accuracy and power
consumption. MPU9250 is not a low power sensor anymore compared to the
LSM6DSM+LIS2MDL anyway. If you don't care about low average power usage,
run the gyro and accel at 1 kHz, the mag at 100 Hz, and the bandwidth at
20% or so of these values.

rafaybeg, 2019-02-22:
Hey, thank you for the quick reply.

I would have loved to use LSM6DSM+LIS2MDL, but size of the sensor matters in this application, so I am stuck using MPU 9250. I think I was able to change the sample rate for gyro/accel to 1kHz (It should be in the code I attached). How do I change the bandwidth in this code?

The power is not an issue in this application, so would getting rid of int be okay? Also, if I am to use the board you suggested, would the code I have still be compatible?

kriswiner, 2019-02-22:
Look in the MPU9250 init section to change the bandwidth. There will be
minor changes required whenever you switch MCUs. The MPU9250 is out of
production as of this month so this is not a viable long term solution.

rafaybeg, 2019-02-22:
Yes, agreed. For now we will work with this, but we'll adapt to newer sensors once they fit our size constraint.

rafaybeg, 2019-02-22:
Hello, am I doing this right to change the sample rate to 1kHz and bandwidth at 20%? Is there anything else I am suppose to change?
[image]
[image]
[image]

rafaybeg, 2019-02-22:
Hey, sorry I think I realized some of the mistakes I was making. Is this correct now?
[image]

kriswiner, 2019-02-22:
You are going to hve to debug your code yourself...

rafaybeg, 2019-02-22:
Sorry, don't want to ask you to debug it. Just want to know if thats where I am suppose to make the changes.

rafaybeg, 2019-02-27:
Hey, I just ordered the Dragonfly STM32L476/96 and had a question. Can I use any pin for interrupt and to toggle between high and low? I have 8 sensors, so I would need 8 interrupt pins and 8 pins to change the addresses on the IMU.

kriswiner, 2019-02-27:
You can use any GPIO as a TTL toggle.

You just have to be sure that the interrupt port numbers do not overlap. So
for example, you can't select both pin 2 (PB12) and pin 11 (PC12) as
interrupts since the 12 port designator can only be used once as an
interrupt. This should still leave you with plenty of options an you should
have not trouble meeting your need for eight interrupts.

rafaybeg, 2019-03-03:
Hey, so only the number at the end needs to be unique? whether it is PB, PC, PD or PA? For example pin 9 and 10 are PA2 and PD2, so I can't use them at the same time? or like pin 21 and pin 8?

kriswiner, 2019-03-03:
Yes, you want to avoid using PA3 and PB3 as interrupts, for example.

rafaybeg, 2019-03-19:
Hello,
I just received the dragonfly today and when I plug in the dragonfly to my pc the rgb light starts flashing and it doesn't get detected by my computer. Is there anything I can do to troubleshoot this kind of issue?

kriswiner, 2019-03-19:
Does the rgb led flash red, green, blue or red and green?

If the former, then Windows doesn't know which driver to use. So,

1) close Arduino IDE
2) plug USB into Dragonfly
3) press and hold boot button, press and release reset button, release boot
button
4) open Arduino IDE, should be no COM port in tools menu. Select Dragonfly
L476RE from tools menu, compile and load a simple blink program.
5) after flashing, Windows should assign a COM port.
6) If you get the 1, 2,, 3, 4, countdown instead, download and run Zadig,
list all options, select STM32 bootloader and click replace driver and try
5) again.

rafaybeg, 2019-03-19:
So I ordered the L496RG and the lights are blinking green, red, blue. i did the reset and the lights stopped blinking. Then I tried uploading a simple sketch and I got the count down. STM32 device is still not showing up in zadig list.

kriswiner, 2019-03-19:
List all options in Zadig, find STM32 boot loader, click replace driver

rafaybeg, 2019-03-19:
Yes, I selected List all options, but STM32 is not in the list. I have other things, but not STM32.

rafaybeg, 2019-03-19:
I have two boards, and they both are having the same issue.

kriswiner, 2019-03-19:
Has to be if the device is in boot mode, must be an STM32 boot loader
there. Look in your device manager, is there no STM32 boot loader there?

rafaybeg, 2019-03-19:
[image]
[image]

Not sure why it is not showing up. I have tried connecting it to different computers, but I get the same result.

kriswiner, 2019-03-19:
Is there a COM port showing in the Arduino IDE?

What does the Zadig display look like?

rafaybeg, 2019-03-20:
No COM port showing in Arduino IDE and Zadig has other devices, like keyboard and camera but not STM32.

kriswiner, 2019-03-20:
Check your USB cable; there is something wrong with your set up.

If you press reset does the Dragonflu start blinking again?

rafaybeg, 2019-03-23:
Hey, So i was able to get it to work. It was the usb cable that was causing issues. I have a question. Everytime I upload a sketch, it does this:

Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 70672

Download	[                         ]   0%            0 bytes
Download	[                         ]   0%            0 bytes
Download	[=                        ]   5%         4096 bytes
Download	[==                       ]   8%         6144 bytes
Download	[==                       ]  11%         8192 bytes
Download	[===                      ]  14%        10240 bytes
Download	[====                     ]  17%        12288 bytes
Download	[=====                    ]  20%        14336 bytes
Download	[======                   ]  26%        18432 bytes
Download	[=======                  ]  28%        20480 bytes
Download	[========                 ]  34%        24576 bytes
Download	[=========                ]  37%        26624 bytes
Download	[==========               ]  40%        28672 bytes
Download	[===========              ]  46%        32768 bytes
Download	[============             ]  49%        34816 bytes
Download	[=============            ]  52%        36864 bytes
Download	[=============            ]  55%        38912 bytes
Download	[==============           ]  57%        40960 bytes
Download	[===============          ]  60%        43008 bytes
Download	[===============          ]  63%        45056 bytes
Download	[================         ]  66%        47104 bytes
Download	[=================        ]  69%        49152 bytes
Download	[==================       ]  72%        51200 bytes
Download	[===================      ]  78%        55296 bytes
Download	[====================     ]  81%        57344 bytes
Download	[=====================    ]  84%        59392 bytes
Download	[======================   ]  89%        63488 bytes
Download	[=======================  ]  92%        65536 bytes
Download	[======================== ]  98%        69632 bytes
Download	[=========================] 100%        70672 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state

Is it suppose to do that and if not then how can stop it? Also is there a way to make the arduino code reset everytime I close the serial port?

kriswiner, 2019-03-23:
It is supposed to do that. I suppose you can turn this off in the IDE
preferences.

You mean cause a device reset when you close the serial port? Huh? Why
would you want to do this? I have no idea how...

rafaybeg, 2019-03-25:
Well I need it to shut off or reset when data is not needed. But otherwise its been working well. I was able to get the two i2cs working. I do have a question. I am running into some power issues. Can I connect a battery to it while it is connected to a computer via usb?

Also is there a way to may be save a csv file for the calibration data on 16gb flash drive? Do you have any example codes of something along this line?

kriswiner, 2019-03-25:
" Can I connect a battery to it while it is connected to a computer via
usb?" yes

Yes, there is also emulated EEPROM with example sketches in the examples.

You can use this sketch to write and reas files from the QSPI flash on the
Dragonfly. Just make sure you select QSPI flash in the tools menu.

https://github.com/kriswiner/Dragonfly/blob/master/SDIOCardReader/ReadWrite.ino

rafaybeg, 2019-03-28:
Hello, thank you for that example. I was able to save all the data to a file and load it at run time from the QSPI Flash. I am running into issues when I increase the sample rate. It works fine at 200Hz, but when I switch it to 1000Hz it stops sending data after the first second or so. I am not sure if its a power issue or something else I am not aware off.

kriswiner, 2019-03-28:
No, FS.h is limited in speed.

rafaybeg, 2019-03-28:
Yes, at run time it loads all the values to their variables, i.e. magBias, accelBias, gyroBias and magScale and just uses those. It is only runs once in the setup function. Also is there any resource where I can look into passthrough mode for the magnometer?

kriswiner, 2019-03-28:
Are you talking about the I2C bus at 1000 kHz or the QSPI read/write speeds?

You can run the EM7180 in passthrough mode and talk directly with the
magnetometer.

rafaybeg, 2019-03-28:
Sorry, i should have cleared that. I am talking about the I2C at 1000hz for the MPU-9250.

kriswiner, 2019-03-28:
The MPU9250 is not rated to support 1 MHz I2C bus speeds, take a look at
the data sheet. Might or might not work, but not with 10 K pullup resistors
most breakout boards use. Which breakout board are you using for the
MPU9250?

rafaybeg, 2019-03-28:
Right. i have the Sparkfun MPU 9250. I am running at the I2C at 400KHz. I get this issue when I increase the sample rate of the MPU9250 from 200Hz to 1000Hz for the accel and gyro. I get data for a second and then it stops working.

kriswiner, 2019-03-28:
No idea. The Sparkfun board is no designed well...