kriswiner/MPU9250 #353: Is it worth to use this MPU9250?

shomedas, 2019-03-19:
I need to know if the MPU9250 I am using **will give me stable AHRS**. I am new to IMUs and I ask your opinion about my MPU9250(I am using the blue board). **Also I need to know what type of calibration is needed?**(For accelerometer, gyro and magnetometer). i have gone through your calibration methods in your sketches(5637.ino) and I need to know whether I need any thing over them. **How do I reduce the gitter in accelerometer, gyroscope & magentometer.**

My connection:
0250---arduino uno
Vcc--5V
Gnd--0V
FSYNC--0V
NCS--3.3V
SDA -- A4
SCL -- A5
INT -- Pin2
Pull up between SDA, SCL to 3.3 is 5k

I recorded the accelerometer, gyroscope and magnetometer while keeping the IMU stationary on a table. The IMU was fixed to a solid cube and placed in three different orientations in three consecutive experiments.

1. YZ on table, X pointing down
2. XY on table, Z pointing up
3. XZ on table, Y pointing up

Below table shows the rms error & peak to peak of ax,ay,az, Gx, Gy,Gz, Mx,My,Mz for three orientations of the IMU. accel, gyro and mag readings are uncalibrated readings from the 5637.ino code. The units for acceleration, gyroscope and magneometer are g, dps & mGauss respectively.
The magnetometer data is captured at 100Hz mode.

|  Orientation1 |   ax   |   ay   |   az    |   gx   |   gy   |   gz   |   mx   |   my   |   mz   |
|      :---:          |  :---:  |  :---:  |  :---:  |  :---: |  :---:  |  :---:  |  :---:   |  :---:   |  :---:   |
|  peak to peak |  0.01 | 0.01  | 0.02  |0.39   |0.45   |0.39   |  68.33 |61.33  |64.36  |
|  RMS error    |4.2e-5|0.0027|0.0032|0.0484|0.0516|0.0463|8.145|7.6624|7.5714 |

|  Orientation2 |   ax   |   ay   |   az   |   gx   |   gy   |   gz   |   mx   |   my   |   mz   |
|      :---:          |  :---:  |  :---:  |  :---:  |  :---: |  :---:  |  :---:  |  :---:   |  :---:   |  :---:   |
|  peak to peak | 0.01  | 0.01   | 0.02  |0.79  |1.07   | 0.46  |  73.7  |  70.35 | 71.32 |
|  RMS error    |4.4e-4|4.6e-5 |0.019|0.049  |0.0512|0.046|7.9524| 7.8141| 7.4709|

|  Orientation3 |   ax   |   ay   |   az   |   gx   |   gy   |   gz   |   mx   |   my   |   mz   |
|      :---:          |  :---:  |  :---:  |  :---:  |  :---: |  :---:  |  :---:  |  :---:   |  :---:   |  :---:   |
|  peak to peak |  0.02 | 0.01  | 0.01  | 0.44 | 0.48   | 0.51  |  70.12 |80.2    | 72.3   |
|  RMS error    |1.13e-4|7.53e-4|0.005|0.0487|0.0524|0.0514|7.7502|8.206|7.8266|

Please find herewith attached the set of images for the case when z faces up and IMU is on table.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

How does the IMU data look overall in its stationary state? Is it reliable enough for AHRS

kriswiner, 2019-03-19:
I am confused by your data.

Shouldn't one of the accel axes be showing gravity at 1000 mg?

Shouldn't the mag magnitude be ~0.5 Gauss?

Shouldn;t the gyro read 0, 0, 0?

Why don't they?

shomedas, 2019-03-19:
This is uncalibrated data. I am showing the raw data here. I have done the calibrations. After calibration the yaw pitch roll values are not changing as expected. So I thought of checking the reliablity of the uncalibrated raw data to start with.

kriswiner, 2019-03-19:
The data is wrong.

shomedas, 2019-03-19:
I am not able to understand how the raw data is wrong. I have not edited it. Can you kindly elaborate?

kriswiner, 2019-03-19:
If the data is properly scaled, then:

 one of the accel axes be showing gravity at 1000 mg

the mag magnitude be ~0.5 Gauss

 the gyro read 0, 0, 0 dps

Do they? This is independent of calibration.

You are either not reading the registers properly or not converting
register bytes to signed integers properly or not scaling the signed
integers properly or all of the above.

So what you are calling the data is not.