kriswiner/MPU9250 #355: MPU9250 - Use in Magnetic environment

adunakhe, 2019-04-11:
Dear Kris,

   I am working on a project which involves the use of MPU9250 in a  magnetic environment. Before I state the issue, I will explain the background and set-up of the project for a clear idea.

**Setup:**
   So, I am designing a system with MPU9250 to be used in a hearing related research project. The sensor will be mounted on the head of the subject. The subject will be sitting inside the soundproof room surrounded by 24 speakers (height of speakers and subject's ears = 4ft). Out of these 24 speakers, 5 speakers can move in verticle plane to allow people to enter and leave the room. Refer to the image of the setup. The sensor will be used when all the speakers are turned on and 5 moving speakers up.  

**About the sensor:**
   We are using the library provided by you and the Arduino communicates to the monitoring station using XBee modules. We followed the magnetometer calibration process (sampling period ~2min). When we calibrated the sensor outside the room and when the 5 moving speakers were down (all speakers on) the calibration seems to generate proper results with moderately round shaped calibration plot. On the other hand (this is where we are facing the problem) when we calibrated the sensor with all the speakers up and turned on the calibration plot seems to distort. 
   In the first case, the sensor works as expected with proper yaw roll pitch angles, however in the second case, irrespective of the rotation of the sensor around the z-axis, the yaw angle remains unchanged.

   So, my question is that is there anything wrong with the calibration process that I am following or do I have to make any changes to the code in order to compensate for any magnetic field that may be present inside the room due to speakers?

   My other question is that, in order to avoid calibration each time we turn on the system, we are trying to save the biases of the magnetometer in the code itself, so is it enough to save only the mag_basises or do I need to consider saving the mag_scale parameter as well?  
 
Ps.: I have attached the image of speakers setup, calibration plots of three conditions 1. Case 1-
Outside the room, 2. Case 2 -  Inside the room with speakers on and up and 3. Case 3 -
 Inside the room with speakers on and down (5 speakers). 

[image]
[image]
[image]
[image]

@kriswiner @angkanal

kriswiner, 2019-04-11:
You will need proper calibration of all three sensors not just the mag. As
a minimum, I would use the more sophisticated calibration methods described
here <https://github.com/gregtomasch/EM7180_SENtral_Calibration> and store
the offset biases (hard iron) as well as the soft iron (3 x 3 matrix)
corrections for the mag as well as offset biases for accel and gyro.

The problem of having fixed magnetic fields (from the speakers) affect the
heading estimation is preventable but requires even more sophisticated
calibration and fusion methods. We would be happy to consult as a fee for
service to help.

adunakhe, 2019-04-14:
Hello Kris,
Thank you for the information. I'll take a look at the document.

In meanwhile I would like to add few more points as I observed data in the last few days. 

So, with the current setup, I am getting fusion rate of 150 Hz and I remember in one of the discussion you told that the fusion rate should be at least 1 KHz. 

- How the fusion rate affects the accuracy of the yaw roll and pitch angles?
- Can you please suggest some techniques to improve this rate? or any documentation regarding this issue?

I am using Atmega32u4 8MHz 3.3V mcu.

kriswiner, 2019-04-14:
You will need something like this
<https://www.tindie.com/products/TleraCorp/ladybug-stm32l432-development-board/>
or its bigger brothers to get fusion rate 5 - 10x the data sample rate. A 8
MHz AVR isn't going to do it.