kriswiner/MPU9250 #356: How to support your work?

zrecore, 2019-04-11:
Hi Kris, I'd like to support your work on the variety of motion sensors you have libraries for. What are some acceptable ways to do that?

- Alex

kriswiner, 2019-04-11:
Hi Alex,

Thank you and I appreciate your help!

Best way is to by my products on Tindie; I have two stores. This
<https://www.tindie.com/stores/onehorse/> one and this
<https://www.tindie.com/stores/TleraCorp/> one.

I will add another motion sensor suite soon that should replace the
MPU9250. It uses the ST 10 DoF sensor suite of the LSM6DSM + LIS2MDL +
LPS22HB similar to a breakout board I currently offer but unlike the
current offering, the new board is intended as a wearable and, more
importantly, to be ganged two or more per I2C bus. Here
<https://oshpark.com/shared_projects/hIizJ2nZ> is the design. I'll post the
appropriate sketches on github when I get the first production batch from
the fab. This will be a great solution for looking at joint angles, etc.

You can make use of the sketches and libraries and let me know how they can
be improved. Or better, improve them yourself and post for others to use.

Lastly, tell me a bit about how you are using motion sensors; maybe I can
help.

Kris

zrecore, 2019-04-11:
Hi Kris! Thank you for replying!  That new board is looking nice! I can't wait to see it in action!

I understand the MPU9250 has been EOL'd in favor of the ICM-20948, however I still need to support the MPU9250 libraries as I've got quite a few coins out in the wild (so to speak). As you are probably aware, the FreeIMU library has support for the MPU9250, but it isn't as up to date as your library... I would prefer to use your work where possible, hence this "issue" thread.

I've noticed quite a few folks have been using motion sensors for human motor control related studies, others for AR/VR controllers, and a few more as a novelty keepsake (I don't always hear back, haha)