kriswiner/MPU9250 #357: unable to read second IMU in dual IMU setup

shomedas, 2019-04-17:
I am using the dual IMU code for MPU9250 with an ESP32 board. I have commented out the RTC related code and change IMU idetification ids to 0x73.
The second IMU is not detected when I run the code.

I get error:

Scanning...
I2C device found at address 0x0C  !
I2C device found at address 0x68  !
done

MPU9250 9-axis motion sensor...
MPU9250_1 I AM 73 I should be 73
MPU9250_2 I AM FF I should be 73
Could not connect to MPU9250 1: 0x73
**Could not connect to MPU9250 2: 0xFF**

Both IMUs work fine when they are run individually.
I2C pull up is 10k for each board. I guess that the int_pin_cfg needs to set/toggled, but dont know where to implement it in the code.
My connection

MPU9250 -- ESP32
SDA -- 21
SCL -- 22
Vcc -- 3,3 V
Gnd -- Gnd
Fsyns -- Gnd
NCS -- 3.3V

Intpin1 =D2
Intpin1 =D4

I2c speed: checked at both 100Khz & 400 Khz. Both failed to detect second IMU.
I realise I need to set the int_pin_cfg but how i dont know to set it. there is an AD0 code in the single IMU version which sets the I2c adress of IMU  to 0x68 or ox69.

kriswiner, 2019-04-17:
Which code specifically are you using? A link please. The proper way to do
this is here <https://github.com/kriswiner/MPU9250/tree/master/Dual_MPU9250>

shomedas, 2019-04-17:
I have used the same code as the one you have suggested. I have removed the lines with RTC and changed 
  attachInterrupt(intPin2, myinthandler2, RISING);   TO 
  attachInterrupt(digitalPinToInterrupt(intPin2), myinthandler2, RISING);

I have changed the intpins to pin D2 & D4. I am using ESP32 NodeMCU

The code is as below for your reference

////////////////////////////////////////////////////////////////////////////////

/* 07/6/2017 Copyright Tlera Corporation
 *  
 *  Created by Kris Winer
 * 
 Demonstrate basic MPU-9250 functionality including parameterizing the register addresses, initializing the sensor, 
 getting properly scaled accelerometer, gyroscope, and magnetometer data out. 
 Addition of 9 DoF sensor fusion using open source Madgwick filter algorithm. 

 Sketch modified to read data from two MPU9250s, one at 0x68 and 0ne at 0x69, on the same I2C bus and get absolute orientation
 Sketch runs on the 3.3 V Ladybug STM32L432 Breakout Board.
 
 Library may be used freely and without limit with attribution.
*/
 
#include "Wire.h"   
#include "MPU9250.h"
//#include "RTC.h"

#define SerialDebug false   // set to true to get Serial output for debugging

// RTC set time using STM32L4 natve RTC class
/* Change these values to set the current initial time */

//const uint8_t seconds = 0;
//const uint8_t minutes = 15;
//const uint8_t hours = 17;

/* Change these values to set the current initial date */
//const uint8_t day = 4;
//const uint8_t month = 7;
//const uint8_t year = 17;

//uint8_t Seconds, Minutes, Hours, Day, Month, Year;

bool alarmFlag = false; // for RTC alarm interrupt

// MPU9250 Configuration
// Specify sensor full scale
/* Choices are:
 *  Gscale: GFS_250 == 250 dps, GFS_500 DPS == 500 dps, GFS_1000 == 1000 dps, and GFS_2000DPS == 2000 degrees per second gyro full scale
 *  Ascale: AFS_2G == 2 g, AFS_4G == 4 g, AFS_8G == 8 g, and AFS_16G == 16 g accelerometer full scale
 *  Mscale: MFS_14BITS == 0.6 mG per LSB and MFS_16BITS == 0.15 mG per LSB
 *  Mmode: Mmode == M_8Hz for 8 Hz data rate or Mmode = M_100Hz for 100 Hz data rate
 *  (1 + sampleRate) is a simple divisor of the fundamental 1000 kHz rate of the gyro and accel, so 
 *  sampleRate = 0x00 means 1 kHz sample rate for both accel and gyro, 0x04 means 200 Hz, etc.
 */
uint8_t Gscale = GFS_250DPS, Ascale = AFS_2G, Mscale = MFS_16BITS, Mmode = M_100Hz, sampleRate = 0x04;         
float aRes, gRes, mRes;      // scale resolutions per LSB for the sensors
float motion = 0; // check on linear acceleration to determine motion
// global constants for 9 DoF fusion and AHRS (Attitude and Heading Reference System)
float pi = 3.141592653589793238462643383279502884f;
float GyroMeasError = pi * (40.0f / 180.0f);   // gyroscope measurement error in rads/s (start at 40 deg/s)
float GyroMeasDrift = pi * (0.0f  / 180.0f);   // gyroscope measurement drift in rad/s/s (start at 0.0 deg/s/s)
float beta = sqrtf(3.0f / 4.0f) * GyroMeasError;   // compute beta
float zeta = sqrtf(3.0f / 4.0f) * GyroMeasDrift;   // compute zeta, the other free parameter in the Madgwick scheme usually set to a small or zero value
bool wakeup;

// Pin definitions
int  intPin1 = 2;  //  MPU9250 1 interrupt
int  intPin2 = 4;  //  MPU9250 2 interrupt
[... 411 lines cut ...]
    Serial.print(yaw2, 2); Serial.print(", "); Serial.print(pitch2, 2); Serial.print(", "); Serial.println(roll2, 2);

} /* end of alarm handling */

}

//===================================================================================================================
//====== Set of useful functions
//===================================================================================================================

void myinthandler1()
{
  intFlag1 = true;
}

void myinthandler2()
{
  intFlag2 = true;
}

void alarmMatch()
{
  alarmFlag = true;
}
///////////////////////////////////////////////////////////////////////////////

kriswiner, 2019-04-17:
Please do not post code here, completely useless.

You seem to have a problem even seeing the second MPU9255 in the I2C scan,
are you sure you have set one address pin to HIGH with the other LOW?

shomedas, 2019-04-17:
No. I havent changed anything in the code other than the RTC part. I understand that one has to be at 0x68 another device at 0x69. How do I do it?
Do I need to add something like

#define ADO 0
#if ADO
#define MPU9250_ADDRESS 0x69  // Device address when ADO = 1
#define AK8963_ADDRESS 0x0C   //  Address of magnetometer
#define MS5637_ADDRESS 0x76   // Address of altimeter
#else
#define MPU9250_ADDRESS 0x68  // Device address when ADO = 0
#define AK8963_ADDRESS 0x0C   //  Address of magnetometer
#define MS5637_ADDRESS 0x76   // Address of altimeter
#endif

shomedas, 2019-04-17:
I realised that I had to connect hardware pins AD0 to gnd and VCC for the two boards resp.
AK8963 2 mag biases (mG)

But now the code doesnot execute the main loop. It is stuck at:
49.48
0.00
0.00
AK8963 2 mag scale (mG)
2.40
1.10
0.60

kriswiner, 2019-04-17:
And what do you expect the code to do after that?

Do you have the interrupts set/configured properly?

Do you have the print statements configured properly?

kriswiner, 2019-04-17:
You are using alarmFlag as a print test and this is never set to true...

shomedas, 2019-04-17:
Yah I removed that alarmflag clause. The code is now running. However the IMU data is changing only for one IMU, **The other IMU readings dont change even if moved around.**

kriswiner, 2019-04-17:
Sounds like another connection problem.

shomedas, 2019-04-17:
The connections are as below:

MPU9250 -- ESP32
SDA -- 21
SCL -- 22
Vcc -- 3,3 V
Gnd -- Gnd
Fsyns -- Gnd
NCS -- 3.3V

AD0 to Vcc(IMU1)
AD0 to Gnd(IMU2)
Intpin1 =D2
Intpin1 =D4

kriswiner, 2019-04-17:
Maybe your interrupts aren't configured properly, who knows?

shomedas, 2019-04-17:
awesome cris. Exactly your guess. Wrong pin connection for one of the interrupts. 
Cant thank you enough for your help.