kriswiner/MPU9250 #360: Yaw changing with roll

Kartha007, 2019-04-23:
Hi Kris,
	This will be a long story, I have been using the famous InvenSense sensor MPU950 for the past 4 months.  It’s one of the Industry leading sensor.
 In the beginning, I simply took the accelerometer values and perform some trigonometry to get the PITCH and ROLL, and used the magnetometer along with accelerometer values and performed the tilt compensation algorithm to get a pretty good YAW, yup things are fine, but am not satisfied with the performance. After  some research I found lot of algorithms around, like Madgwick(*8), Mahony, Kelman, extended kelman, complementary filter algorithm etc.  As I’m not at all good in math I just gone with the complementary filter algorithm, which I understood much better. After few days I came to know that (*** REAL PROBLEM ****) whenever I change the ROLL OR PITCH my YAW angle is getting changed in a nasty manner, which I can’t accept. After several searches I came to know about the gimbal lock / singularity (*1) when dealing with Euler angles. So I decided to go with the Madgwick algorithm even though I don’t know what is exactly happening inside that. I took this (*2) decision because when I read about this algorithm I came to know that the output is in Quaternions (afraid of that). The good thing about Quaternions I believe is, they don’t exhibits Singularity as in Euler angles. So I assumes that the YAW PITCH and ROLL calculated from (*3) the quaternions using some standard formulas from Wiki won’t have singularity, I don’t know whether this is true. Following are the setting I have done prior to the running of madgwick algorithm.

(Controller used PIC32MZ series, MIPS: 200 MHz, have FPUs, caches etc.)

Here I will share the sensor configurations and other things I have done prior to the algorithm.

1.	Sets the corresponding bits of the power management 1 register for - Automatically select 
        the clock either PLL or 20MHZ internal clock
2.	Then disabled the sleep of MPU9250.
3.	Configured the accelerometer for 1 kHz sampling rate with a band width of 5 Hz (*4), and a 
        full scale range of 4G.
4.	Configured the gyroscope for 1 KHz bandwidth by setting the FCHOICE to 0 and a 
        bandwidth of 5 Hz, the full scale rage was 250 DPS.
5.	After that I waited for almost 1 second and then took 6000 samples from gyroscope (after 
        each reading a delay of 1002 uS is given to ensure no same data where taken) and 
        averaged it, then loaded it to the offset registers of gyroscope as given in the data sheet. 
        There by I removed the offset of gyroscope up to an extent.
6.	I enabled the pass through mode for accessing the magnetometer (AK8963C) using I2C 
        module.
7.	Then configured the magneto for 100Hz mode (*5) and 16 bit resolution.
8.	Then I collected the fuse values from the magneto and stored in a global variable for 
        multiplying it with the readings as said in the datasheet (3 fuse values are there, each for x 
        y and z, multiplied x fuse value with each reading from x before using it for the sensor 
        fusion. And so on).

       Now I have done all my setting with the MPU9250. So after that I done the following things 
       once:

1.	Calibrated the accelerometer to get +1G and -1G, when pointed the axis perpendicular to 
        the ground, by simply adding offset values. This offset values where stored in the memory 
        and the ACC data is offset properly before using in the algorithm.
2.	Gyro is already offset calibrated as mentioned above.
3.	Calibrated the magnetometer properly and I got a perfect circle centered at 0,0 when 
        plotted X VS y and x VS z in excel. And used the calibration constants (mx + c format, hard 
         iron and soft iron compensation) every time I use the magneto reading. (**** I think there 
        is a problem with data from AK8963C, the X axis of AK8963C is aligned with the Y axis of 
        accelerometer, so I flipped the data every time, means X data of magnetometer is taken 
        from Y register and Y data is taken from X register ****).

Now the time for data collection and Algorithm iteration.

After reading one of your replies (https://github.com/kriswiner/MPU9250/issues/23) to ReyBreslin in GitHub I came to know that the algorithm should run as faster as possible. So I configured 2 timers one for 1mS inside which data is collected from all sensors, convert them to their basic units (ACC reading in G, and GYRO readings in Radian per second, and no conversions for magnetometer data)and then 9 filters (LPF)where used to filter the 9 data to make it smoother, then store it in a buffer.
 The next timer is running at 100 uS inside which the algorithm is called (10000 Hz) speed (*6). I confirm that the data to the algorithm where not corrupted due to any race condition.(Data from sensors are stored in a double buffered structure).
The algorithm and data collection is working perfectly and I am getting YAW PITCH AND ROLL correctly (Thanks to all those who worked on this beautiful piece of code.)

Now I will come to my real problem,

1.	I believe that the YAW PITCH and ROLL angles calculated from the Quaternion from 
        madgwick algorithm won’t present any singularity (By the word Singularity, I mean that 
        whenever the PITCH or ROLL angles increased beyond a certain limit the yaw angle 
        changes. For example, when ROLL or PITCH is 0 degrees no problem with YAW it works 
        great, but if ROLL increases beyond +-25(for example) degrees the YAW began to change, 
        same with pitch), is that TRUE?
2.	In my requirement the pitch angle won’t go beyond 5 degrees so am not bothering about 
        that, but the roll angle can vary from 0 to 360 degrees. As I said above I’m not getting a 
        stable yaw angle. Please refer to the attached file “Atti Roll 0 - 90 Yaw change.txt”, the data 
        in that file is YAW PITCH and ROLL calculated using the quaternion from madgwick 
        algorithm(*7). As you can see when roll is changed from 0 to 270 degree (90 degree 
        change) YAW is gone somewhere. Are u also getting the same result Kris?
3.	Can you please tell me if I’m missing something?  Please help me with this issue as this is 
        haunting me.

Below are the *references:
*1 -> I don’t know whether  gimbal  lock and singularity is same or not. 
*2 ->https://www.youtube.com/watch?v=0VAc_G79POE, after seeing this video I decided to 
        switch to madgwick algorithm.  In this video the UAV with quaternion based stabilization is 
        mind blowing, so I think they are calculating yaw pitch roll from the quaternion and 
        applying it to the PID as set points, please correct  me if I’m wrong.
3* -> Will the YAW PITCH AND ROLL calculated from quaternions exhibit singularity? Please 
         help me with this question.
4* -> I don’t know which bandwidth I have to choose, am also using an LPF in my code for every 
         set of data (9 filters for 9 data).
*5 ->which one I have to use 100 Hz or 8 Hz mode.
*6 -> is this speed good. (I also change the symbolic constant “sampleFreq” inside the 
         madgwick file to 10000, since the algorithm is running at 10000 Hz speed.)
*7 ->I used this function to get YAW PITCH SND ROLL from quaternion, please suggest if better 
        equations are there.
voidGetAttitude(Quaternions q, Attitude *result)
{
result -> Roll = atan2(q.y * q.z + q.w * q.x, 0.5f - (q.x * q.x + q.y * q.y)); 
result -> Pitch = asin(2.0f * (q.x * q.z - q.w * q.y));                         
result -> Yaw = atan2(q.x * q.y + q.w * q.z, 0.5f - (q.y * q.y + q.z * q.z)); 
}
 *8 -> I’m using the madgwick algorithm code from these site (also attached)( the data is send 
          to the filter as said in the function itself ->  mx r/s, my r/s, mz r/s, ax G, , ay G, az G, mx, my, mz) : https://x-io.co.uk/open-source-imu-and-ahrs-algorithms/
          By clicking the c code (bottom of the page) I got a source and header file which is used in 
          my controller, the only change I made is changed the sample rate to 10000.00f.

Thanks In advance.

[madgwick_algorithm_c.zip](https://github.com/kriswiner/MPU9250/files/3108702/madgwick_algorithm_c.zip)

kriswiner, 2019-04-23:
" the data is send
to the filter as said in the function itself -> mx r/s, my r/s, mz r/s, ax
G, , ay G, az G, mx, my, mz"

This is wrong. The sensor order should be NED.

Kartha007, 2019-04-24:
Hi Kris, Thank you very much for your fast reply. After seeing your mail, I changed two things in my coed
1. I replaced my current algorithm code to your algorithm which I got from your repository quaternionFilters.ino. 
2. Then I changed the data input sequence like this : 
MadgwickQuaternionUpdate(-ax, ay, az, gx, -gy, -gz, mx, -my, mz).

After these two things the algorithm start working correctly (Thanks a lot, u helped me a lot). I got stable yaw angle, while varying the roll from 0 to 360 degree.

Few more doubts, I am still getting 3 to 4 degree error in my yaw angle when roll angle is changed from 0 to 360 degree, is that usual, or do you suspect any thing wrong, please make a suggestion. Next things is when pitch is increased beyond 45 degrees yaw is going some where, do u also have the same response,(Even though I'm not bothered about pitch since in my requirement pitch wont go beyond 20 degree) I am very curious about that, if no please tell me some way to correct it. Once again thanks a lot for your help.

kriswiner, 2019-04-24:
This MadgwickQuaternionUpdate(-ax, ay, az, gx, -gy, -gz, mx, -my, mz).
can't be right for the MPU9250. Do you know why?

The sensors need to be fed into the filter in NED or ENU or some other
conventional order. accel/gyro axes are not aligned with the mag axes, so
above cannot be correct.

So aN, aE, aD, gN, gE, gD, mN, gE, gD or you can use ENU (any right-handed
coordinate convention).

Kartha007, 2019-04-24:
Hi Kris,
am afraid I given the sensor data as shown here:
MadgwickQuaternionUpdate(-ax, ay, az, gx, -gy, -gz, my, -mx, mz).
But you have seen it as MadgwickQuaternionUpdate(-ax, ay, az, gx, -gy, -gz, mx, -my, mz), I think it could be a technical error.
Any way I will try the data sequence you have suggested (aN, aE, aD, gN, gE, gD, mN, **gE, gD**)( did u mean  mE, mD) and will update my result. 
One more doubt, why my yaw angle changes when pitch angle go beyond 45 degrees? Is this singularity,
any effective way to get rid of this?
Thank you so much.

kriswiner, 2019-04-24:
This MadgwickQuaternionUpdate(-ax, ay, az, gx, -gy, -gz, my, -mx, mz).could
be right. You need to check for your board configuration.

If you have the right sensor order and adequate calibration, the absolute
orientation should be correct for all pitch and roll values until you get
quite close to the poles.

Kartha007, 2019-04-25:
Hi kris,
According to this input sequence (MadgwickQuaternionUpdate(-ax, ay, az, gx, -gy, -gz, my, -mx, mz)) every thing works fine. The only error is, my yaw angle is changing rapidly when pitch angle is increased beyond +-45 degrees, and as you said this could be the lack of proper calibration, not because of singularity. Please verify the procedure I followed for calibrating accelerometer and magnetometer:

Accelerometer:

1. I just printed the values from accelerometer (x,y,x) inside a while loop with a delay of 10 mS.
2. While printing I logged the values by setting a log file.
3. Then I pointed the x axis perpendicular to the ground as good as I can, and hold it there for few seconds. (since the earth offers 1G acceleration I should get 8192 (since accelerometer full scale range is 4G)). Same procedure is done for -x, y, -y, z and -z.

4. After that I took the data to excel and find the max and min of each axis, then calculate the difference of max to 8192 and min to -8192 of each axis, then properly offsetted them from the raw data before converting to G values. 

Given below is the code

void Mpu9250OffsetGValues(FloatVect *Gvalue)
{
    if(Gvalue -> x > 0.0) Gvalue -> x -= 128.0;       // since + and - axis have different offsets.
    else  Gvalue -> x -= 152.0;
    if(Gvalue -> y > 0.0) Gvalue -> y -= 108.0;
    else  Gvalue -> y -= 142.0;
    if(Gvalue -> z > 0.0) Gvalue -> z += 152.0;
    else  Gvalue -> z += 288.0;
}

 here Gvalue is pointer to a structure in which row acceleration data is stored. Is this approach good? or do I need to calibrate the accelerometer in any other way, like the way magnetometer is calibrated.

Magnetometer:

1. printed the data from magneto(x,y,z) inside a while loop with a delay of 10mS.
2. While printing I logged the values by setting a log file.
4. Then moved the sensor in figure 8 shape.

3.  Using excel I calculated the hard iron and soft iron values, and properly offset them. I am attaching this excel file so u can verify the procedures I have gone through,
[AK8963C calib.xlsx](https://github.com/kriswiner/MPU9250/files/3115506/AK8963C.calib.xlsx)

the functions I used for calibration are as given below,

void HardIronCompensateAk8963(FloatVect *MagData) -> First called this function
{
    MagData -> x -= AK8963_X_OFFSET; 
    MagData -> y -= AK8963_X_OFFSET; 
    MagData -> z -= AK8963_X_OFFSET; 
}

void SoftIronCompensateAk8963(FloatVect *MagData) -> then called this function
{
    MagData -> x *= AK8963_X_SOFT_FACTOR; 
    MagData -> y *= AK8963_Y_SOFT_FACTOR; 
    MagData -> z *= AK8963_Z_SOFT_FACTOR; 
}

here MagData is the pointer to a structure in which data from magnetometer is stored.

So can you please notify me if I am doing something wrong. 

Thank you very much for all your support.

kriswiner, 2019-04-25:
Seems reasonable. How do you check calibration?. Can you verify your
Madgwick function is receiving sensor data in NED configuration or are you
just guessing?

Kartha007, 2019-04-26:
Hi Kris,

. How do you check calibration?. , Yeah, after calibration I just placed the accelerometer x axis perpendicular to the ground and I got a value very close to 8192 (FS is 4G), and when placed -x axis perpendicular to ground I got values very close to -8192, same procedure for all axis  this is how I test the calibration of accelerometer.
And checked the mag calibration by plotting X vs Y and X vs Z, and check whether the plot looks like a circle centered at 0,0. And I also got that pretty well.

One more thing, are you getting yaw change only when pitch approaches close to +-90, or whether it changes beyond +-45 degrees. 

I am getting a yaw change of 2 to 3 degrees when roll is changed 0 to 360 degree, is this normal Kris???

Thank you very much for your help.

kriswiner, 2019-04-26:
"I
* am getting a yaw change of 2 to 3 degrees when roll is changed 0 to 360
degree" Just what I would expect, yaw is accurate to +/- 4 degree rms with
this method.*

Kartha007, 2019-04-26:
Thanks Kris, can you please tell me whether you are  getting yaw change only when pitch approaches close to +-90, or whether it changes beyond +-45 degrees.

kriswiner, 2019-04-26:
Only near 90 degrees. This is the problem of finding the direction of South
when you are standing on the North pole.

Kartha007, 2019-04-26:
Hi Kris, Now my problems where somewhat cleared. Thank you for all your help and support, and I will surely try to upgrade the calibration and try to come out of singularity issue. Thanks a lot.

kriswiner, 2019-04-26:
We will have a calibration tool soon for sale on Tindie that should
significantly help you.

Kartha007, 2019-04-27:
Hi Kris,
Am very much happy to hear that you are going to launch a calibration setup for motion sensors. can I use it for calibrating all accelerometer, gyroscope and magnetometer? Once it is launched please let me know.

Actually I made a a calibration setup for calibrating my magnetometer. I made a gimbal structure using wood(to remove any magnetic interference) and stainless steel rods, which is capable of rotating independently in all axis around 360 degrees. But I got a mistake, I just used stepper motors for rotating the platforms which affects the calibration of the sensor, even though the motors are pretty far from the sensor. Any way I took a break on that project.

All the best for the calibration setup and once again thanks for your help.