kriswiner/MPU9250 #361: Fix the IMU relative frame respect to the absolute of the earth

cwarrior777, 2019-04-24:
Hello Kris,
First of all thank you for this amazing code!
After dealing with the calibration of the magnetometer, I have encountered the drawback of the absolute frame on which the IMU is referenced.
I am using the program MPU9250_MS5611_BasicAHRS_t3.ino, but I can not find in what point of the code you fix what is the relative frame with respect to the absolute picture.
I ask you this since I want to save an unique relative frame, because every time I initialize the serial monitor with my IMU in a different position it sets that initial position as absolute affecting the true measurement I'm looking for, I really do not know at what point and how can I save the relative frame from the desired position so that my sensor already starts referencing.

Regards!

kriswiner, 2019-04-24:
Yeah, don't understand this comment since the whole point of 9 DoF sensor
fusion is absolute orientation estimation. Absolute means wrt to gravity
and true north. The frame if done properly is the global Earth frame so
that the heading,pitch, and roll always read the same when the sensor is in
the same orientation.

I suspect one or more of three common problems: 1) not running the sensor
fusion fast enough, 2) no properly calibrating the sensors, 3) not feeding
the sensor fusion the data in the proper order, or all three.

cwarrior777, 2019-04-24:
Hi Kris,
First of all, thanks for answering.
I think it does not explain me well, what I'm looking for is to be able to start the sensor in any position, but with the same one already marking the degrees with respect to a position already fixed previously, I think I have solved it, it comes by the side of manually loading the data of biases and offset in the records of accelerometer and gyroscope, is this true ?.

Regarding your recommendations, it may be true that I have to improve my MCU, since I am using an Arduino Mega to perform all operations with quaternions.
When you say that the sensor may not be well calibrated, I noticed that Mz is close to -Mz and I think it's acceptable, but even so I'm trying to get a good calibration when doing the 8-way, you could advise me some way in which can perform a good figure 8 to achieve excellent calibration? (through photos or videos).

Regards!