kriswiner/MPU9250 #362: Regsiter setp for dual MPU9250 on common i2c bus

shomedas, 2019-04-27:
I am trying to get two MPU9255 to work with an ESP32 on a single I2C bus. I use your code at [https://github.com/kriswiner/MPU9250/tree/master/Dual_MPU9250](url). 

The code works fine only if I comment two lines reading the interrupt in the main loop 
 if(intFlag1 == true) {   ...}
 if(intFlag2 == true) {   ...}

Otherwise either one or both MPU readings will be stuck and condition(intFlag == true ) is always false for one or both of the IMUs. This implies that interrupt reading is not set properly.
I realize that two registers need to be set. INT_PIN_CFG and INT_ENABLE. In your code it is 

writeByte(MPUnum, INT_PIN_CFG, 0x10);  // INT is 50 microsecond pulse and any read to clear  
writeByte(MPUnum, INT_ENABLE, 0x01); // Enable data ready (bit 0) interrupt

What are the correct values for INT_PIN_CFG & INT_ENABLE for the dual MPU9250 on a single I2C bus

My connection:

MPU9255 -- ESP32
SDA -- 21
SCL -- 22
Vcc -- 3.3 V
Gnd -- Gnd
Fsyns -- Gnd
NCS -- 3.3V

AD0 to Vcc(IMU1)
AD0 to Gnd(IMU2)
Intpin1 =D2
Intpin1 =D4

kriswiner, 2019-04-27:
Isn't this the same problem you had before, I thought you fixed this?

Try using different ESP32 pins as interrupts, and make sure to define them
in the sketch.

shomedas, 2019-04-27:
I could read both the IMUs for a while. So i thought it was sorted. After more tests i realized that some times the readings for one or both MPUs are frozen. On checking I found that they were not executing the  parts where the interrupt is used in the if condition.

which value should i set to INT_PIN_CFG : 0x10, or 0x12(bypass enabled) or 0x02
Also do i need to do any write to int_status?
should i check the interrupt waveforms with CRO?

kriswiner, 2019-04-27:
Leave the settings as they are in the github sketch. You need to select
some other interrupts beside pin 2. Make sure you are defining whatever
pins you use in the sketch.

shomedas, 2019-04-27:
trying all pin combinations now. another related problem is that for every time i run the program and connect to serial terminal of ardiuno sdk to check data, i need to do a power reset. 

else i get

MPU9250 9-axis motion sensor...
MPU9250_1 I AM FF I should be 73
MPU9250_2 I AM FF I should be 73
Could not connect to MPU9250 1: 0xFF
Could not connect to MPU9250 2: 0xFF

do i need to add extra pull ups. i have 10k pull up on each board(So 5k in parallel). My cable length is less than 50cm with 50pf/metre.I use Flat ribbon cable.

kriswiner, 2019-04-27:
No idea.

shomedas, 2019-04-27:
What if read the data without using the interrupts? Will the accuracy of the data be lesser. Would it be corrupted by noise? what problems can i face?

kriswiner, 2019-04-27:
None, just wasting power, but with the ESP32 who cares.

shomedas, 2019-04-28:
when i do a dual fusion i get a rate of only 250hz for each IMU. What do you think about the fusion rate? Is 250Hz pretty low to get good results for AHRS?

kriswiner, 2019-04-28:
Too low. Should be 5 - 10 x your sample rate. This is why I use an STM32L4
like this
<https://www.tindie.com/products/TleraCorp/dragonfly-stm32l47696-development-board/>
one.

shomedas, 2019-04-28:
stm32 seems to 80Mhz while the ESP32 is 240Mhz. Isnt ESP32 supposed to be faster then?

kriswiner, 2019-04-28:
I have seen much faster fusion times that 250 Hz using the ESP32 from a
single sensor board, yes. But in your case, you are probably spending a lot
of time in I2C transactions because the I2C peripheral software is, to use
the technical term, "crappy". So it doesn't matter how fast your MCU clock
speed is if you are standing around waiting most of the time for I2C
transactions to complete. There might be other problems as well, I don't
know exactly what you are doing. I do know that I am able to run using the
Dragonfly with fusion rates of several kHz with two of these sensor boards.

shomedas, 2019-04-28:
I have seen 1khz on esp32 for a single mpu. Do you mean that reading two I2c devices on the same bus is causing the delay or is it the esp32 I2c driver that's crappy.  Thanks for all of your detailed replies Kris.

kriswiner, 2019-04-28:
" Do you mean that reading two I2c devices on the same bus is causing the
delay or is it the esp32 I2c driver that's crappy" yes