kriswiner/MPU9250 #367: Accessing MPU9250's magnetometer via SPI

Qammar-25, 2019-05-10:
Hi,

I am using MPU9250 to get accelerometer and gyroscope readings via SPI. However, i am unable to sort out how can i access the magnetometer through SPI? Any help regarding the issue will be appreciated.

kriswiner, 2019-05-10:
You could treat the AK8963 as an I2C slave to the MPU9250 and read mag data
directly from MPU9250 registers with SPI like in this
<https://github.com/kriswiner/MPU9250/tree/master/AK8963_as_slave> example.
Not here I am using I2C to read MPU9250 registers but you can replace this
with SPI and it should work just fine.

Qammar-25, 2019-05-13:
Thank you Kris. I successfully read the Who Am I register by making AK8963 as an I2C slave (SLAVE 4) of MPU9250. However, now i want to read magnetometer values for which i want to configure my CNTL1 register for continuous measurement mode 2. I am trying to write to this register but when i read it back it shows 0x00. Why i cannot write to this register?

kriswiner, 2019-05-13:
No idea. Are you using the MPU9250 methods to write to the slave sensor via
the MPU9250 registers?

Qammar-25, 2019-05-14:
Yes

kriswiner, 2019-05-14:
You should be able to write and read back register data, no idea why it
doesn't work.

Qammar-25, 2019-05-14:
The datasheet says it is a read-only register.

kriswiner, 2019-05-14:
Well, then of course you can't write to it.

Qammar-25, 2019-05-14:
How can i set the continuous measurement mode?

kriswiner, 2019-05-14:
Most of the AK8963C registers can be written to. You have to write the
correct configuration to the appropriate register as in the sketch I linked
to.

kriswiner, 2019-05-14:
Like this:

void MPU9250::initAK8963Slave(uint8_t Mscale, uint8_t Mmode, float *
magCalibration)
{
   // First extract the factory calibration for each magnetometer axis
   uint8_t rawData[3];  // x/y/z gyro calibration data stored here

   writeByte(MPU9250_ADDRESS, I2C_SLV0_ADDR, AK8963_ADDRESS);           //
Set the I2C slave address of AK8963 and set for write.
   writeByte(MPU9250_ADDRESS, I2C_SLV0_REG, AK8963_CNTL2);              //
I2C slave 0 register address from where to begin data transfer
   writeByte(MPU9250_ADDRESS, I2C_SLV0_DO, 0x01);                       //
Reset AK8963
   writeByte(MPU9250_ADDRESS, I2C_SLV0_CTRL, 0x81);                     //
Enable I2C and write 1 byte

   writeByte(MPU9250_ADDRESS, I2C_SLV0_ADDR, AK8963_ADDRESS);           //
Set the I2C slave address of AK8963 and set for write.
   writeByte(MPU9250_ADDRESS, I2C_SLV0_REG, AK8963_CNTL);               //
I2C slave 0 register address from where to begin data transfer
   writeByte(MPU9250_ADDRESS, I2C_SLV0_DO, 0x00);                       //
Power down magnetometer
   writeByte(MPU9250_ADDRESS, I2C_SLV0_CTRL, 0x81);                     //
Enable I2C and write 1 byte
   delay(50);
   writeByte(MPU9250_ADDRESS, I2C_SLV0_ADDR, AK8963_ADDRESS);           //
Set the I2C slave address of AK8963 and set for write.
   writeByte(MPU9250_ADDRESS, I2C_SLV0_REG, AK8963_CNTL);               //
I2C slave 0 register address from where to begin data transfer
   writeByte(MPU9250_ADDRESS, I2C_SLV0_DO, 0x0F);                       //
Enter fuze mode
   writeByte(MPU9250_ADDRESS, I2C_SLV0_CTRL, 0x81);                     //
Enable I2C and write 1 byte
   delay(50);

   writeByte(MPU9250_ADDRESS, I2C_SLV0_ADDR, AK8963_ADDRESS | 0x80);    //
Set the I2C slave address of AK8963 and set for read.
   writeByte(MPU9250_ADDRESS, I2C_SLV0_REG, AK8963_ASAX);               //
I2C slave 0 register address from where to begin data transfer
   writeByte(MPU9250_ADDRESS, I2C_SLV0_CTRL, 0x83);                     //
Enable I2C and read 3 bytes
   delay(50);
   readBytes(MPU9250_ADDRESS, EXT_SENS_DATA_00, 3, &rawData[0]);        //
Read the x-, y-, and z-axis calibration values
   magCalibration[0] =  (float)(rawData[0] - 128)/256.0f + 1.0f;        //
Return x-axis sensitivity adjustment values, etc.
   magCalibration[1] =  (float)(rawData[1] - 128)/256.0f + 1.0f;
   magCalibration[2] =  (float)(rawData[2] - 128)/256.0f + 1.0f;
   _magCalibration[0] = magCalibration[0];
   _magCalibration[1] = magCalibration[1];
   _magCalibration[2] = magCalibration[2];
   _Mmode = Mmode;

   writeByte(MPU9250_ADDRESS, I2C_SLV0_ADDR, AK8963_ADDRESS);           //
Set the I2C slave address of AK8963 and set for write.
   writeByte(MPU9250_ADDRESS, I2C_SLV0_REG, AK8963_CNTL);               //
I2C slave 0 register address from where to begin data transfer
   writeByte(MPU9250_ADDRESS, I2C_SLV0_DO, 0x00);                       //
Power down magnetometer
   writeByte(MPU9250_ADDRESS, I2C_SLV0_CTRL, 0x81);                     //
Enable I2C and transfer 1 byte
   delay(50);

   writeByte(MPU9250_ADDRESS, I2C_SLV0_ADDR, AK8963_ADDRESS);           //
Set the I2C slave address of AK8963 and set for write.
   writeByte(MPU9250_ADDRESS, I2C_SLV0_REG, AK8963_CNTL);               //
I2C slave 0 register address from where to begin data transfer
   // Configure the magnetometer for continuous read and highest resolution
   // set Mscale bit 4 to 1 (0) to enable 16 (14) bit resolution in CNTL
register,
   // and enable continuous mode data acquisition Mmode (bits [3:0]), 0010
for 8 Hz and 0110 for 100 Hz sample rates
   writeByte(MPU9250_ADDRESS, I2C_SLV0_DO, Mscale << 4 | Mmode);        //
Set magnetometer data resolution and sample ODR
   writeByte(MPU9250_ADDRESS, I2C_SLV0_CTRL, 0x81);                     //
Enable I2C and transfer 1 byte
   delay(50);
   writeByte(MPU9250_ADDRESS, I2C_SLV0_ADDR, AK8963_ADDRESS | 0x80);    //
Set the I2C slave address of AK8963 and set for read.
   writeByte(MPU9250_ADDRESS, I2C_SLV0_REG, AK8963_CNTL);               //
I2C slave 0 register address from where to begin data transfer
   writeByte(MPU9250_ADDRESS, I2C_SLV0_CTRL, 0x81);                     //
Enable I2C and transfer 1 byte
   delay(50);
}

Qammar-25, 2019-05-21:
Thank you for your help Kris. I have resolved the issue which was in the sequence of sending commands. The DO register of slave 4 should have been written first before enabling the slave. I was doing the other way around.

Qammar-25, 2019-05-21:
I can read and write to registers now. However, when i enter fuse mode to access Axial Sensitivity Adjustment Values, all three registers return me zero. Are these values correct?

kriswiner, 2019-05-21:
No these values should be 1 +, like 1.10 or 1.05, etc