kriswiner/MPU9250 #371: Self - test way too off

borstan, 2019-05-11:
Hello there, I have a problem with **self - test** values that show more than 40% difference from the factory values. In other examples, I saw that output should be as close to 0 as possible. Could someone help me solve this one? Thank you for any advice! 
I am using Sparkfun ESP32 Thing, MPU9250, Interrupt connected to the GPIO35. 

CODE: MPU9250_MS5637_ESP32 Basic Example Code (I commented all lines with MS5637 since I am not using it)

------
00:01:52.300 -> Scanning...
00:01:52.300 -> I2C device found at address 0x0C  !
00:01:52.300 -> I2C device found at address 0x68  !
00:01:52.300 -> done
00:01:52.300 -> 
00:01:52.300 -> MPU9250 9-axis motion sensor...
00:01:52.300 -> MPU9250 I AM 71 I should be 71
00:01:53.286 -> MPU9250 is online...
00:01:53.557 -> x-axis self test: acceleration trim within : -48.5% of factory value
00:01:53.591 -> y-axis self test: acceleration trim within : -51.7% of factory value
00:01:53.591 -> z-axis self test: acceleration trim within : -49.1% of factory value
00:01:53.591 -> x-axis self test: gyration trim within : -51.5% of factory value
00:01:53.591 -> y-axis self test: gyration trim within : -26.7% of factory value
00:01:53.591 -> z-axis self test: gyration trim within : -36.8% of factory value
00:01:54.607 ->  Calibrate gyro and accel
00:01:54.978 -> accel biases (mg)
00:01:54.978 -> 27.34
00:01:54.978 -> -53.83
00:01:54.978 -> 126.89
00:01:54.978 -> gyro biases (dps)
00:01:54.978 -> 8.69
00:01:54.978 -> -0.16
00:01:54.978 -> -18.47
00:01:56.371 -> MPU9250 initialized for active data mode....
00:01:56.371 -> AK8963 I AM 48 I should be 48
00:01:57.424 -> AK8963 initialized for active data mode....
00:01:57.424 -> Mag Calibration: Wave device in a figure eight until done!
00:02:19.389 -> Mag Calibration done!
00:02:19.389 -> AK8963 mag biases (mG)
00:02:19.423 -> 223.61
00:02:19.423 -> 270.64
00:02:19.423 -> 180.19
00:02:19.423 -> AK8963 mag scale (mG)
00:02:19.423 -> 1.09
00:02:19.423 -> 1.01
00:02:19.423 -> 0.92
00:02:21.422 -> X-Axis sensitivity adjustment value 1.18
00:02:21.422 -> Y-Axis sensitivity adjustment value 1.19
00:02:21.422 -> Z-Axis sensitivity adjustment value 1.14
00:02:22.405 -> ax = 0.00 ay = 0.00 az = 0.00 mg
00:02:22.439 -> gx = 0.00 gy = 0.00 gz = 0.00 deg/s
00:02:22.439 -> mx = 0 my = 0 mz = 0 mG
00:02:22.439 -> q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
00:02:22.439 -> Gyro temperature is 31.4 degrees C
00:02:22.439 -> Yaw, Pitch, Roll: 3.80, 0.00, 0.00
00:02:22.439 -> Grav_x, Grav_y, Grav_z: 0.00, 0.00, 1000.00 mg
00:02:22.439 -> Lin_ax, Lin_ay, Lin_az: 0.00, 0.00, -1000.00 mg
00:02:22.439 -> sumCount = 0
00:02:22.439 -> sum = 0.00
00:02:22.439 -> rate = nan Hz
00:02:22.914 -> ax = -17.88 ay = 13.43 az = 1303.41 mg
00:02:22.914 -> gx = 0.56 gy = 1.30 gz = 0.88 deg/s
00:02:22.982 -> mx = 1 my = -19 mz = 22 mG
00:02:22.982 -> q0 = 0.39 qx = 0.05 qy = 0.02 qz = -0.92
00:02:22.982 -> Gyro temperature is 31.5 degrees C
00:02:22.982 -> Yaw, Pitch, Roll: 229.92, 6.76, -0.15
00:02:22.982 -> Grav_x, Grav_y, Grav_z: 2.64, 117.78, 993.04 mg
00:02:22.982 -> Lin_ax, Lin_ay, Lin_az: -20.52, -104.35, 310.37 mg
00:02:22.982 -> sumCount = 1020
00:02:22.982 -> sum = 35.70
00:02:22.982 -> rate = 28.57 Hz
00:02:23.457 -> ax = -42.36 ay = 58.11 az = 1344.97 mg
00:02:23.457 -> gx = -2.86 gy = 0.56 gz = 1.82 deg/s
00:02:23.491 -> mx = 23 my = -10 mz = -4 mG
00:02:23.491 -> q0 = 0.50 qx = 0.03 qy = 0.02 qz = -0.86
00:02:23.491 -> Gyro temperature is 31.6 degrees C
00:02:23.491 -> Yaw, Pitch, Roll: 244.07, 4.41, -0.15
00:02:23.491 -> Grav_x, Grav_y, Grav_z: 2.65, 76.95, 997.03 mg
00:02:23.491 -> Lin_ax, Lin_ay, Lin_az: -45.00, -18.85, 347.94 mg
00:02:23.491 -> sumCount = 1010
00:02:23.491 -> sum = 0.52
00:02:23.491 -> rate = 1944.59 Hz

kriswiner, 2019-05-11:
Check your code with this:

https://github.com/kriswiner/MPU9250/blob/master/Documents/AN-MPU-9250A-03%20MPU-9250%20Accel%20Gyro%20and%20Compass%20Self-Test%20Implementation%20v1%200_062813.pdf

The self test uses only the 8 LSBits.

borstan, 2019-05-12:
Hello Kris, I read true the sheet you sent me but I am not very clear which are the trim values I should change in your code to have it working properly.

In this test I printed out only Yaw, Pitch and Roll (data is in this order) and when I did 8 pattern I am getting quite stabile output (you said that when the MPU is stationary on the table the Pitch and Roll should be larger than 1). Problem is when I don't do the 8 pattern data pitch and roll gets >6. How can I achieve calibrated data without performing 8 pattern before every start of communication with MPU?

16:27:26.165 -> Scanning...
16:27:26.199 -> I2C device found at address 0x0C  !
16:27:26.199 -> I2C device found at address 0x68  !
16:27:26.199 -> done
16:27:26.199 -> 
16:27:26.199 -> MPU9250 9-axis motion sensor...
16:27:26.199 -> MPU9250 I AM 71 I should be 71
16:27:27.183 -> MPU9250 is online...
16:27:27.521 -> x-axis self test: acceleration trim within : -46.0% of factory value
16:27:27.521 -> y-axis self test: acceleration trim within : -43.1% of factory value
16:27:27.521 -> z-axis self test: acceleration trim within : -42.9% of factory value
16:27:27.521 -> x-axis self test: gyration trim within : -45.3% of factory value
16:27:27.521 -> y-axis self test: gyration trim within : -24.0% of factory value
16:27:27.521 -> z-axis self test: gyration trim within : -34.8% of factory value
16:27:28.502 ->  Calibrate gyro and accel
16:27:28.875 -> accel biases (mg)
16:27:28.875 -> -10.44
16:27:28.875 -> -38.64
16:27:28.875 -> -87.34
16:27:28.875 -> gyro biases (dps)
16:27:28.875 -> 6.25
16:27:28.875 -> -0.16
16:27:28.875 -> -10.29
16:27:30.267 -> MPU9250 initialized for active data mode....
16:27:30.267 -> AK8963 I AM 48 I should be 48
16:27:31.320 -> AK8963 initialized for active data mode....
16:27:31.320 -> Mag Calibration: Wave device in a figure eight until done!
16:27:53.317 -> Mag Calibration done!
16:27:53.317 -> AK8963 mag biases (mG)
16:27:53.317 -> 101.16
16:27:53.317 -> 292.01
16:27:53.317 -> 61.78
16:27:53.317 -> AK8963 mag scale (mG)
16:27:53.317 -> 1.06
16:27:53.317 -> 0.64
16:27:53.317 -> 2.02
16:27:55.319 -> X-Axis sensitivity adjustment value 1.18
16:27:55.319 -> Y-Axis sensitivity adjustment value 1.19
16:27:55.319 -> Z-Axis sensitivity adjustment value 1.14
16:27:56.305 -> 3.80, 0.00, 0.00
16:27:56.815 -> 248.49, 0.37, -0.07
16:27:57.323 -> 248.75, 0.40, -0.64
16:27:57.799 -> 248.34, -0.32, -0.51
16:27:58.309 -> 248.69, 0.35, 0.05
16:27:58.817 -> 248.41, 0.02, -0.87
16:27:59.327 -> 250.03, 0.33, 0.64
16:27:59.836 -> 249.06, 0.46, 0.22
16:28:00.313 -> 248.04, 0.12, -0.48
16:28:00.822 -> 248.41, -0.03, 0.05
16:28:01.331 -> 249.18, 0.81, 0.39
16:28:01.840 -> 248.74, 0.35, -0.35
16:28:02.315 -> 247.66, 0.73, -0.74
16:28:02.823 -> 248.96, 0.26, 0.12
16:28:03.331 -> 248.59, 0.04, 0.30
16:28:03.807 -> 248.63, -0.49, -0.77
16:28:04.317 -> 248.75, 0.08, 0.38
16:28:04.826 -> 247.59, 0.43, -0.36
16:28:05.334 -> 249.26, -0.02, -0.30

kriswiner, 2019-05-12:
Store the calibration corrections in the main program so you don't have to
redo them every time.

borstan, 2019-05-12:
Okay I'll do that, and what do you think about pitch, roll and Yaw values? I thought after calibration pitch will get stabilized at 360 if I keep the mpu stationary.

kriswiner, 2019-05-12:
Pitch and roll should be 0 is stationary and flat if the sensors are
properly calibrated.

borstan, 2019-05-14:
Kris, thank you for your advice. I put my calibrated data into the main program and calibration process works fine now, while not moving, I am getting pitch and roll values under 1 (even tho I have constant changes of sign of Pitch while not moving - don't know if it is an issue or not). 

But I did an observation and find out that:
1) When I move MPU in the Yaw direction while resting on the table everything works fine.
2) When I move in the Pitch direction pitch starts increasing with the Yaw value.
3) While I do roll I can achieve 30° and then Yaw starts increasing, 

Do you know what might be a problem here? 

Input to my Quaternion update is like this:

 MahonyQuaternionUpdate(ax, ay, az, gx * PI / 180.0f, gy * PI / 180.0f, gz * PI / 180.0f, my, mx, mz);

kriswiner, 2019-05-14:
" MahonyQuaternionUpdate(ax, ay, az, gx * PI / 180.0f, gy * PI / 180.0f, gz
* PI / 180.0f, my, mx, mz);"

This cannot be correct. The data needs to be fed in NED or ENU order, a RH
coordinate system.

So first decide which accel axis will be designated as North. The determine
which of the remaining axes is pointing toward North, East and Down. The
input the data into the Update function accordingly.

I know the above cannot be correct because the gyro and mag z-axes are
opposite each other, for example.

robertoostenveld, 2019-05-14:
Hi @borstan, you may want to check #370.

borstan, 2019-05-14:
Thank you, guys. I used the help from the #370 and now it is much better I am feeding data in this order (_MahonyQuaternionUpdate(ax, ay, az, gy * PI / 180.0f, gx * PI / 180.0f, gz * PI / 180.0f, my, mx, -mz)_). 

But while having a pitch close to +-90° Yaw is getting a bit crazy. I tried to print ax, ay, az; gx, gy, gz; mx, my, mz and compare it to some values that Kris founds as reasonable. 

**Observation:**
I think that accelerometer is well calibrated as the ax and ay don't go above 10 miliGs and az oscillates around 1000.
But I am concerned about the Mag values, are they in norm? I already did the calibration and put the MagBiases and scales into the code. 

Do you know what might be a problem?

Thank you for any help.

_TAB: Calibrated raw data output_

ax | ay | az | gx | gy | gz | mx | my | mz
-- | -- | -- | -- | -- | -- | -- | -- | --
1.65 | 0.37 | 998.05 | 0.00 | 0.02 | 0.04 | 95 | 276 | 228
2.62 | 1.10 | 1000.49 | 0.00 | 0.04 | 0.12 | 92 | 308 | 228
1.65 | -0.12 | 998.78 | 0.08 | 0.06 | -0.02 | 95 | 304 | 216
1.53 | -1.83 | 998.84 | -0.02 | -0.05 | 0.02 | 96 | 278 | 210
2.14 | 0.55 | 999.57 | -0.03 | 0.02 | -0.02 | 104 | 284 | 212
1.89 | 3.23 | 997.56 | 0.18 | 0.02 | 0.05 | 108 | 290 | 214
3.17 | -1.59 | 994.81 | 0.00 | 0.01 | 0.05 | 95 | 300 | 209
2.38 | -0.43 | 1000.12 | 0.08 | -0.01 | 0.02 | 102 | 298 | 222
3.11 | 0.24 | 999.27 | 0.01 | -0.02 | 0.02 | 99 | 294 | 218
1.89 | -0.18 | 998.05 | 0.09 | -0.03 | -0.07 | 113 | 308 | 224
-0.12 | -0.31 | 998.54 | 0.01 | 0.05 | 0.04 | 92 | 292 | 220
2.08 | -0.12 | 998.78 | 0.00 | -0.06 | 0.01 | 104 | 292 | 216
1.40 | -1.40 | 998.29 | -0.02 | -0.06 | 0.08 | 98 | 296 | 216
1.04 | 0.73 | 999.02 | -0.07 | -0.03 | 0.05 | 107 | 300 | 224
1.59 | 0.85 | 996.28 | -0.01 | -0.04 | 0.03 | 98 | 292 | 220
2.20 | -0.24 | 997.99 | 0.05 | 0.05 | 0.00 | 95 | 288 | 216
1.34 | -0.79 | 1001.04 | -0.02 | -0.02 | 0.14 | 87 | 282 | 214
0.55 | 1.95 | 999.33 | -0.05 | -0.01 | 0.06 | 98 | 284 | 220
0.61 | -0.49 | 997.38 | 0.05 | -0.02 | 0.05 | 90 | 318 | 222
1.28 | 0.31 | 999.27 | -0.04 | 0.02 | 0.05 | 101 | 284 | 220
1.22 | 3.36 | 999.76 | -0.01 | 0.02 | 0.02 | 90 | 294 | 210
-0.06 | 0.67 | 1000.67 | -0.05 | -0.04 | 0.08 | 101 | 304 | 216
2.32 | -1.40 | 997.62 | 0.00 | -0.02 | -0.05 | 93 | 302 | 214
0.79 | 0.43 | 997.68 | -0.02 | -0.04 | 0.07 | 98 | 280 | 209
0.24 | -0.43 | 1000.37 | 0.03 | -0.03 | -0.03 | 96 | 294 | 233
0.73 | 0.61 | 996.40 | 0.05 | -0.07 | 0.02 | 102 | 286 | 222
3.60 | 1.10 | 998.60 | 0.01 | 0.02 | 0.05 | 101 | 288 | 209
0.31 | -0.31 | 997.80 | -0.01 | 0.00 | 0.01 | 98 | 300 | 209
1.89 | 0.00 | 1000.92 | 0.03 | -0.02 | 0.06 | 93 | 290 | 207
3.36 | -0.12 | 999.27 | -0.02 | -0.07 | 0.11 | 90 | 310 | 222
1.95 | -0.37 | 998.05 | 0.04 | -0.02 | 0.03 | 105 | 298 | 229
1.53 | 1.16 | 998.84 | 0.02 | 0.01 | 0.02 | 113 | 296 | 197
2.44 | 0.24 | 999.57 | 0.04 | -0.01 | 0.06 | 96 | 282 | 222
2.93 | -0.31 | 997.38 | 0.04 | -0.08 | -0.05 | 96 | 286 | 218
2.62 | -1.34 | 996.40 | -0.01 | -0.02 | 0.00 | 104 | 288 | 231
2.75 | -0.31 | 1001.16 | -0.02 | -0.05 | 0.03 | 90 | 282 | 222
2.44 | -0.85 | 1000.37 | 0.08 | -0.03 | 0.13 | 99 | 298 | 214
2.75 | -1.10 | 1001.40 | 0.05 | 0.02 | 0.13 | 99 | 302 | 222
-0.37 | -0.67 | 1000.24 | 0.02 | -0.05 | 0.11 | 99 | 278 | 222
2.99 | 2.26 | 999.45 | -0.02 | -0.02 | -0.05 | 119 | 296 | 231
2.38 | 0.79 | 1000.18 | -0.15 | -0.07 | 0.00 | 98 | 308 | 220
2.26 | -1.46 | 1000.37 | -0.08 | -0.11 | -0.03 | 96 | 278 | 184
1.77 | -0.55 | 1000.61 | 0.08 | -0.03 | 0.05 | 101 | 300 | 212
1.16 | 0.67 | 997.31 | 0.05 | 0.07 | -0.03 | 95 | 296 | 224
1.40 | -0.43 | 998.72 | -0.02 | -0.07 | 0.05 | 110 | 300 | 220
1.89 | 0.31 | 1001.59 | 0.04 | -0.01 | 0.05 | 95 | 284 | 209
1.40 | -0.61 | 998.84 | -0.06 | -0.09 | 0.08 | 101 | 296 | 205
1.71 | -0.92 | 999.15 | -0.03 | -0.08 | 0.07 | 99 | 294 | 218
3.36 | 0.79 | 996.58 | 0.01 | -0.06 | 0.01 | 98 | 296 | 212
3.23 | -0.61 | 1000.31 | -0.01 | 0.00 | 0.01 | 98 | 280 | 209
0.00 | 0.55 | 994.87 | 0.05 | -0.04 | 0.02 | 92 | 280 | 197
1.34 | -1.22 | 997.25 | -0.01 | 0.02 | 0.05 | 105 | 302 | 210
2.87 | -0.43 | 997.31 | -0.02 | 0.02 | 0.11 | 110 | 296 | 209
2.56 | 0.98 | 996.22 | -0.07 | -0.07 | 0.01 | 104 | 304 | 224
-0.24 | 0.06 | 997.25 | -0.02 | 0.01 | 0.04 | 95 | 308 | 228
-0.06 | 0.67 | 999.33 | 0.09 | 0.02 | 0.00 | 107 | 284 | 212
0.98 | 1.10 | 999.39 | 0.05 | 0.00 | -0.05 | 95 | 296 | 212
2.32 | 0.43 | 997.62 | -0.02 | -0.04 | 0.05 | 101 | 296 | 209
2.99 | 0.24 | 999.76 | 0.06 | 0.00 | -0.02 | 102 | 302 | 222
3.91 | 0.67 | 999.57 | 0.05 | -0.05 | -0.02 | 98 | 288 | 216
2.99 | -0.12 | 998.78 | 0.00 | -0.08 | -0.02 | 104 | 284 | 220
0.98 | 0.55 | 994.26 | 0.00 | 0.00 | 0.04 | 89 | 300 | 224
0.92 | -0.31 | 998.78 | 0.02 | -0.02 | -0.02 | 90 | 286 | 218
1.04 | 0.12 | 998.90 | -0.03 | -0.05 | -0.01 | 86 | 296 | 220
1.10 | -0.37 | 999.88 | -0.06 | 0.01 | 0.02 | 96 | 294 | 207
2.14 | 0.85 | 999.39 | -0.05 | 0.02 | 0.09 | 101 | 296 | 209
3.30 | 1.65 | 1001.10 | -0.03 | -0.02 | 0.00 | 99 | 290 | 226
2.87 | -0.67 | 996.46 | -0.06 | -0.05 | 0.00 | 87 | 306 | 210
2.26 | 0.49 | 997.62 | -0.01 | -0.04 | -0.04 | 93 | 286 | 218
2.50 | -1.40 | 998.17 | 0.07 | -0.03 | 0.06 | 102 | 310 | 222
3.23 | -0.55 | 1000.61 | 0.02 | 0.01 | -0.02 | 105 | 282 | 229
3.11 | -0.43 | 1002.38 | 0.08 | -0.11 | 0.00 | 99 | 298 | 218

kriswiner, 2019-05-15:
" (*MahonyQuaternionUpdate(ax, ay, az, gy * PI / 180.0f, gx * PI / 180.0f,
gz * PI / 180.0f, my, mx, -mz)*)"

sure the gyro is right

kriswiner, 2019-05-15:
Not for the MPU9250, since the accel and gyro are aligned...so if Ax is
North, so is Gx.

robertoostenveld, 2019-05-15:
My experience with a calibration scheme like [this one](https://github.com/sparkfun/SparkFun_MPU-9250_Breakout_Arduino_Library/blob/ea379ce8c930ef90f4ebc37799544d68108e3126/src/MPU9250.cpp#L622) (which takes max/min values along the cardinal axes), is that waving in a figure of eight does not work as well as rotating it as a cube/dice, making sure that each axis (side of the cube/dice) at a certain moment points towards the magnetic north, about 45 degrees downward. 

Note that the magnetic north field direction is not aligned along the surface of the earth, but pointing quite steep down (for me in Europe). See 

[image]

which I took from [here](https://www.researchgate.net/publication/279446083_Magnetoreception_in_Birds_and_Its_Use_for_Long-Distance_Migration/figures?lo=1).

kriswiner, 2019-05-15:
> [image:
> The-Earths-magnetic-field-the-geomagnetic-field-Notice-that-the-southern-and-northern]

Yes, we have developed a calibration cube to do this all better. The idea
is to have a fixed bracket such that the precision cube faces are always
oriented with a reference surface. Then by taking data at the 24 unique
cube positions and doing some math both the accel and mag are calibrated
and the cross axis error is also compensated for. The result is < 1 degree
heading accuracy or better. This device will be for sale on Tindie in a few
weeks time.