kriswiner/MPU9250 #372: Sensor can not be found

Papierkor654, 2019-05-12:
Hello

I have a problem connection my MPU 9250 ( https://www.sparkfun.com/products/13762 ) to my Arduino Mega 2560. I've got it connected as follows:

_MPU9250 --------- Arduino Mega 2560
 VDD ---------------------- 3.3V
 SDA ----------------------- A20
 SCL ----------------------- A21
 GND ---------------------- GND_

When I run a I2C scan, a device is found at 0x68:

_I2C device found at address 0x68  !_

so the mpu9250 should be recognised right?

But if I run the code / library from sparkfun ( https://github.com/sparkfun/SparkFun_MPU-9250_Breakout_Arduino_Library/tree/master/examples/MPU9250BasicAHRS_I2C ) which is nearly the same code as the one from kriswiner (as far as I know its heavily based on his code) the sensor is not found:

_MPU9250 I AM 0xFF I should be 0x71
Could not connect to MPU9250: 0xFF
Communication failed, abort!_ 

Does anyone have an idea, why the sensor isn’t found? I have already tried connection ADO to high and low and nothing worked so far…

Best Regards
Marco

kriswiner, 2019-05-12:
What address does the code expect to find the MPU9250? I am betting 0x69.

Papierkor654, 2019-05-12:
Hey Kris

Thanks for the quick response. 
As far as I can tell, it expects 0x71. 

What does the 0xff mean? Is this a sign, that the code can't see the sensor?

kriswiner, 2019-05-12:
No, the MPU9250 can have one of two I2C addresses, 0x68 or 0x69. Your has
0x68 )ADO is pulled LOW) so in your code, you need to set the MPU9250
address to 0x687. I am betting that it is set to 0x69.

Papierkor654, 2019-05-13:
As far as I can tell the code is the same as yours. So there are these two lines:
_#define MPU9250_ADDRESS_AD1 0x69  // Device address when ADO = 1
#define MPU9250_ADDRESS_AD0 0x68  // Device address when ADO = 0_
But then there is also another code line:
_if (c == 0x71) // WHO_AM_I should always be 0x71_ 

Where would I change the address from 68 to 69?

And if the code returns 0xff for the WHO_AM_I isn't this a sign that the code doesn't seem to find the sensor?

I've tried putting AD0 high, and the address remains 68, any suggestions why this is so?

kriswiner, 2019-05-13:
" Where would I change the address from 68 to 69? " It's in the code
somewhere, find it.

" And if the code returns 0xff for the WHO_AM_I isn't this a sign that the
code doesn't seem to find the sensor?" Yes.

" I've tried putting AD0 high, and the address remains 68, any suggestions
why this is so? " Crappy breakout design?.

Papierkor654, 2019-05-13:
Hey Kris

Thanks a lot for the response. Luckily I've been able to get the IMU running with a different code (which is also based on your code). Had to change a 1 and a 2 to a 1L and a 2L... Luckily the solution was out there on the internet, I wouldn't have been able to find the solution on my own.

Thanks a lot for your support, and obviously for the amazing code / library you've created.
Have a great day ^^