kriswiner/MPU9250 #373: ESP32 SPI DMP

Lynxu, 2019-05-19:
Hello @kriswiner do you know how can i get working dmp on a ESP32 on spi. I'm preety new to stuff like this. I got stuck on a project where i have three MPU9250 and need the quaternions for them.

kriswiner, 2019-05-19:
No, I don't use the DMP on the MPU9250. I prefer to do the sensor fusion on
the host MCU.

Lynxu, 2019-05-19:
Do you have something that could help me to get the quaternions? Any information is needed right now.

kriswiner, 2019-05-19:
This:  https://github.com/kriswiner/ESP32/tree/master/MPU9250_MS5637

and this: https://github.com/kriswiner/MPU9250/tree/master/Dual_MPU9250

Lynxu, 2019-06-08:
@kriswiner which of your sketches think would work the best for esp32

kriswiner, 2019-06-08:
Maybe this one:
https://github.com/kriswiner/ESP32/tree/master/MPU9250_MS5637

Lynxu, 2019-06-08:
@kriswiner did you made something that could work on spi too?

kriswiner, 2019-06-08:
No, but this should be a straightforward change to the read/writeByte
functions.