kriswiner/MPU9250 #374: instability in roll & yaw when pitch reaches ninety degrees

shomedas, 2019-05-22:
I am using  a mpu9250 with esp32(from kriswiner/ESP32/tree/master/MPU9250_MS5637). My setup is perfectly calibrated. I get a fusion rate of nearly 2kHz. There is no drift.

I align the MPU9250 with NED. The quaternion reads nearly 1,0,0,0. yaw, pitch, roll is nearly zero each. Now I rotate the IMU only along the pitch axis. Initially, only the pitch changes. But as the pitch approaches +/-90 degrees the yaw and roll show spiky behavior.  Is this the phenomenon of gimbal lock? Is it possible to get independent angle change for all axis? Or this kind of abrupt change in yaw & roll is unavoidable near +/-90degrees.

[image]

kriswiner, 2019-05-22:
This is just how absolute orientation sensors work. Can you answer this
question?

When you are standing on the North Pole, which direction is South?

shomedas, 2019-05-22:
There are infinite solutions to your question regrading north pole.

In my application i need to avoid the region where pitch tends to +/-90. The other thing I saw is that this effect starts way before 90deg. It starts at 80 degrees. Is there any rule of thumb of the safe distance from 90 degree to avoid gimbal lock?

kriswiner, 2019-05-22:
Yes, yaw is undefined at 90 degrees. However, if the sensors are properly
calibrated and the fusion filter is properly configured the yaw should be
stable and accurate up to a few degrees from 90. This is not gimbal lock
per se, this is simply the fact that at 90 degrees the yaw is degenerate,
i.e. can take on any value.

Some applications need to operate with the sensor oriented at pitch == 90
degrees. In this case, the solution is to redefine the NED assignments to
make this orientation the normal one. But there is no way to maintain
accurate yaw from pitch == 0 to pitch == 90.

shomedas, 2019-05-22:
I plotted the Quaternion components. They seemed to be fine i.e. without any sudden changes in value. So I guessed it has to do with the way Euler angles are defined. Also this problem occurs only with pitch values near 90. I did not observe it while changing yaw or roll to 90 or near 90 degrees

kriswiner, 2019-05-22:
Roll goes from -180 to 180, heading remains well defines on the equator,
just has trouble at the poles. well-behaved quaternions is why people use
them, no gimbal lock. You still have the problem that South (or West or
East) is undefined at the poles.