kriswiner/MPU9250 #375: correct order for NED with x or y axis axis pointing north

shomedas, 2019-05-23:
I use https://github.com/kriswiner/ESP32/blob/master/MPU9250_MS5637/MPU9250_MS5637_AHRS.ino
MadgwickQuaternionUpdate(-ax, ay, az, gx*pi/180.0f, -gy*pi/180.0f, -gz*pi/180.0f, my, -mx, mz);
I read from your other posts that order for NED should be 
(AN,AE,AD,GN,GE,GD,MN,ME,MD). Why are the gyroscope expression sign and the accelerometer expression sign opposite in madgwick expression? For example **-**ax is used as is **+**gx*pi/180. Similarly **+**ay is used as is **-**gy*pi/180.0f \

I get quaternion  q[0] = 1.0, q[1]=0.004,q[2]=-0.056,q[3]=-0.001 in the following image  IMU orientation. However my alignment with NED looks inconsistent with the madgwick convention above. ax points north while -ay points east in image but in madgwick code above -ax points north and ay points east. Is something wrong here?

[image]

Also what is the correct order for NED with y axis of the accelerometer pointing north and x axis pointing east.  I tried 
MadgwickQuaternionUpdate(ay, ax, -az, gy*pi/180.0f, gx*pi/180.0f, -gz*pi/180.0f, mx, my, mz);
I get quaternion in the following image IMU orientation q[0] = 0, q[1]=**-**1, q[2]=0, q[3]=0

[image]

kriswiner, 2019-05-23:
I usually invert the accel data to account for gravity being +1 up. But
also because the EM7180 does this so it is easiest to compare.

If you select a different axis as North just use the same procedure to
identify which other axes point East and Down.

shomedas, 2019-05-23:
How to set the gyroscope component sign in the MadgwickQuaternionUpdate(-ax, ay, az, gxpi/180.0f, -gypi/180.0f, -gzpi/180.0f, my, -mx, mz). i can read the ax, ay, az directions. But how to understand gyroscope direction/sign?

kriswiner, 2019-05-23:
Which direction do the gyro axes point? That's how.

shomedas, 2019-05-23:
The gyro axes and the accel axis point in the same direction in the datasheet. But you invert the sign of the accelerometer. So gyro axes and accel axes point in opposite direction. Is that what you mean? Does it hold good in all of your codes?

kriswiner, 2019-05-23:
Not sure what hold good means, this is what I usually do, YMMV.

shomedas, 2019-05-23:
I usually go through all the issues and also parts of your codr in your GitHub before I ask a question. So I asked if all your codes followed the same convention.

kriswiner, 2019-05-23:
No.