kriswiner/MPU9250 #377: endless loop with vibration

Papierkor654, 2019-05-31:
I'm currently building an arduino drone and I have the problem, that the IMU code gets stuck somewhere if the Propeller speed is at a certain rate. This only happens, if the IMU is sticked to the drone. If I take the IMU of the drone and hold it in my hands, there is no error and everything runs fine.

Dos anyone have an idea why this could happen? Is it possible that the IMU gets stuck, when it experiences to much vibration?

The soldering of the IMU should be fine. There is no problem, if you touch / bend the cable.

(I've already posted this problem to the place where I've got the software from (https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library) but since the code is based on the code from this side and because this side is more active, I've decided to post the problem here as well.)

zrecore, 2019-05-31:
The MPU-9250 has microscopic mechanism. Have you tried adding dampening
material to the IMU board mount?

Papierkor654, 2019-05-31:
I've put a layer of velcro under the IMU. Don't know if this is enough. 

Put is it really possible, that the IMU gets stuck, if there is to much vibration?

Papierkor654, 2019-05-31:
I've found a solution:

Apparently the reason was, that there was to much vibration and because of that, the IMU had an error. (no  idea why exactly) 
I've straped something heavy onto the IMU, to give it some mass in order to prevent vibration (I've used a AA battery). This solved the problem for me.

closed 2019-05-31 by Papierkor654