kriswiner/MPU9250 #378: MPU9250 I AM 78 I should be 71

meta-meta, 2019-06-04:
I was getting ```
MPU9250 I AM FF I should be 71
Could not connect to MPU9250: 0xFF
```

I've added an I2C scan which reports:
```
Scanning...
I2C device found at address 0x68  !
done
```
So I switched ADO to 0.

Now I get:
```
MPU9250 I AM 78 I should be 71
Could not connect to MPU9250: 0x78
```

I am using a Pro Micro 5v with a cheap Chinese board which takes 3-5v: https://www.amazon.com/gp/product/B07KQ65HXQ/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

Any ideas why I'd get `I AM 78`?

kriswiner, 2019-06-04:
Switch the address to 0x68 in the code. 0x78 sounds like a random value.

meta-meta, 2019-06-05:
Thanks for the help though I'm confused. Which address?

Since I set ADO to 0, I expect `MPU9250_ADDRESS` to be `0x68`

`WHO_AM_I_MPU9250` is not the same thing as `MPU9250_ADDRESS` right?

`readByte(MPU9250_ADDRESS, WHO_AM_I_MPU9250);` is returning `0x78`

Is this a bad idea?
`writeByte(MPU9250_ADDRESS, WHO_AM_I_MPU9250, 0x71);`

kriswiner, 2019-06-05:
" Since I set ADO to 0, I expect MPU9250_ADDRESS to be 0x68"

expect? Can you verify this?

" WHO_AM_I_MPU9250 is not the same thing as MPU9250_ADDRESS right?" right

"Is this a bad idea?
writeByte(MPU9250_ADDRESS, WHO_AM_I_MPU9250, 0x71);"

yes, very

meta-meta, 2019-06-06:
```
 Serial.print("MPU9250_ADDRESS: ");
  Serial.println(MPU9250_ADDRESS, HEX);
  
  byte c = readByte(MPU9250_ADDRESS, WHO_AM_I_MPU9250);  // Read WHO_AM_I register for MPU-9250
```
> MPU9250_ADDRESS: 68
> MPU9250 I AM 78 I should be 71

(I'm going to try with a breakout board from another manufacturer tomorrow)

kriswiner, 2019-06-06:
Yes, looks like you have an exotic version of the "MPU9250". The MPU9255
has WHO_AM_I of 0x73, but I have not heard of any variant with 0x78.

meta-meta, 2019-06-06:
OK! new chips work like a charm. Thank you for your guidance! I'll leave a poor review on Amazon for the other chip. I hope this helps someone in the future.

closed 2019-06-06 by meta-meta