kriswiner/MPU9250 #383: Yaw angular velocity measurement (rate of change of rotation)

cwarrior777, 2019-08-01:
Hi Kris,

I am writing to you because I need you to advise me about the sensor.
I must obtain the angular change rate (in degrees / seconds) to measure the rotation in an airplane at its yaw angle. I really don't know whether to use the gyroscope's output or if I have to calculate after I get the YAW values from Madgwick / Mahonny.
What method would you recommend for this?

I've been watching the post https://github.com/kriswiner/MPU9250/issues/132
but I have not finished understanding why this happens.

Keep in mind that I have already calibrated the 3 sensors very acceptably, but I don't know if there will be drift in the gyroscope that affects me in the measurement (I know that the drift appears when the gyroscope's output is integrated to obtain the angle).

Thank you very much and I hope you can help me!

kriswiner, 2019-08-01:
You will want to use the quaternions to obtain absolute orientation, and
the the angle about the yaw axis (the roll) is easily found.

cwarrior777, 2019-08-01:
Yes, but, how can i do that?, I have seen in that post that i've mentioned that he put the following equations:
// Compute Angular Rate : w = 2 x qDot x qStar
AngRat[0] = 2.0f * (qDot1*q[0] + qDot2*q[2] + qDot3*q[3] + qDot4*q[4]);
AngRat[1] = 2.0f * (qDot1*q[2] - qDot2*q[1] - qDot3*q[4] + qDot4*q[3]);
AngRat[2] = 2.0f * (qDot1*q[3] + qDot2*q[4] - qDot3*q[1] - qDot4*q[2]);
AngRat[3] = 2.0f * (qDot1*q[4] - qDot2*q[3] + qDot3*q[2] - qDot4*q[1]);
I don't really know if that equations are well.
But it doesn't work, and i don't know how calculate from yaw to obtain the angular rate of that angle.

kriswiner, 2019-08-01:
Just use the roll, google it.