kriswiner/MPU9250 #387: Quick Question on using MPL3115A2 Functions

mjs513, 2019-09-14:
Hi Kris

Long time.   Have a quick question for you.  No problems for once.  I am playing around with Paul's NXPMotionSense library and just want to let you know that I am planning on using MPL3115A2 functions into that library if that is alright with you.  Hate reinventing the wheel so to speak.

kriswiner, 2019-09-17:
Sorry for the late reply, no problem-it's all open source. But this
pressure sensor is terribly down rev, is it not?

mjs513, 2019-09-17:
Not a problem.  Not 100% sure on the web page its identified as a MPL3115A2.

Anyway the pieces I moved over was the code to read barometric and altimeter modes, as using toggleOneShot.  I also reduced the OS to 32ms as opposed to 512ms that was currently in the code.  Was going to add more but figured I would keep it simple

Getting descent results and a bit faster for the pressure and altimeter.  I also added I function to set local sea level pressure.