kriswiner/MPU9250 #388: heading angle problem

alieng1994, 2019-09-23:
Hi Kris, thanks for the great job you doing my problem is I'm getting wrong value for heading angle and I,m using your code and calibration method so can you help me?
thanks

closed 2019-09-23 by alieng1994

kriswiner, 2019-09-23:
Fixed so fast, that's great!

alieng1994, 2019-09-23:
no ,it is the first time using GitHub I thought by clicking close mean it will publish but my problem still open and definitely I need help

kriswiner, 2019-09-23:
What is the problem?

alieng1994, 2019-09-23:
the heading angle in the mpu9250 it gives one value and does not change according to the rotation

kriswiner, 2019-09-23:
What do your sensor show? accel, gyro and mag?

alieng1994, 2019-09-23:
the sensor showing roll,pitch and yaw all the valu are corecet but the yaw is incorrect and I'm using the quatronain filter for the sensor fusion

kriswiner, 2019-09-23:
Then your sensors are not calibrated.

alieng1994, 2019-09-23:
I'm using the calibration code in the library and it is show on the Serial monitor a calibration values for the three sensors

kriswiner, 2019-09-23:
How do you know the sensors are calibrated? How did you test this?

alieng1994, 2019-09-23:
I compared the result to a mobile phone app the roll and the pitch were correct but the yaw wasn't but if your are suggest that the calibration were wrong what can I do?

alieng1994, 2019-09-23:
and this the code I used 

#include "Wire.h"  
#include "MPU9250.h"
uint8_t Gscale = GFS_250DPS, Ascale = AFS_2G, Mscale = MFS_16BITS, Mmode = M_100Hz, sampleRate = 0x04;         
float aRes, gRes, mRes;      // scale resolutions per LSB for the sensors
float pi = 3.141592653589793238462643383279502884f;
float GyroMeasError = pi * (40.0f / 180.0f);   // gyroscope measurement error in rads/s (start at 40 deg/s)
float GyroMeasDrift = pi * (0.0f  / 180.0f);   // gyroscope measurement drift in rad/s/s (start at 0.0 deg/s/s)
float beta = sqrtf(3.0f / 4.0f) * GyroMeasError;   // compute beta
float zeta = sqrtf(3.0f / 4.0f) * GyroMeasDrift;   // compute zeta, the other free parameter in the Madgwick scheme usually set to a small or zero value

int16_t MPU9250Data[7]; // used to read all 14 bytes at once from the MPU9250 accel/gyro
int16_t magCount[3];    // Stores the 16-bit signed magnetometer sensor output
float   magCalibration[3] = {0, 0, 0};  // Factory mag calibration and mag bias
float   temperature;    // Stores the MPU9250 internal chip temperature in degrees Celsius
float   SelfTest[6];    // holds results of gyro and accelerometer self test

uint32_t delt_t = 0;                      // used to control display output rate
uint32_t count = 0, sumCount = 0;         // used to control display output rate
float pitch, yaw, roll;                   // absolute orientation
float a12, a22, a31, a32, a33;            // rotation matrix coefficients for Euler angles and gravity components
float deltat = 0.0f, sum = 0.0f;          // integration interval for both filter schemes
uint32_t lastUpdate = 0, firstUpdate = 0; // used to calculate integration interval
uint32_t Now = 0;                         // used to calculate integration interval

float lin_ax, lin_ay, lin_az;             // linear acceleration (acceleration with gravity component subtracted)
float q[4] = {1.0f, 0.0f, 0.0f, 0.0f};    // vector to hold quaternion
float eInt[3] = {0.0f, 0.0f, 0.0f};    

// These can be measured once and entered here or can be calculated each time the device is powered on
float   gyroBias[3] = {0.96, -0.21, 0.12}, accelBias[3] = {0.00299, -0.00916, 0.00952};
float   magBias[3] = {71.04, 122.43, -36.90}, magScale[3]  = {1.01, 1.03, 0.96}; // Bias corrections for gyro and accelerometer

float ax, ay, az, gx, gy, gz, mx, my, mz; // variables to hold latest sensor data values 

int  intPin = 12;  //  MPU9250 interrupt
int  myLed  = 13; // red led

bool intFlag = false;
bool newMagData = false;
MPU9250 MPU9250(intPin); // instantiate MPU9250 class
void myinthandler(){
  intFlag = true;
}

void setup() {
  Serial.begin(115200);
  delay(1000);
  
  Wire.begin(); // set master mode, default on SDA/SCL for Ladybug   
  Wire.setClock(400000); // I2C frequency at 400 kHz
  delay(1000);

  MPU9250.I2Cscan(); // should detect BME280 at 0x77, MPU9250 at 0x71 
  
  // Set up the interrupt pin, it's set as active high, push-pull
  pinMode(myLed, OUTPUT);
  digitalWrite(myLed, HIGH); // start with orange led on (active HIGH)

  pinMode(intPin, INPUT);

   /* Configure the MPU9250 */
  // Read the WHO_AM_I register, this is a good test of communication
  Serial.println("MPU9250 9-axis motion sensor...");
  uint8_t c = MPU9250.getMPU9250ID();
  Serial.print("MPU9250 "); Serial.print("I AM "); Serial.print(c, HEX); Serial.print(" I should be "); Serial.println(0x71, HEX);
  delay(1000);
  
  if (c == 0x71 ) // WHO_AM_I should always be 0x71 for MPU9250, 0x73 for MPU9255 
[... 199 lines cut ...]
            norm = 1.0f/norm;
            s1 *= norm;
            s2 *= norm;
            s3 *= norm;
            s4 *= norm;

            // Compute rate of change of quaternion
            qDot1 = 0.5f * (-q2 * gx - q3 * gy - q4 * gz) - beta * s1;
            qDot2 = 0.5f * (q1 * gx + q3 * gz - q4 * gy) - beta * s2;
            qDot3 = 0.5f * (q1 * gy - q2 * gz + q4 * gx) - beta * s3;
            qDot4 = 0.5f * (q1 * gz + q2 * gy - q3 * gx) - beta * s4;

            // Integrate to yield quaternion
            q1 += qDot1 * deltat;
            q2 += qDot2 * deltat;
            q3 += qDot3 * deltat;
            q4 += qDot4 * deltat;
            norm = sqrtf(q1 * q1 + q2 * q2 + q3 * q3 + q4 * q4);    // normalise quaternion
            norm = 1.0f/norm;
            q[0] = q1 * norm;
            q[1] = q2 * norm;
            q[2] = q3 * norm;
            q[3] = q4 * norm;

        }

kriswiner, 2019-09-23:
Check the sensor values, not roll and pitch. Do the sensor values make
sense?

alieng1994, 2019-09-23:
the accelerometer and the gyro make sense but the compass sometimes even freeze after some time

kriswiner, 2019-09-23:
yup, problem with the mag. Are you suing the interrupt?

alieng1994, 2019-09-23:
when I used the interrupt the register address changed so I working without an interrput is that a problem?

kriswiner, 2019-09-23:
No, but it tells me you don't really know what you are doing.

I think you have the mag configuration wrong. Please sort this out. You
should be able to get sensible magnetometer readings from the sensor, and
you should be able to calibrate this sensor. You can check by comparing the
Mz component with the sensor facing up versus facing down, they should be
equal but opposite in sign, for example.

alieng1994, 2019-09-23:
so what I need to exactly you suggest using the basic  code and then check the calibration ? thanks for your time and help and I hope everything goes fine