kriswiner/MPU9250 #390: Compensate for off-centre IMU placement

savvn001, 2019-09-26:
Hi, 

Trying to get my head around an issue that I've been thinking about for the last few days. I'm using your library with an MPU9250 to orient a quadcopter, however due to using a breakout board it wasn't possible to place the sensor in the middle of the quadcopter, where the centre of gravity/rotation would be. I instead placed it about halfway between the centre and the 'nose' of the quadcopter. I feel like this should require some compensation as the 'true' orientation of the quadcopter would be about its centre of gravity/rotation. 

Do you have any ideas as to how I could compensate for this in the code? I think it should be somewhere in the sensor fusion, particularly the acceleration (x axis in this case) but can't figure out what changes I should actually make.

Thanks!

kriswiner, 2019-09-26:
I would recommend you ignore it at first and see what kind of absolute
orientation you can achieve. In our experience this is not a barrier to
accurate absolute orientation estimation when the offsets are small enough.
There are ways to account for this, perhaps my colleague Greg can give you
a clue.

kriswiner, 2019-09-27:
The IMU provides absolute orientation; that is pitch, roll and yaw angles. Even if the IMU sensors are off-center, the orientation estimate will still settle to the right value. However, being off-center does impact the dynamic response of the IMU. Specifically, airframe rotations induce false linear accelerations. Linear accelerations do influence the indicated orientation in general. As a practical matter, this effect will be seen more at higher airframe rotation rates.

 

Placement of the IMU sensors an inch or two off of the center of rotation is typical. There are no static effects that need to be compensated but the dynamical effects are difficult to compensate… For a multirotor, if you have a problem, you will experience some instability/overshoot in the copters attitude if you are using an Euler angle stabilized flight mode. If you are using Euler angle stabilization for enhanced hovering stability and gentle manuevering, you should have no problem…

 

For more violent maneuvers I would recommend suspending Euler angle stabilization and simply use the gyroscopes for control; the gyros are fairly insensitive to the offending linear acceleration artifacts.

 

From: Tlera Corporation <tleracorp@gmail.com> 
Sent: Thursday, September 26, 2019 9:17 AM
To: kriswiner/MPU9250 <reply@reply.github.com>
Cc: Greg Tomasch <poirot@earthlink.net>
Subject: Re: [kriswiner/MPU9250] Compensate for off-centre IMU placement (#390)

 

I would recommend you ignore it at first and see what kind of absolute orientation you can achieve. In our experience this is not a barrier to accurate absolute orientation estimation when the offsets are small enough. There are ways to account for this, perhaps my colleague Greg can give you a clue.